lumapu/ahoy #1150: Feature-Request: CMT2300A GPIO Pins einstellbar in der oberfläche von Ahoy

Riker18, 2023-09-12:
Ich würde darum bitten die Pin-Belegung vom CMT2300A auch komplett frei wählbar/ einstellbar zu machen in der Weboberfläche.

Wenn man eine fertige Platine kauft die z.B. OpenDTU Default belegung hat funktioniert CMT nicht mit Ahoy da einige GPIO Pins fest im Code sind.   so geht TX nicht - weiter unten hatte das ja schon mal jemand geschrieben der dann auf opendtu gewechselt ist.

Default CMT Belegung  von openDTU

            "clk": 12,
            "sdio": 14,
            "cs": 27,
            "fcs": 26,
            "gpio2": -1,
            "gpio3": -1

So wäre man flexibel und könnte auch Platinen mit NRF+CMT frei über das Webinterface  belegen.

Vielen Dank im Voraus 

Teste ich gerne bin ja auch im Chat jeden Tag

knickohr, 2023-09-12:
Oh, schau mal, Issue #1046 schein ähnlich zu sein 😉

lumapu, 2023-09-12:
sind beide heute noch gefixt .. `0.7.50` ist im Anmarsch 😁

Riker18, 2023-09-13:
[image]

Vielen dank fürs schnelle einbauen - leider funktioniert das speichern noch nicht nach reboot immer wieder SCK 18 und SDIO 23

Riker18, 2023-09-13:
in dem exportiertem json file sind bei CMT nur die "vorherigen" 3 Werte drin - ich vermute das dass überhaupt noch nicht gespeichert wird - müssten ja jetzt 5 Werte sein.

lumapu, 2023-09-13:
ok, dann gehe ich mal reparieren, habe schon eine Vermutung

Riker18, 2023-09-13:
sonst guck auch mal im discord da gibts auch priobleme beim update vom fusion - kann aber auch sein das es damit zusammenhängt

lumapu, 2023-09-14:
ja, das SPI kann nicht mehr korrekt initialisiert werden, echt eigenartig. Bin noch nicht dazu gekommen.

DanielR92, 2023-09-14:
Könnte hier die Lösung sein? ****s://github.com/*spressif/esp-idf/issues/2455

Riker18, 2023-09-16:
guten morgen, hab kurz die 51 getestet - das speichern der CMT Ports geht leider immer noch nicht.  in der exportierten Config weiterhin nur 3 Werte drin.

muss erst mal arbeiten - teste heute Nachmittag mal mit einem werksreset.

Riker18, 2023-09-16:
hab noch schnell  einen werks-reset gemacht dann geht der NRF auch nicht mehr da sind dann die GPIOS default vom CMT drin da ist noch was vertauscht.   CMT speichern geht weiterhin nicht.

knickohr, 2023-09-16:
Hier geht’s weiter :  Issue #1159

lumapu, 2023-09-16:
gefixt in `0.7.54`

Jakobi88, 2023-10-21:
Frage am Rande:
ist der GPIO3 vom CMT2300A zwingend nötig? Mein PCB hat offenbar keine Verbindung zwischen GPIO3 und dem ESP32...

closed 2023-12-29 by lumapu