lumapu/ahoy #1151: [Bug] Logo erscheint Nachts nicht mehr im ePaper

knickohr, 2023-09-12:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.49, vielleicht auch schon 0.7.48

### Github Hash

…

### Build & Flash Method

ESP Tools (flash)

### Setup

MQTT, ePaper, Fusion-Board

### Error description

Gerade eben festgestellt das das Logo nicht mehr bei Offline im ePaper erscheint, trotz angehaktem „Turn off while inverters are offline“.

Problem kann schon seit der 0.7.48 bestehen, habe nicht darauf geachtet.

lumapu, 2023-09-12:
laut Code kann das nicht sein, hast du die DTU neu gestartet nachdem das Setting aktiv gemacht wurde? -> klar, aber ich muss das fragen 😉

lumapu, 2023-09-12:
evtl. ist dein `TotalPower` noch auf einem Wert > 0 hängen geblieben.

knickohr, 2023-09-12:
Du hast Recht, es ist tatsächlich ein Wert bei 1,2W hängen geblieben 😵

lumapu, 2023-09-12:
dann brachen wir evtl. einen Schwellwert

knickohr, 2023-09-12:
Nee, man müßte das von der State-Machine abhängig machen. Wenn (alle Inverter) offline, dann aus die Maus.

knickohr, 2023-09-21:
Konnte es bis jetzt nicht verifizieren, gehe aber davon aus das es nicht mehr vorkommt wenn ein Inverter noch ein paar Milliwatt stehen läßt.

knickohr, 2023-09-29:
Mach mer mal dicht 😉

Bisher nicht mehr aufgetreten.

knickohr, 2023-09-29:
Verdammt !

Gerade erst zu gemacht und schon wieder auf machen. Er hat es wieder getan, bei einem Inverter sind knapp 2W hängen geblieben und somit erscheint das Logo Nachts nicht 😢

[image]

lumapu, 2023-09-30:
hast du auch die 15 Minuten abgewartet? Wir hängen doch jetzt an der Statemachine

knickohr, 2023-09-30:
Ja habe ich. War bis vor einer Stunde immer noch ohne Logo 😅

You69Man, 2023-09-30:
@lumapu Der Fix den du hierfür in commit 181cc6fc gemacht hast, wirkt imho nur auf die mono Displays. 
Die mDisplayData Struktur wird vom ePaper (noch) nicht verwendet (wäre aber auch mal gut).

Vorerst müsste man daher also auch hier (in Display.h) noch folgendes ändern:
#if defined(ESP32)
        else if (mCfg->type == 10) {
        # mEpaper.loop(totalPower, totalYieldDay, totalYieldTotal, nrprod); -> alt
            mEpaper.loop((allOff) ? 0.0 : totalPower, totalYieldDay, totalYieldTotal, nrprod); -> alt
            mRefreshCycle++;
        }

lumapu, 2023-09-30:
@You69Man wird genauso übernommen, danke für den Hinweis

knickohr, 2023-11-15:
Auch hier kann man zu machen, ist nicht mehr aufgetreten. Hoffentlich ! 😅

closed 2023-11-15 by knickohr