lumapu/ahoy #1154: Details Webgui, System

Peertje-62, 2023-09-13:
Hi, happy with version, 7.50, ESP32, Build b06f8e9.
I am not using MqTT and since MqTT info is moved to system, it is  labelled ‘False’ and red (other than  CMT2300A; not enabled ).

I would call it 'Icing on the cake' if this could be changed.
Kind regards

lumapu, 2023-09-15:
what do you want to be changed? The color or the text? Please give an example

Peertje-62, 2023-09-15:
Dear Lukas,
I like uniformity, Short and sweet.
NRF24L01 – connected and green
CMT2300A – Not enabled and red

I am not using Mqtt = enabled  – false and red 

my proposal would be:
an option to disable Mqtt – not enabled and red.
If enabled and working – connected and green.
If enabled and not working – false and red.

If possible, no hurry, until now it’s working nice 0.7.50 esp 32
I am well aware that I am not the only user.
Other users may have different ideas.

Kind regards
Herman

Peertje-62, 2023-09-15:
and may be '' Sunrise & Sunset'' instead of Sun, in system.
sorry

lumapu, 2023-09-15:
This is how it is in `0.7.50`:
MqTT not enabled:
[image]

MqTT enabled and connected:
[image]

MqTT enabled but not connected:
[image]

Should I change anything, I think your request is already met.

Peertje-62, 2023-09-15:
Hi, I am using 7.50, mine looks differend, what did I do wrong? or do I get it wrong!
(also Night behaviour if disabled could be red)
[image]
good night

Peertje-62, 2023-09-16:
v0.7.51:I don't see any differences, it doesn't matter
Kind regards, Herman
[image]

closed 2023-09-16 by Peertje-62