lumapu/ahoy #1156: Version 0.7.36 unregelmäßige Unterbrechungen mit ESP8266 und ESP32

hajulied, 2023-09-15:
Moin,
habe bisher auf einem ESP8266 AhoyDtu weitgehend störungsfrei mit der Version ß.6.9 betrieben.
Nach dem Update auf 0.7.36 treten unregelmäßig Unterbrechungen der Verbindung auf. Heute habe ich 0.7.36 auf einen ESP32 geflasht, aber auch hiermit treten die Unterbrechungen auf.
Beide Kits sind auf einer Platine aufgebaut und die Änderungen der PIN-Belegung sind berücksichtigt. Mit 0.6.9 läuft Ahoy wie beschrieben fast störungsfrei.

lumapu, 2023-09-16:
bitte auch mal die Development-Version testen, wir sind schon wieder um einiges weiter ... aktuell `0.7.53`
https://fw.ahoydtu.de/fw/dev/

hajulied, 2023-09-16:
Moin,
  
Danke für den Tipp.   
  
Bin gerade dabei. Habe mir vorhin die V 0.7.51 runtergeladen und installiert. Bin gerade am Testen.

hajulied, 2023-09-18:
mmhhh auch mit V 0.7.51 hatte ich die Aussetzer.
Jetzt habe ich die V 0.7.36 auf einen anderen ESP32 (38 polig mit der etwas abgesetzten WLAN Antenne am oberen Ende) geflasht. Seit gut einer Stunde habe ich keine Verbindungsunterbrechungen mehr. 
Ich weiß zwar nicht warum, aber es läuft.
Vielen Dank für die Hilfe.