lumapu/ahoy #1157: Leichte verbesserung: Gesamtertrag im dem Feld unter die weisse Linie setzenp, rechts davon (Gesamtertrag in €)  und noch eine Eingabefeld in System für die Stromkosten (0,40€)vorsehen .Somit weiss jeder was gespart wird

persil14, 2023-09-16:
Ihr könnt mir gerne antworten.. Pesilber@gmx.de Grüße Peter

rmayergfx, 2023-09-16:
Antworten erfolgen sinnvollerweise immer im zugehörigen Thread, damit auch jeder mitlesen kann was bereits geschrieben wurde. PNs führen ein Git als absurdum, da nichts mehr nachvollzogen werden kann. 

Bitte in Zukunft auch etwas mehr Mühe geben und nicht alles in den Titel reinschreiben sondern dazu auch eine vernünftige Beschreibung in den Text und v.a. mit welchem System gearbeitet wird. 

Zur Info, Yield Total wird intern im Inverter gespeichert, eine Berechnung müsste auf dem System erfolgen und je nach Aufbau stimmt dieser "Gesamtertrag in €" gar nicht, einmal von der Begrifflichkeit, es gibt auch noch andere Länder die keinen € haben und dennoch AhoyDTU einsetzen und zum anderen kann man ohne die Einbindung der Zähler und einer sogenannten Nulleinspeisung überhaupt nicht nach diesem Wert gehen, da sind dann viele Leute am Ende des Jahres erstaunt, das diese "Einsparung" sich nicht auf der Stromrechnung niedergelassen hat, weil einfach zu viel ins Netz eingespeist wurde, ohne jegliche Vergütung.

knickohr, 2023-09-16:
Ich kann den Betreff leider nicht editieren. Kann wohl nur der Author selbst und Auserwählte 😉

Zum Thema :  Nette Idee, wird aber vermutlich schon allein an den variablen Strompreisen scheitern. Nicht jeder ha5 einen fixen Tarif.

persil14, 2023-09-16:
Hi, werde Besserung geloben.
Grüße Peter 

Sent by from my Smartphone.  

Am 16. September 2023, 14:22, schrieb rmayergfx:

Antworten erfolgen sinnvollerweise immer im zugehörigen Thread, damit auch jeder mitlesen kann was bereits geschrieben wurde. PNs führen ein Git als absurdum, da nichts mehr nachvollzogen werden kann.

Bitte in Zukunft auch etwas mehr Mühe geben und nicht alles in den Titel reinschreiben sondern dazu auch eine vernünftige Beschreibung in den Text und v.a. mit welchem System gearbeitet wird.

Zur Info, Yield Total wird intern im Inverter gespeichert, eine Berechnung müsste auf dem System erfolgen und je nach Aufbau stimmt dieser "Gesamtertrag in €" gar nicht, einmal von der Begrifflichkeit, es gibt auch noch andere Länder die keinen € haben und dennoch AhoyDTU einsetzen und zum anderen kann man ohne die Einbindung der Zähler und einer sogenannten Nulleinspeisung überhaupt nicht nach diesem Wert gehen, da sind dann viele Leute am Ende des Jahres erstaunt, das diese "Einsparung" sich nicht auf der Stromrechnung niedergelassen hat, weil einfach zu viel ins Netz eingespeist wurde, ohne jegliche Vergütung.

rmayergfx, 2023-09-16:
@persil14 
Bitte keine Fullquotes und die E-Mail Adresse würde ich auch nicht überall im Klartext reinschreiben, dafür bekommst du vom System eine automatische E-Mail wenn jemand auf deine Beiträge antwortet. Geh bitte mal an einen PC oder Tablet und editiere das vernünftig. Danke

technics42, 2023-09-16:
To me this doesn't make sense, due to arguments above and in addition: you are only accounting produced power, but not consumed power by yourself! (Most people are not consuming all generated power and/or do not use "zero export")

MetaChuh, 2023-09-16:
@persil14 

> Hi, werde Besserung geloben.
Grüße Peter

vielen dank, wir werden dir, für die durchführung deiner gelobung, die kontodaten schicken, damit du 50% deiner 2023er jahrensersparnis an dieses open source projekt überweisen kannst.

greetings,
metachuh

stefan123t, 2024-10-29:
@persil14 ich kann den Vorschlag verstehen, man möchte wissen wieviel man theoretisch eingespart hat durch seine Investition in eine PV-Anlage. So machen es die Discounter mit einer Solaranlage auf dem Dach ja auch seit Jahren. Da wird dann froh verkündet wieviele Tonnen CO2 und wieviel Stromkosten man sich bereits zusammen "gespart" hat. Dass es eine sog. Milchmädchenrechnung ist, sollte uns allen klar sein.

Mir persönlich gefiele die Zusatz-Option mit der CO2 Ersparnis, da die als Faktor m.E. sogar fest hinterlegbar sein sollte ?

> Die Formel lautet:
> Jährliche Produktionsleistung der PV-Anlage (kWh / a) x 0,584 kg. Das Ergebnis teilen Sie durch 1.000 kg/t. 
> Mit dieser Berechnung erfahren Sie, wie viele Tonnen CO2 Sie mit Ihren Solarzellen pro Jahr einsparen.
https://www.ibc-solar.de/solar-ratgeber/co2-ersparnis-photovoltaik/

Man könnte ja neben dem Stromkostenfaktor, z.B. 0,40 €/kWh noch ein Freitext-Feld (String) für die Landeswährung Euro, Dollar, Rubel, Pesos, Rupien, Drachmen, Lira, etc. pp. erlauben, da kann dann jeder das Unicode Zeichen oder bis zu drei Buchstaben für seine Währung eingeben.