lumapu/ahoy #1158: [Bug] Unauthorized Reboot

panic2k, 2023-09-16:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

ba218edbdb1b0a168e0c721bc2259fcc97c57f8a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

This DTU monitors 3 inverters, sometimes it freezes so i wanted to add some crometab

### Error description

Unauthorized Reboot is possible

curl 'http://192.168.XXX.XXX/reboot' --compressed -H 'User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:109.0) Gecko/20100101 Firefox/117.0' -H 'Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,*/*;q=0.8' -H 'Accept-Language: de,en-US;q=0.7,en;q=0.3' -H 'Accept-Encoding: gzip, deflate' -H 'DNT: 1' -H 'Connection: keep-alive' -H 'Referer: http://192.168.XXX.XXX' -H 'Upgrade-Insecure-Requests: 1' -H 'Pragma: no-cache' -H 'Cache-Control: no-cache'

No cookie stuff needed - easy - but..... hmmmm......

lumapu, 2023-09-16:
is your installation protected by a password? You mention that you are able to call `/reboot` without any authentification?

MetaChuh, 2023-09-16:
@lumapu 
was always like this.
calling dtu_ip/reboot always triggers a reboot, regardless of the protection mask.

some of us use this hack to reboot the dtu if the api is stuck at delivering „null“ instead of a json, so if you change anything ask the others first … if you don‘t get any response, it‘s ok.

lumapu, 2023-09-16:
ok got your point, the security risk is low about that. In general: Ahoy isn't secure by itself.
I will wait for response by others

MetaChuh, 2023-09-16:
@panic2k 
good eyes and thanks for sharing.
as long as you don‘t expose your dtu to the web, or at least geo protect your port forwarding or reverse proxy, it will be little of concern for now.

panic2k, 2023-09-17:
My DTU is certainly protected against unauthorized changes with password. In local network, I would actually not have to worry about it - but if someone wants remote access without homeassistant, it could be used at least for denial of service. I thought it was a little unusual. Just wanted to mention this - risk is for sure low

lumapu, 2023-09-17:
thank you for reporting, I leave it open for a while,  maybe someone has an idea to solve this with a small implementation

stefan123t, 2024-10-29:
@lumapu OpenDTU and OpenDTU-OnBattery (OBOD) developers switched to [ESPAsyncWebServer](https://mathieu.carbou.me/ESPAsyncWebServer/) fork from **mathieucarbou**.
This allows for an Authentication Middleware to intercept and secure these API requests in case you want that too ?