lumapu/ahoy #1159: CMT Pin belegung wird nicht gespeichert

teletapi, 2023-09-16:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.51

### Github Hash

GIT SHA: 77ba278 :: 0.7.51

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

Keine änderung

### Error description

Fusionsboard V2 mit FW.07.51 und Epaper, CMT enable geht und wird auch gespeichert
Ändern der Pinbelegung CMT kann nicht gespeichert werden getestet habe ich die vorgeschlagene Belegung SCLK von Pin 6 auf "Off" nach speichern und neustart wurde "off" nicht übernommen und Pin 6 ist wieder hinterlegt
Siehe discord "reference-board"

lumapu, 2023-09-16:
gefixt in `0.7.53`

Riker18, 2023-09-17:
53er Version per webtool geflasht da kein Update ging.

Default-Config noch verdreht siehe Bild - wenn man sie aber ändert und speicher ist nach einem reboot CMT und NRF verbunden :)

[image]

teletapi, 2023-09-17:
Ich schließe das dann mal dürfte ja jetzt funktionieren. Danke an alle

closed 2023-09-17 by teletapi

Jakobi88, 2023-09-23:
@Riker18:
blöde Frage:
hast du es geschafft einen HMS-Inverter mit 
SCLK auf GPIO12 
SDIO auf GPIO 14
CSB auf GPIO 27
FCSB auf GPIO26

 zu verbinden?

Bei mir klappt's aktuell nicht :/

Riker18, 2023-09-23:
Ja läuft

Jakobi88, 2023-10-12:
Darf ich fragen, was du für eine Antenne auf der 868 Mhz-Seite verwendest??

Riker18, 2023-10-12:
https://www.amazon.de/dp/B06Y1RNFBT?psc=1&ref=ppx_yo2ov_dt_b_product_details