lumapu/ahoy #1160: dowload last development software

Peertje-62, 2023-09-17:
Hardware
  * [x ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

* [x ] external antenna (SMA)

### Power Stabilization:
* [x ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.7.54

### Build & Flash Method:
* [x ] ESP Tools

Hi,
dowload last development software **ahoy_v0.7.54**
in in zip file : 
esp32;  230916_ahoy_0.7.53_181cc6f_esp32.bin, ota, reboot,
Ahoy Dtu; page You are using development version 0.7.51. In case of issues you may want to try the current stable release: 0.7.36
System page; version displays 0.7.51

esp 8266; 230916_ahoy_0.7.53_181cc6f_esp8266.bin, ota reboot
Ahoy Dtu page ; You are using development version 0.7.51. In case of issues you may want to try the current stable release: 0.7.36
System page version displays 0.7.53

Gubi2023, 2023-09-17:
jupp, 7.53 wird bei mir auch nich installiert...

dtuuser, 2023-09-17:
Bei mir auch.
Warte gespannt auf

added active power control in W to live view #201, #673

Peertje-62, 2023-09-17:
Hi
ota, 7.55 on esp8266 & esp32 same problem, as before.
Kind regards

Peertje-62, 2023-09-17:
Hi,
ESP-Flasher 7.55  on esp8266 & esp32 works
Kind regards

Peertje-62, 2023-09-17:
Don't know what happened, In set up!
Radio gone, 
Inverters gone
Display gone

knickohr, 2023-09-17:
This Version is buggy, use 0.7.55 !

rmayergfx, 2023-09-17:
0.7.51 is not updating to 0.7.55 with 230917_ahoy_0.7.55_a18a738_esp32.bin from https://github.com/lumapu/ahoy/actions/runs/6213014391

Ollipop030, 2023-09-17:
> 0.7.51 is not updating to 0.7.55

OTA ist broken in 0.7.51 - 0.7.54, use ESP Flasher or similar. Save your settings before flashing, ahoy will lose it´s settings.

Peertje-62, 2023-09-17:
Thx,
''set up from scratch''
works

closed 2023-09-17 by Peertje-62