lumapu/ahoy #1161: Mqtt ClientId wird nicht benutzt

martinriedel, 2023-09-17:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

ba218edbdb1b0a168e0c721bc2259fcc97c57f8a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Unwichtig

### Error description

Kurz und schmerzlos. Die gesetzte Mqtt client id wird nicht benutzt, stattdessen weiterhin {Device Name - die letzten 6 Zeichen der MAC Adresse}

lumapu, 2023-09-17:
der Code selbst spricht was anderes. Dort sieht alles korrekt aus, evtl. muss man mit Debug-Messages prüfen, ob die funktion angesprochen wird.

martinriedel, 2023-09-17:
Das wundert mich auch. Habe es ja schließlich eingebaut, nur konnte ich es damals nicht kompilieren und dachte mir 'sieht gut aus, wird passen'.
Debuggen funktioniert nur über VS Code? Oder kann man noch anders loggen?

martinriedel, 2023-09-21:
Hab den Fehler gefunden, in der settings.h -> jsonMqtt fehlt der clientId Eintrag. Ich stelle einen PR

lumapu, 2023-09-21:
sehr gut, ich weiß die Stelle die du meinst. Ist zwar oben in der Struktur, aber nicht unten beim speichern und laden dabei 😄

closed 2023-12-29 by lumapu