lumapu/ahoy #1162: Feature Request: Webapi - power_ack (enable/disable inverter)

reserve85, 2023-09-17:
Hi,

ich hätte einen Request für die WebAPI: könnten die Befehle "Power-On" und "Power-Off" ebenfalls quittiert werden? Ähnlich dem power_limit_ack nach einem Limit-Befehl...

Idee: Ggf. wäre es gut bei den Control-Commands (Power-On/Power-Off, Power-Limit) einen (optionalen) Identifier mit übergeben zu können (z.B. ein uint16).
Dann könnte man den Command-Identifier nach dem Acknowledge vom Inverter einfach in der WebApi als "LastCommandAcknowledge" publishen.

z.B.:
`http://{AHOY_IP}/api/ctrl -> '{{"id": 0, "cmd": "limit_nonpersistent_absolute", "val": 500, "identifier": 12345}}'''`
und in der RestAPI kommt dann wenn es erfolgreich zurückgemeldet wurde:
`http://{AHOY_IP}/api/inverter/id/0/last_ack = 12345`
Wenn es nicht acknowledged wurde bleibt der alte Identifier in dem Feld stehen.

lumapu, 2023-09-17:
weiß garnicht, ob der Wechselrichter auf diese beiden commands antwortet. Muss ich erst prüfen

knickohr, 2023-09-17:
Sollte es funktionieren, dann hätte ich diese beiden Ack auch gerne über MQTT, natürlich mit QOS=2. by the way, funktioniert eigentlich Power on/of Topic, es fehlt ja in der Doku ?  Habe aber schon mitbekommen das es anscheinend geht.

stefan123t, 2024-10-29:
@lumapu eigentlich sollten alle Befehle vom WR mit dem Kompletemn `MainCmd | 0x80` quittiert werden.
Die [DEVCONTROL_ALL (`0x51`) Commands](https://github.com/lumapu/ahoy/wiki/Protocol#welche-device-control-devcontrol_all-0x51-sub-kommandos-gibt-es) findest Du hier im Protocol Wiki.
Die Quittungen müssten also eigentlich alle mit `0xD1` beginnen, vielleicht mal in die Log schauen ?