lumapu/ahoy #1164: [Bug]

You69Man, 2023-09-18:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.51 ff

### Github Hash

3b4365

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

Start without time sync (useful in AP-Only-Mode)

### Error description

Wie hier schon beschrieben: https://github.com/lumapu/ahoy/issues/1125

>Ich fürchte, da hat's noch eine Kleinigkeit mit der neuen mainloop in der 0.7.51 (commit https://github.com/lumapu/ahoy/commit/77ba2783a6299497ccee5fb43e24835e678a4185).
Gerade testhalber aufgespielt (ESP8266, HM800): Wifi ist on, MQTT ist on, RF24 ok, aber Inverter bleibt not available, und Display status daher "offline".
Zurück auf 0.7.50 ist alles wieder gut.

Das Phänomen exisitert seit 0.7.51, ist aber auch in 0.7.56 noch vorhanden.

Mittlerweile habe ich festgestellt, dass dieses Verhalten mit der Option `Start without time sync (useful in AP-Only-Mode)` zusammenhängt. Ist diese gesetzt, beginnt das Inverter-Polling *nicht* nach dem Start. Ist sie nicht gesetzt, funktioniert es wie erwartet. 
Bis 0.7.50 war das Verhalten anders. Hier hatte ich die Option bewusst gesetzt, weil ich eine lokale Display-Anzeige der Leistungswerte haben wollte, auch wenn mein WLan Router ausgeschaltet war. Ich denke, das sollte auch in Zukunft möglich sein, eine lokale Inverter-Abfrage und Display Anzeige, unabhängig von Wlan und MQTT zu haben.

lumapu, 2023-09-18:
Danke für's ermitteln, das klingt logisch.
Werde prüfen wie man das am besten behebt.

fraggy2k, 2024-05-01:
Moin Moin, 
gab es hier schon was neues? 
Heute musste ich auch feststellen, dass bei der 0.8.83 mit gesetzter Option `Start without time sync` der Inverter nur als offline im Display angezeigt wird.

Sobald ich dann das Smartphone mit dem Ahoy WLAN verbunden habe, ohne auf die GUI zu wechseln, wurden die Daten empfangen und auf dem Display angezeigt. Die Zeit wurde nicht synchronisiert, wie mit der Option gewünscht.

Hardware ESP8266

lumapu, 2024-05-30:
in der aktuellen Dev sollte dieses Problem nicht mehr auftreten. https://fw.ahoydtu.de/fw/dev/

closed 2024-08-16 by lumapu