lumapu/ahoy #1165: Active Power Control in Live View auf Mobilgeräten

gone-for-coding, 2023-09-18:
Die Änderung vom Text in Active Power Control sorgt auf kleinen Mobilgeräten für relativ viel Platzverschwendung durch die Umbrüche.

Folgende Ideen hätte ich:

- Auf Mobilgeräten einen anderen Text anzeigen (z.B. wieder Limit)
- Generell abkürzen, z.B. APC mit HTML acronym Tag als Hover für den vollen Text
- Elemente anders sortieren könnte auch helfen

lumapu, 2023-09-18:
verstehe das Anliegen, mal sehen was sich am einfachsten implementieren lässt

Gubi2023, 2023-09-19:
blaues Feld zu schmal für Yield total
[image]

gone-for-coding, 2023-09-19:
Danke @lumapu! Sieht wieder gut aus :) Von mir aus können wir den hier zu machen oder für das andere Thema offen lassen.

Ollipop030, 2023-09-19:
> blaues Feld zu schmal für Yield total

Scheint ein kleines Display am Handy zu sein. Kannst du im Browser vielleicht die Schriftgröße verkleinern? Oder hast du Bedienungshilfen eingeschaltet? Die Schrift sieht irgendwie so groß aus.

Gubi2023, 2023-09-19:
stimmt, "grösserer Text" war eingestellt. Jetzt sieht es so aus:
[image]

Gubi2023, 2023-09-19:
könnte aber bei vierstelligen Anzeigen evtl. auch zu schmal werden(?)

closed 2023-09-19 by gone-for-coding