lumapu/ahoy #1166: In Vers 7.55 u. 7.56 wird in den Settings keine Änderung angenommen[Bug]

Karlo49, 2023-09-18:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.55 u 0.7.56

### Github Hash

xxx

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

x

### Debug Serial Log output

```shell
xx
```

### Error description

Nach Änderung in den Settings und drücken der Taste "Save" passiert erst garnichts und dann keine Verbindung mehr.
Nach ein paar Minuten ist die Seite wieder da ohne Änderung

Ollipop030, 2023-09-18:
Grad probiert mit 0.7.56, bei mir gehts. Hast du da spezielle Settings bei denen es nicht klappt?

Ollipop030, 2023-09-18:
Habe kein MQTT, aber grad mal was eingetragen. Es wird zumindest gespeichert, Powel Level auch. Allerdings bin ich auf einem ESP32, evtl. betrifft das nur den 8266?

lumapu, 2023-09-18:
bei Möglichkeit wäre das serielle Log während dem Speichern sehr interessant. Dafür einfach einen zweiten Tab öffnen und dort die WebSerial mitlaufen lassen. Dann in ersten Tab auf save klicken und den Inhalt der WebSerial analysieren und uU. hier posten

blueline13, 2023-09-19:
Ich habe bei mir genau das gleiche Problem mit einem ESP8266 auch mit der 57 Version.

1) Controller mit Erase geflasht
2) Nur Wifi Parameter gesetzt
1) System Reset bis auf WiFi
2) Serielle Console auf
3) Wert ändern und speichern
4) Nun passiert auf der Weboberfläche nichts mehr. Es fehlen die üblichen Meldung:
    - Save Settings
    - Settings successfully saved. Rebooting. Automatic redirect in 20 seconds.

5) Nach dem Speichern kommt in der Serielle Konsole nur noch(die Stack Infos habe ichweggelassen):

===========
Soft WDT reset

Exception (4):
epc1=0x401002f8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
===========

Also rennt er irgendwie in einen Loop und hängt. Keine Ahnung ob das hilft oder irgendetwas damit zu tun hat, ich habe noch die "alte" default Belegung vom NRF (CS=D8, CE=D4, IRQ=D3).

Ich hoffe das hilft erst mal.

Karlo49, 2023-09-19:
@lumapu  hier das gewünschte Log:
[Aenderung Settings 0756.txt](https://github.com/lumapu/ahoy/files/12657711/Aenderung.Settings.0756.txt)

Karlo49, 2023-09-19:
Habe jetzt den ESP8266 mini mit dem ESPTool unter Linux neu geflasht.
Selbes Ergebnis beim ändern der Settings!
[ESPTool 07566.txt](https://github.com/lumapu/ahoy/files/12657958/ESPTool.07566.txt)

lumapu, 2023-09-19:
habe eben mit einem ESP8266 (mit Elko) und der Version `0.7.57` getestet. Alles hat einwandfrei geklappt:
1. ESP auf 0.7.57 geflasht
2. Factory Reset (aus WebGUI) gemacht
3. Per Smartphone WiFi mit meinem heimischen WLAN verbunden -> save, reboot
4. Haken für "Serial Debug" bei "System Config" gesetzt, einen 4-kanaligen Wechselrichter angelegt -> save, reboot
5. Mqtt mit broker, user, clientId, passwort gesetzt -> save, reboot

Wie war euer vorgehen?

lumapu, 2023-09-19:
jetzt habe ich mit dem Online-Installer gestartet, daher Version `0.7.36`
1. Per Smartphone WiFi mit meinem heimischen WLAN verbunden -> save, reboot
2. Update per WebUi auf `0.7.57`
3. Inverter, MqTT -> save, reboot -> crash 🔥
4. debugging ....

Karlo49, 2023-09-19:
1. Esp ist auf 0.7.50  -> Ok
2. Update per WebUi auf 0.7.56 -> Ok
3. Settings Änderung Power Level  -> save  -> Absturz

lumapu, 2023-09-19:
Fehler gefunden 😀
War noch ein Artifakt aus der Loopumstellung in Version `0.7.58`. Wieder mal danke an alle, die hier so fleißig reporten und testen.

MetaChuh, 2023-09-19:
@lumapu
hoffe du hast nicht den gleichen blöden symetrie gedanken wie ich:
tripple seven 0.7.77 soll die most resilient version ever sein, drum nach 0.7.76 die sub-sub-sub versionierung 0.7.76.1 etc. 😉

ernst und mega appreciation: danke dir vielmals für alles was du für uns alle machst 🤗

Karlo49, 2023-09-19:
Sprung von 0.7.57 auf 0.**5**.58 ???

lumapu, 2023-09-19:
ja Fehler passieren in der hektik, immerhin nur der git-commit und das Changelog. Die Version im ESP ist mit `0.7.58` korrekt.

blueline13, 2023-09-19:
Getestet mit der eben hochgeladenen Version. Es funktioniert wieder wie gewohnt!

Die Versionsnummer ... wir sind alle nur Menschen und da passieren nun mal Fehler  .... Ist ja auch schon etwas später ;-)

Wirklich eine Super Leistung die du und alle andere hier erbringt. Bitte weiter so!!!

closed 2023-12-29 by lumapu