lumapu/ahoy #1168: Total AC Power halbiert bei kleiner 100 Watt???

soelli22, 2023-09-19:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.7.57
Github Hash: _______

### Build & Flash Method:

* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

Hallo zusammen.

Bei mir tritt folgendes Phänomen auf:
Ich speise über einen Akku (Zendure Solar flow) in meinen HM800 konstant 60 Watt ein.
Dabei erscheint jedoch bei der DTU auf dem Display und über die Website nur etwa die halbe Leistung.
Die Leistung je Eingang stimmt, nur nicht die Gesamtleistung.
Die angeschlossene Fritz-Steckdose erfasst die korrekten 60 Watt und auch mein Shelly Pro 3PM zeigt 60 Watt Einspeisung an.
Es halbiert aber nicht generell die Anzeige. Bei etwa 120 Watt oder mehr passt es plötzlich wieder.
Gehe ich auf 60 Watt zurück, halbiert sich die Leistung AC Power total wieder.

[image]
[image]

MetaChuh, 2023-09-19:
@soelli22 
wie gehst du in den hm-800 von der batterie rein ?
weil auf den ersten blick sieht es auf deinen screenshots aus, als ob du per y kabel auf beide ports in den hm-800 reingehst, was die beiden hm mppt messungen wegen nicht isloierter leitungen durcheinander bringen kann.

das einzige was eine beliebige dtu darstellt sind normalerweise die werte die der inverter per funk liefert.

welche messwerte hast du, wenn du zb. nur einen der beiden hm mc4 eingänge verwendest ?
und falls du vorher bereits eine ältere ahoy dtu verson probiert hast, war es anders, mit vorherigen versionen ?

lumapu, 2023-09-19:
DC Power wird richtig gemessen denke ich. Allgemein muss ich @MetaChuh Recht geben, die Werte stammen direkt vom Wechselrichter

Ollipop030, 2023-09-20:
Aber warum ist die Efficiency so niedrig? Die sollte locker über 90% liegen. Steigt der Wert sobald du auf 120W regelst?

Kenne das Datenblatt jetzt nicht im Detail, aber haben die HM nicht eine Startspannung von 22V? Laufen die mit so einer kleinen Spannung stabil?

soelli22, 2023-09-20:
Guten Morgen.

Danke an alle für die Gedankenanstöße.
Ja, der Akku hängt über ein Y-Kabel am HM800.
Lt. Fritz DECT werden die 60 Watt auch ohne große Probleme konstant über Nacht an den WR geliefert.
Einzeln habe ich es noch nicht probiert.
Mit einer älteren Ahoy Version habe ich noch nicht drauf geachtet, kann also keine Aussage dazu treffen.

Die Startspannung vom HM600/700/800 liegt tatsächlich bei 22 Volt. Trotzdem läuft es mit 20,8 Volt auch gut. (die LED vom HM800 blinkt aber auch da nur im "halben Takt".

Die Efficiency wird wahrscheinlich auch halbiert, ab ca. 120 Watt läuft sie mit 95%.

Wahrscheinlich kann Hoymiles unter 120 Watt und 22 Volt nicht richtig rechnen... ;-) Das letztendliche Einspeiseergebnis stimmt aber.

kiu77, 2023-09-20:
@soelli22 
Die Frage war, ob Du vom Akku per Y-Kabel auf beide Eingänge des Inverter gehst.
Das wäre nicht nur nicht nötig bei den Leistungen, sondern könnte Probleme verursachen, da die beiden MPPT_Tracker sich sonst gegenseitig auskegeln und Chaos veranstalten können.
Der HM800 kann es definitiv gut mit kleinen Leistungen und hohem Wirkungsgrad. Bei mir tut er es auch mit unter 10W korrekt. Außerdem ist die Spannung wirklich zu gering. Wenn Du mit einem 24V-Akku hantierst, wäre die neue Serie, also der HMS-800 die bessere Wahl, denn der geht ab 16 V. Oder aber Du sattelst auf ne 48V-Lösung um...

MetaChuh, 2023-09-20:
datenblatt: hier noch mal das "richtige" datenblatt der hm's mit 2 eingängen.
bei vielen übersetzungen wurden die unterschiedlichen mppt spannungsbereiche und andere werte, ausser der endleistung,  zwischen hm 600/700/800 für alle gleich angegeben, sodass der anschein entstand, dass alle geräte gleich und nur gedrosselt sind.
dadurch auch die unterschiedlichen erfahrungen mit 24 v direkteinspeisung aus batterien zwischen zb hm 300/600 und 400/800.

[Datasheet_HM-600-700-800_AP_EN_V202110.pdf](https://github.com/lumapu/ahoy/files/12676421/Datasheet_HM-600-700-800_AP_EN_V202110.pdf)

@soelli22 bitte mal einen mc4 strang abhängen und schauen was langfristig passiert, wenn du's noch nicht gemacht hast.

ps: wenn dich die kriechspannungswerte des offenen eingangs stören, dann einen pulldown terminator plug kaufen oder anfertigen und anstecken:
mc4 männchen > 1kΩ widerstand mit min 0.25w > mc4 weibchen

knickohr, 2023-09-21:
Gans da nicht mal eine etwas ausführlichere Diskussion im Discord ?  Ich meine mich daran zu erinnern.

MetaChuh, 2023-10-12:
@soelli22 

bitte schließe dieses inaktive issue ticket, wenn geht mit einer kurzen info deiner schritte nach deinem letzten post, welche ausgereicht haben, um dieses issue hier nicht weiter aktiv zu verfolgen.

vielen dank

soelli22, 2023-10-14:
Hiermit schließe ich das Issue, da der Fehler nicht auf Seiten Ahoy liegt, sondern wohl grundsätzlich an der Konstellation SolarFlow/Hoymiles. Danke für eure Unterstützung.

closed 2023-10-14 by soelli22