lumapu/ahoy #1173: Inverter kommen sich bei zu kurzem Intervall in die Quere

rejoe2, 2023-09-21:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.59

### Github Hash

3a29d6c

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

- 6 MI-1500 3rd gen (# 0 - # 5) 
- 1 MI-600 (2nd gen) (# 6)
- 1 HM-1500 (# 7, hängt an einem Akku und bekam zum Testzeitpunkt noch alle 15 Sek. ein limit-Kommando)
- Intervall 2 Sek.
- retransmits (5, default)

### Debug Serial Log output

```shell
07:37:35 I: (#6) sth. missing: Request Retransmit 0x09
07:37:35 I: TX 11 CH23 | 09 #6 (MI-600) 86 09 76 68 09 f2
07:37:36 I: (#1) enqueued cmd failed/timeout
07:37:36 I: (#1) Requesting Inv SN #1
07:37:36 I: (#1) enqueCommand: 0x0b
07:37:36 I: (#1) prepareDevInformCmd 0x0b
07:37:36 I: TX 27 CH40 | 15 #1 86 09 76 68 80 0b 00 65 0b d6 a0 00 00 00 00 00 00 00 00 d3 14 a4
07:37:36 I: (#2) no complete Payload received! (retransmits: 5)
07:37:36 I: (#2) Requesting Inv SN #2
07:37:36 I: (#2) enqueCommand: 0x0b
07:37:36 I: (#2) prepareDevInformCmd 0x0b
07:37:36 I: TX 27 CH61 | 15 #2 86 09 76 68 80 0b 00 65 0b d6 a0 00 00 00 00 00 00 00 00 d3 14 01
07:37:36 I: (#3) enqueued cmd failed/timeout
07:37:36 I: (#3) Requesting Inv SN #3
07:37:36 I: (#3) enqueCommand: 0x0b
07:37:36 I: (#3) prepareDevInformCmd 0x0b
07:37:36 I: TX 27 CH75 | 15 #3 86 09 76 68 80 0b 00 65 0b d6 a0 00 00 00 00 00 00 00 00 d3 14 17
07:37:36 W: (#1) nothing received
07:37:36 W: (#2) nothing received
07:37:36 W: (#3) nothing received: complete retransmit
07:37:36 I: (#3) prepareDevInformCmd 0xb
07:37:36 I: TX 27 CH3 | 15 #3 86 09 76 68 80 0b 00 65 0b d6 a0 00 00 00 00 00 00 00 00 d3 14 17
07:37:36 I: RX 27 CH23, -75dBm | 95 #3 #3 03 00 1e 00 04 8d 67 00 04 81 92 00 00 00 00 09 1d 65
07:37:36 I: RX 27 CH23, -75dBm | 95 #3 #3 84 13 84 00 74 00 d4 00 05 01 e2 00 a2 00 01 c6 51 f4
07:37:36 W: (#3) Frame 1 missing: Request Retransmit
07:37:36 I: TX 11 CH23 | 15 #3 86 09 76 68 81 c2
07:37:36 I: RX 27 CH61, -75dBm | 95 #3 #3 01 00 01 01 1b 00 0b 00 0b 00 1e 00 1f 00 04 83 a4 ad
07:37:36 W: (#3) Frame 2 missing: Request Retransmit
07:37:36 I: TX 11 CH40 | 15 #3 86 09 76 68 82 c1
07:37:37 W: (#3) Frame 2 missing: Request Retransmit
07:37:37 I: TX 11 CH61 | 15 #3 86 09 76 68 82 c1
07:37:37 W: (#3) Frame 2 missing: Request Retransmit
07:37:37 I: TX 11 CH75 | 15 #3 86 09 76 68 82 c1
07:37:38 I: (#4) enqueued cmd failed/timeout
07:37:38 I: (#4) Requesting Inv SN #4
07:37:38 I: (#4) enqueCommand: 0x0b
07:37:38 I: (#4) prepareDevInformCmd 0x0b
07:37:38 I: TX 27 CH3 | 15 #4 86 09 76 68 80 0b 00 65 0b d6 a2 00 00 00 00 00 00 00 00 b3 0d cd
07:37:38 I: (#5) Requesting Inv SN #5
07:37:38 I: (#5) enqueCommand: 0x0b
07:37:38 I: (#5) prepareDevInformCmd 0x0b
07:37:38 I: TX 27 CH23 | 15 #5 86 09 76 68 80 0b 00 65 0b d6 a2 00 00 00 00 00 00 00 00 b3 0d 31
07:37:38 I: (#6) no complete Payload received! (retransmits: 5)
07:37:38 I: (#6) Requesting Inv SN #6 (MI-600)
07:37:38 I: (#6) enqueCommand: 0x09
07:37:38 I: (#6) enqueCommand: 0x05
07:37:38 I: (#6) prepareDevInformCmd 0x09
07:37:38 I: TX 11 CH40 | 09 #6 (MI-600) 86 09 76 68 05 fe
07:37:38 W: (#4) nothing received: complete retransmit
07:37:38 I: (#4) prepareDevInformCmd 0xb
07:37:38 I: TX 27 CH61 | 15 #4 86 09 76 68 80 0b 00 65 0b d6 a2 00 00 00 00 00 00 00 00 b3 0d cd
07:37:38 W: (#5) nothing received: complete retransmit
07:37:38 I: (#5) prepareDevInformCmd 0xb
07:37:38 I: TX 27 CH75 | 15 #5 86 09 76 68 80 0b 00 65 0b d6 a2 00 00 00 00 00 00 00 00 b3 0d 31
07:37:38 I: (#6) retransmit on failed first request
07:37:38 I: TX 11 CH3 | 09 #6 (MI-600) 86 09 76 68 09 f2
07:37:38 I: RX 27 CH23, -75dBm | 95 #4 #4 01 00 01 01 1c 00 0b 00 0b 00 1f 00 1f 00 05 10 42 df
07:37:38 I: RX 27 CH3, -75dBm | 95 #4 #4 02 00 04 fb ec 00 00 00 00 01 1c 00 0b 00 0b 00 21 b8
07:37:38 W: (#4) Frame 3 missing: Request Retransmit
07:37:38 I: TX 11 CH23 | 15 #4 86 09 76 68 83 61
07:37:38 W: (#5) nothing received: complete retransmit
07:37:38 I: (#5) prepareDevInformCmd 0xb
07:37:38 I: TX 27 CH40 | 15 #5 86 09 76 68 80 0b 00 65 0b d6 a2 00 00 00 00 00 00 00 00 b3 0d 31
07:37:38 I: (#6) nothing received
07:37:39 I: RX 27 CH61, -75dBm | 95 #4 #4 03 00 20 00 05 06 84 00 05 25 93 00 00 00 00 09 23 a8
07:37:39 W: (#4) Frame 4 missing: Request Retransmit
07:37:39 I: TX 11 CH61 | 15 #4 86 09 76 68 84 66
07:37:39 W: (#5) nothing received: complete retransmit
07:37:39 I: (#5) prepareDevInformCmd 0xb
07:37:39 I: TX 27 CH75 | 15 #5 86 09 76 68 80 0b 00 65 0b d6 a2 00 00 00 00 00 00 00 00 b3 0d 31
07:37:39 W: (#4) Frame 4 missing: Request Retransmit
07:37:39 I: TX 11 CH3 | 15 #4 86 09 76 68 84 66
07:37:39 W: (#5) nothing received: complete retransmit
07:37:39 I: (#5) prepareDevInformCmd 0xb
07:37:39 I: TX 27 CH23 | 15 #5 86 09 76 68 80 0b 00 65 0b d6 a2 00 00 00 00 00 00 00 00 b3 0d 31
07:37:39 I: RX 27 CH40, -75dBm | 95 #4 #4 84 13 84 00 78 00 d5 00 05 01 ea 00 b9 00 01 71 a3 af
07:37:39 I: (#4) procPyld: cmd:  0x0b
07:37:39 I: (#4) Payload (62): 00 01 01 1c 00 0b 00 0b 00 1f 00 1f 00 05 10 42 00 04 fb ec 00 00 00 00 01 1c 00 0b 00 0b 00 21 00 20 00 05 06 84 00 05 25 93 00 00 00 00 09 23 13 84 00 78 00 d5 00 05 01 ea 00 b9 00 01
07:37:39 W: (#5) nothing received: complete retransmit
```

### Error description

Anknüpfend an https://github.com/lumapu/ahoy/issues/718#issuecomment-1727094341 ist es vermutlich besser, das separat zu diskutieren (@beegee3 @lumapu @knickohr).

> > Allerdings sehe ich nicht, wieso sich die Inverter in die Quere kommen können. Das Zeitmanagement beim Abhorchen der Kanäle ist m.E. unverändert, so dass Antworten des vorherigen Inverters bei Abfrage des nächsten nicht mehr kommen dürften. 🤔 sehr merkwürdig.
> 
> Hmmm, wenn der Inverter noch sendet, kann Ahoy das auch noch empfangen und gibt es an den passenden iv zur Auswertung weiter. Wenn dann da die nächste Abfrage gemacht wird, weil noch nicht alles da ist, überlappen sich die Anfragen uU. gegenseitig (mein Invervall ist mit 2 sec. aber unverändert kurz!).
> 
> Diesen Effekt hatte ich teilweise aber auch schon zu Zeiten gesehen, in denen das fastNext noch nicht drin war. Werde mir das aber nochmal ansehen, weil ich einige komische Effekte sehe, die irgendwie auch mit der fastNext-Geschichte zusammenhängen könnten. Andererseits ist es "nur" eine Ausweitung des "retransmit"-Requests, und damit müßte app::tickSend eigentlich klar kommen. Aber vermutlich übersehe ich noch irgendwas.
> 

Wie man am anliegenden log-Auszug sehen kann, kommt das gesamte timing in bestimmten Konstellationen bei mäßigem Empfang durcheinander. Der einfache nächste Versuch (in https://github.com/lumapu/ahoy/pull/1170 eingeführt) scheint in meiner (extremen!) Testkonstellation das Fass vollends zum Überlaufen zu bringen, und sicher würde es schon helfen, entweder das Invervall zu erhöhen, die Zahl der retransmit-Versuche zu verringern, oder wenigstens das "Regelungs-Störfeuer" zu verringern :grin:  (aber es geht ja ums Testen und Optimieren und nicht um workarounds).

Wie an anderer Stelle schon erwähnt: ähnliches (aber (abgesehen von meinen Heuristik-Versionen) nicht so extrem) meine ich immer mal wieder gesehen zu haben, auch schon völlig unabhängig von der aktuellen "Doppelung".  

Bin mal auf eure Ideen gespannt und werde vermutlich als nächstes erst mal das rx-Timeout ansehen.

knickohr, 2023-09-21:
Blöde Frage, woran erkenne ich es ?

rejoe2, 2023-09-21:
> Blöde Frage, woran erkenne ich es ?
> 
> Habe 5s Intervall und 10 Retries pro payload. Vielleicht reichts da noch aus 😉

Bei 5s dürfte in deinem setup (lauter 1ch-Inverter) keine Gefahr bestehen. Braucht nur 2 frames, wenn ich's richtig im Kopf habe (4ch: 4, MI-2ch auch 4). Macht max. 12 tx, bis ein einzelner Inverter durch ist... (erster erfolgloser retry, dann Erfolg + 10 (oder nur noch 9?) retries).

Man sieht es daran, dass irgendwann der nächste oder schon übernächste Inverter angefunkt wird, und dann wieder irgendein retransmit von einem "alten" Inverter kommt (wie eben exemplarisch in dem seriellen Auszug, der oben anhängt).

knickohr, 2023-09-21:
Hmmm, dann hilft nur der gesunde Menschenverstand und etwas Adam Ries ?

Ich glaube @oberfritze hat schon mal irgendwo im Discord eine Formel eingeschmissen. Sollte halt jeden klar sein, wie weit er mit dem Intervall runter kann und wieviele Retries man einstellt, damit noch alles im Rahmen paßt.

Ne einfache Faustformel wäre gut.

rejoe2, 2023-09-21:
Jein. Natürlich hat alles eine Grenze, aber hier soll es eben nicht um workarounds gehen, sondern darum, wie man 
- das interne timing optimiert, um unnötige (!) interne Wartezeiten zu minimieren (dann paßt mehr in das zur Verfügung stehende Zeitfenster), und eventuell
- wie man User"fehler" ggf. kompensieren kann, damit die gar nicht erst rechnen müssen. Sonst geht es erfahrungsgemäß sogar in den Fällen gelegentlich schief, in denen der User das Problem überhaupt als solches erkannt hat...

oberfritze, 2023-09-24:
Bezüglich Optimierung des Antwortintervalls: probiere doch mal, für Deine 4-kanaligen HM/MI (3rd Gen) den Teil der Optimierung für den HM-300 aus dem all-in-one fork entsprechend anzupassen, also z.B. so:

hmRadio.h:
        void prepareReceive (inv_type_t invType, uint8_t txChan, uint8_t rxFrameCnt) {
            ...
            } else if (invType == INV_TYPE_HMCH4) {
                mRxChannels = (uint8_t *)(rf24RxDefChan[txChan]);  // no change
                mMaxRxChannels = RX_DEF_MAX_CHANNELS;  // no change
                mRxAnswerTmo = rxFrameCnt * (RX_WAIT_SFR_TMO + RX_WAIT_SAFETY_MRGN); // current formula for hm inverters
                if (mRxAnswerTmo > RX_ANSWER_TMO) {
                    mRxAnswerTmo = RX_ANSWER_TMO;
                }
                mRxChanTmo = RX_CHAN_TMO;                    // no change
           } else ...
und

        void prepareDevInformCmd(uint64_t invId, inv_type_t invType, uint8_t txChan, uint8_t cmd, uint32_t ts, uint16_t alarmMesId, bool isRetransmit, uint8_t reqfld=TX_REQ_INFO) { // might not be necessary to add additional arg.
            ...
            prepareReceive (invType, txChan,
                // could be optimized for other HM inverter types as well
                ((cmd == RealTimeRunData_Debug) && (invType == INV_TYPE_HMCH1)) ? 2 :
                ((((cmd == RealTimeRunData_Debug) && (invType == INV_TYPE_HMCH4)) ? 4 :
                 (((invType == INV_TYPE_HMCH1) || (invType == INV_TYPE_HMCH4)) && ((cmd == InverterDevInform_All) || (cmd == SystemConfigPara) || (GetLossRate))) ? 1 : 10));

Hoffe, ich hab mich mit den Klammern nicht vertippt. Das sollte die Situation schon mal entspannen.
Darüber hinaus kannst Du natürlich auch noch auf die Tx-spezifischen 3 (statt 5) Antwortfrequenzen lauschen, auf denen Deine 4-Kanal Inverter wirklich antworten. Je mehr Fehler man vermeidet, desto seltener reizt man die Antworttimeouts aus.

rejoe2, 2023-09-29:
Seems to be solved (in latest dev) by ticker resets

closed 2023-09-29 by rejoe2