lumapu/ahoy #1175: [Bug] Update von 0.7.55auf 0.7.59 klappt nicht

MartinBachmannHD, 2023-09-21:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.55

### Github Hash

3a29d6c

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Upload der Firmware von 0.7.55 zu 0.7.59

### Error description

1) Ich verwende aktuell die 0.7.55 Version, funktioniert sehr gut und stabil mit HM-600 (auch ohne Elko) auf einem ESP32-D0WD-V3.
2) Wie immer wollte ich die aktuelle Version updated (0.7.59). Habe das Zip heruntergeladen, ausgepackt und über das Web-UI versucht, die Firmware (firmware -> ESP32 -> 230920_ahoy_0.7.59_3a29d6c_esp32.bin) zu installieren
3) Update resultiert in 'success'. Danach Reboot
4) Jetzt wird leider immer noch 0.7.55 angezeigt, nicht 0.7.59
5) Ich habe es von zwei Geräten probiert (Handy und Laptop, beide male Chrome Browser). 
6) Reboot aus WebUi -> System hat auch nichts geändert (nach dem Reboot ist die 'Uptime' aber nicht auf 0 Days, sondern läuft unbeeindruckt weiter)

MartinBachmannHD, 2023-09-21:
Es gab glaube ich mal eine Diskussion über zu wenig Speicher bei update - System sagt hier:

heap free:  150924
sketch used: 1163
heap total: 288228

Ollipop030, 2023-09-21:
Mach die DTU mal stromlos für ein paar Sekunden. Konnte meinen ESP heute auch erst nicht updaten (von 0.7.57 auf 0.7.59), nach dem Neustart ging es dann aber.

rejoe2, 2023-09-21:
Wenn "success" angezeigt wird, sollte es eigentlich geklappt haben. UU. sind da auch noch Reste im Browser gespeichert, die man mit einem full refresh der Seite weg bekommt.

Ollipop030, 2023-09-21:
> Wenn "success" angezeigt wird, sollte es eigentlich geklappt haben. UU. ist es "nur" ein Anzeigeproblem und es sind noch Reste im Browser gespeichert, die man mit einem full refresh der Seite weg bekommt.

Das war auch erst mein Gedanke, allerdings hat das Update heute morgen erst nach dem kompletten Stromlosmachen geklappt. Hab es vorher 4x versucht. Das "success" wurde immer angezeigt, obwohl es nicht geklappt hatte.

MartinBachmannHD, 2023-09-21:
Genau, ich hatte auch mit Shift+F5 im Browser probiert, es war kein Caching Problem im Browser. Und ja, das Stromlos-machen hat funktioniert, danach hatte ich direkt die .59 Version :-). Und mit der 0.7.59 Version klappt auch der 'Reboot' knopf wieder... 
Danke für die schnelle Hilfe!!!

closed 2023-09-21 by MartinBachmannHD