lumapu/ahoy #1177: [Bug]

teletapi, 2023-09-22:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.59

### Github Hash

GIT SHA: 3a29d6c :: 0.7.59

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

.

### Debug Serial Log output

```shell
keine Fehler
```

### Error description

YieldDay werte werden nach Sonnenuntergang wenn der WR ausgeht in der DTU genullt in den Ordnern channel 0, channel 1 und channel 2. die im Iobroker per Mqtt abgerufen werden.
Der Total Ordner hier werden  die gesamtwerte angezeigt, hier bleiben die Werte erhalten

[image]
[image]

teletapi, 2023-09-22:
[image]
[image]

knickohr, 2023-09-23:
Bin ich wahrscheinlich dran Schuld mit Issue #1151 ?  Sollte aber nicht Nullen, sondern nur das Logo einblenden 🤔

MetaChuh, 2023-09-23:
@knickohr 
mit 13 invertern bist du prinzipiell schuld an allem … du brauchst definitiv nen 14ten, der nicht an einem freitag gekauft wird 😂

MetaChuh, 2023-09-23:
@teletapi 
sorry für wieder ne extreme off topic meldung, hab's mir nicht verkneifen können 🙈🙏

teletapi, 2023-09-24:
Jetzt wirds seltsam, ich komme gerade nachg hause und schau dabei da sind alle YieldDay werte da.  2teage hintereinander nicht und heute sind sie da. Ich mach das Issue mal vorerst zu um keine weitere Verwiirung zu stifften. Vielleicht ist es ja auch ein eigenes problem .... Sollte es weiter hin nicht funktionieren schreib ich das  wieder.

closed 2023-09-24 by teletapi

knickohr, 2023-09-24:
Vielleicht sollten wir den Issue als Kuriositätensammlung warm halten.

Hatte gestern einen seltsamen Effekt. Nach fast 2 Tagen Uptime fing die DTU plötzlich an doppelt so schnell zu werden. Die Werte von den Invertern kamen nicht mehr, wie eingestellt, alle 5s, sondern jetzt alle 2,5s 😲 Keine Ahnung was da passiert ist. Nun kann man sich streiten ob es aus Ahoy kommt oder ob es durch das neue Radioprotokoll und die Heuristik rein kommt.  Wollte noch einen seriellen Output tragen, aber beim Aufruf der Konsole verlor die DTU alle Inverter. Weg, nix mehr da, also kein Output.

Nach Reboot und Import war alles wieder als ob nichts gewesen wäre 🤔

Gubi2023, 2023-09-24:
war bei mir auch so, alle 2,5 sec. Werteupdate, obwohl ich 15 sec. eingestellt habe....

Gubi2023, 2023-09-24:
DTU startet sich urplötzlich neu, Timestamp 2 Std falsch!
[image]

knickohr, 2023-09-24:
Ich habe leider nicht auf die Auer geachtet. Aber mit einen scheinbaren Neustart kann ich auch dienen. Ich habe am Vorabend die IP in der Fritze geändert. Das wird aber erst nach Neustart des Gerätes übernommen. Hat mich nur gewundert, das am nächsten Morgen die DTU mit der neuen IP ansprechbar, obwohl ich eigentlich keinen Neustart gemacht habe. Wir müssen das weiter beobachten, zumal es ja kein Einzelfall mehr zu sein scheint 😲