lumapu/ahoy #1178: DTU startet sich urplötzlich neu, Timestamp 2 Std falsch!

Gubi2023, 2023-09-24:
DTU startet sich urplötzlich neu, Timestamp 2 Std falsch!
[image]

_Originally posted by @Gubi2023 in https://github.com/lumapu/ahoy/issues/1177#issuecomment-1732510351_

knickohr, 2023-09-24:
Welche Version hast Du ?  Ich hatte gestern einen ähnlichen Effekt, auch wenn ich das mit der Uhr nicht mehr kontrollieren kann. Aber bei mir war das Intervall auch auf 2,5s verstellt, obwohl ich 5s eingestellt hatte.

Bei mir war es die 0.7.59, genauer gesagt eine modifizierte Version (0.7.591) dessen mit einer erweiterten Funktion und anderem Radioprotokoll des NRF. Das ist auch der Grund warum ich noch keinen Issue aufgemacht habe, weil es eben keine offizielle Version ist. Scheint aber so, als ob der Bug im offiziellen Teil von Ahoy ist. Müssen wir weiter beobachten.

Gubi2023, 2023-09-24:
Version 0.7.59

knickohr, 2023-09-24:
Sag mal, wie konntest Du den Reboot im Log eigentlich so genau erwischen ?  Zufall ?  Denn eigentlich müßte nach/ einem Reboot das Log doch leer sein ? 🤔

Hast Du was an der IP verändert, oder hat die DTU eine neue IP bekommen ?

Gubi2023, 2023-09-24:
das Log läuft in separatem Fenster, wird bei mir nicht gelöscht, auch nicht, wenn die DTU-Zeit neu startet...

knickohr, 2023-09-24:
OK, habe ich noch nicht probiert. Dann kann ich auch davon ausgehen das sich die IP nicht geändert hat.

Gubi2023, 2023-09-26:
Zusatz: DTU sendet/empfängt sporadisch 10 min. lang keine Signale vom WR
[image]

lumapu, 2023-09-27:
siehe https://github.com/lumapu/ahoy/issues/1179#issuecomment-1735071508
wird sehr wahrscheinlich in der nächsten Version gefixt sein.

closed 2023-12-29 by lumapu