lumapu/ahoy #1179: [Bug] double WiFi ticker

You69Man, 2023-09-24:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.51 ff

### Github Hash

77ba2783

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

-

### Error description

Mir ist aufgefallen, dass seit V0.7.51 (vermutlich im Zuge der Änderung der mainloops) der tickWifiLoop zweimal pro Sekunde aufgerufen wird:
Ticker 0 wifiL
Ticker 1 webSc
Ticker 2 disp
Ticker 9 wifiL
Ticker 0 wifiL
Ticker 1 webSc
Ticker 2 disp
Ticker 9 wifiL
Vorher war es nur 1x. Das bewirkt u.a. dass ein Zähler mCnt doppelt so oft inkrementiert wird, und die Funktion isTimeout() doppelt so schnell anschlägt, was auch immer das wiederum zur Folge hat...
Vielleicht verstehe ich es ja einfach auch nicht, aber ich habe das Gefühl, dass das nicht beabsichtigt war. 
Und vielleicht gibt' auch einen Zusammenhang mit #1178 oder #1173 ??

You69Man, 2023-09-24:
Der Ticker wird einmal in app::setup() und später noch einmal in app::onNetwork() gestartet

beegee3, 2023-09-26:
mit den loop Änderungen in v0.7.51 ist in `app::onNetwork` der Ticker Neustart (reset und reinstall) abhanden gekommen.
@lumapu war das Absicht? Nicht nur `wifiL` läuft doppelt, auch `tSend` und `tsCmd` werden mit jedem `onNetwork` Aufruf vervielfacht. Also, wenn Wifi mal abreißt und wieder neu verbindet.
Zur Abhilfe bitte hinter `mNetworkConnected = gotIp;` wieder einfügen:
```
    ah::Scheduler::resetTicker();
    regularTickers();  // reinstall regular tickers
```
Und bei der Gelegenheit in `app.h` die überflüssige Zeile 68 `void loopStandard(void);` löschen.

You69Man, 2023-09-26:
Habe gerade noch ein paar Versuche gemacht. Tatsächlich häufen sich bei WiFi Verbindungsabbruch und -wiederaufbau neben `wifiL `auch noch andere Ticker an, z.B. auch `mqttS `und `tSend`... was daher auch Ursache für andere bereits gemeldete Probleme sein dürfte.
Der Vorschlag von beegee3 oben scheint das Problem zu beheben!

Evt. sollte man zusätzlich auch noch im Scheduler einen Schutzmechanismus einbauen, dass beim Versuch einen Ticker zu starten, dessen Name bereits vergeben ist, diesen entweder ablehnt, oder den aktuell bestehenden zuerst löscht. Weiß aber nicht, das im Sinne der Erfinderst des Schedulers ist, oder ob man bewusst mehrere Ticker mit gleichem Namen zulassen möchte.

lumapu, 2023-09-27:
danke für die Analyse der Probleme! Ich war der Meinung, die Ticker laufen durchgehend und müssen daher nicht mehr gelöscht und neu aufgesetzt werden. Werden den PR von @rejoe2 entsprechend mergen.
Der Scheduler führt die Namen der Ticker nur zu Debugzwecken daher lag das Augenmerk nie darauf, dass ein Ticker doppelt vorhanden sein darf oder nicht. Im Prinzip dürfte jeder Ticker nur einmal exisitieren, ich weiß aber nicht, ob wir wirklich die Logik in jeder Installation mitschleppen wollen, die prüft ob Ticker `unique` sind.

beegee3, 2023-09-27:
@lumapu: auch wenn du eben die neue Version erstellt hast, falls es dir besser gefällt, kannst du alternativ die beiden eingefügten Zeilen in `app.cpp` wieder löschen und in `scheduler.h` in `addTicker` auf das Vorhandensein eines Tickers nach Name prüfen und ggfs. diesen aktualisieren. Dazu die Funktion folgendermaßen ändern:
```
            inline uint8_t addTicker(scdCb c, uint32_t timeout, uint32_t reload, bool isTimestamp, const char *name) {
                uint8_t idx = MAX_NUM_TICKER;
                for (uint8_t i = 0; i < MAX_NUM_TICKER; i++) {
                    if (!mTickerInUse[i] && (idx == MAX_NUM_TICKER)) {
                        idx = i;
                    }
                    if ((strncmp(name, mTicker[i].name, 5) == 0)) {
                        idx = i;
                        break;  // ticker already in mTicker list, just update 
                    }
                }
                if (idx < MAX_NUM_TICKER) {
                    mTickerInUse[idx] = true;
                    mTicker[idx].c = c;
                    mTicker[idx].timeout = timeout;
                    mTicker[idx].reload = reload;
                    mTicker[idx].isTimestamp = isTimestamp;
                    memset(mTicker[idx].name, 0, 6);
                    strncpy(mTicker[idx].name, name, (strlen(name) < 6) ? strlen(name) : 5);
                    if(mMax == idx)
                        mMax = idx + 1;
                    return idx;
                }
                return 0xff;
            }
```
Wichtig ist, dass die `for()` Schleife weiterläuft, auch wenn man einen unbenutzten Ticker gefunden hat. Sonst findet man einen ggfs. vorhandenen Ticker nicht, falls der einen größeren Index hat.

lumapu, 2023-09-29:
da finde ich doch die beiden vorherigen Zeilen eleganter, zumal wir nie sichergehen können nicht versehentlich zwei Ticker mit idemtischem Namen anzulegen. Danke für die Variante 2.

You69Man, 2023-09-30:
Ich fände zwar den Vorschlag von beegee3 auch nicht schlecht, aber ich denke, es funktioniert auch so, wie es derzeit in der 0.7.60 gelöst wurde -> ich close das damit einmal

closed 2023-09-30 by You69Man