lumapu/ahoy #1183: zusätzlicher Sensor an AhoyDTU (ESP32 oder 8266)

grericht, 2023-09-26:
ich wollte fragen ob es irgendwie möglich ist an die DTU zusätzlich einen Sensor anzuschließen (es geht um einen "Abstandsmesser" SR04 um den Füllstand der Zisterne zu messen). und die Daten dann mit an Home Assistant zu schicken. Ich habe aktuell quasi 2 ESP8266 nebeneinander liegen. Der eine mit AhoyDTU der andere mit Tasmota und SR04. Lohnt es daran zu arbeiten nur einen ESP8266 oder 32 daraus zu machen?

MartinBachmannHD, 2023-09-26:
Bei mir liegen auch zwei nebeneinander... (https://sensor.community/de/sensors/airrohr/). Anscheinend ist aber vor allem beim ESP8266 nur noch wenig Platz. Bin sicher da kommen viele mögliche Kandidaten zusammen... 
Ansonsten könnte ich noch 'Werbung' für die Idee machen, den NRF-Sensor per USB an einen NAS/RPi/... anschließbar zu machen... : https://github.com/lumapu/ahoy/issues/1008

grericht, 2023-09-26:
ein raspi verbraucht dann aber doch schon ganz ordentlich strom. und den will ich der witterung auch nicht unbedingt aussetzen. am esp32 und 8266 ist ja genug platz um den nrf sensor und einen schall-entfernungsmesser parallel zu betreiben. ich will nur wissen ob das überhaupt vorstellbar ist. schön wäre eine tasmota version der dtu. dann könnte man wohl flexibel sensoren kombinieren.

MartinBachmannHD, 2023-09-26:
...ja, RPi nur, wenn er wegen Mosquitto, OpenHAB, ... eh' schon da ist... und ja, nur drinnen :-)

You69Man, 2023-09-26:
Der ESP8266 hat noch den A0 Analogeingang frei. Der ist allerdings etwas mit Vorsicht zu genießen, und kann nur für langsame Analog- oder Digitalmessungen verwendet werden. Für den SR04 scheint er mir nicht geeignet zu sein, denn hierfür bräuchte man eine schnelle und hochgenaue Zeitmessung. Der A0 ist aber weder interruptfähig, noch mag es der ESP8266 , wenn der A0 schnell gepollt wird, denn dann fällt das Wlan aus (der Analogeingang wird vom ESP parallel im Hintergrund auch vom WiFi controller genutzt).

lumapu, 2023-09-27:
am besten einfach einen Fork meines Repositories anlegen und dort die Änderungen einbauen, die du benötigst. So kannst du ganz einfach deinen Code neben dem bestehenden und zukünftigen Ahoy-Code haben.

knickohr, 2023-09-27:
Was hat ein Füllstandsmesser mit einen Photovoltaik DTU zu tun ?

Nee, bitte nicht, der nächste möchte dann eine Wetterstation dran hängen oder andere Sensoren.

grericht, 2023-09-27:
> Was hat ein Füllstandsmesser mit einer Photovoltaik DTU zu tun ?
> 
> Nee, bitte nicht, der nächste möchte dann eine Wetterstation dran hängen oder andere Sensoren.

beides befindet sich bei mir im garten in einer entfernung, dass es noch ins heimische wlan kommt und die zisterne ist nahe genug an den wechselrichtern um auch dort hin zu funken. und beides kann auf einem chip zusammen verarbeitet werden. 
was muss es mehr miteinander zu tun haben?

grericht, 2023-09-27:
> am besten einfach einen Fork meines Repositories anlegen und dort die Änderungen einbauen, die du benötigst. So kannst du ganz einfach deinen Code neben dem bestehenden und zukünftigen Ahoy-Code haben.

das kann ich aktuell nicht verstehen aber werde mal versuchen damit klar zu kommen. grundlegend heißt es, dass es eine möglichkeit gibt?

grericht, 2023-09-28:
dann frage ich nochmal kurz andersrum: ist der code von ahoydtu irgendwie mit tasmota kompatibel? bei tasmota will ich mich evtl eh mehr einarbeiten und habe schon selber compiliert. wenn es möglich wäre den code von ahoy zu nehmen und "auf die betreffenden pins eines mit tasmota bespielten esp32 oder 8266 zu spielen, könnte ich mir das evtl mal anschauen". wenn die sprache aber eine komplett andere ist, dann würde das ja nicht funktionieren. ich will nur eine einschätzung ob es sorum möglich wäre.

cinatic, 2023-10-24:
hatte endlich mal zeit einen bme280 sensor dranzubauen, kann gut sein dass ich etwas außer acht gelassen habe (bin cpp noob) aber es funktioniert für meine zwecke: https://github.com/lumapu/ahoy/compare/main...cinatic:ahoy:add_bme280

es wäre nicht schlecht wenn es den part für das auslesen der werte seperat gäbe

knickohr, 2023-10-24:
Kannst es ja als Pull Request in Ahoy einfließen lassen. Da es ein I3C Interface ist, wird sich das auch recht einfach und ohne Probleme einbauen lassen.

grericht, 2023-10-25:
Das klingt super! Da ich die AhoyDTU perspektivisch draußen haben werde, ist schon ein Temperatursensor einfach eine sinnvole Ergänzung. Persönlich wäre mir aber noch wichtiger einen SR04 als Abstandsmesser anzuschließen. Ist das komplizierter?

cinatic, 2023-10-25:
@knickohr 
naja will ja nicht jeder den bme drin haben. Wäre nicht schlecht, wenn man modular irgendwie was setzen kann wie, --enable sensor-xyz und man müsste dann nur sorgen dass die verkabelung passt, aber dafür kenn ich mich zuwenig im projekt aus

@grericht 
eigentlich nicht, "einfach" die sachen hier https://github.com/lumapu/ahoy/compare/main...cinatic:ahoy:add_bme280#diff-c395850a24ba1bf6f63a172c7c4cc33425a7dde3cfb9fc48b72fb75a7c6bdaf8 entsprechend ersetzen, also den sensor und das senden via mqtt

closed 2024-02-29 by lumapu