lumapu/ahoy #1184: AhoyDTU "verliert" immer wieder Daten von Invertern - webSerial, buffer overflow!

contentpirate, 2023-09-26:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

ba218ed

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Siehe Screenshots (Eigentlich ausgehend von Basiseinstellungen nichts verändert)

### Debug Serial Log output

```shell
webSerial, buffer overflow!
webSerial, buffer overflow!
webSerial, buffer overflow!
webSerial, buffer overflow!
webSerial, buffer overflow!
00 05 b5 45 01 03 01 07 09 20 13 8b 0a 26 00 02 00 6f 03 e8 01 b8 00 01 
/ch2/Irradiation: 4.795 %
11:36:45 I: WRNord4/ch2/MaxPower: 20.900 W
11:36:45 I: WRNord4/ch0/U_AC: 232.800 V
11:36:45 I: WRNord4/ch0/I_AC: 0.180 A
11:36:45 I: WRNord4/ch0/P_AC: 41.500 W
11:36:45 I: WRNord4/ch0/F_AC: 50.030 Hz
11:36:45 I: WRNord4/ch0/PF_AC: 1.000 
11:36:45 I: WRNord4/ch0/Temp: 22.900 °C
11:36:45 I: WRNord4/ch0/ALARM_MES_ID: 1.000 
11:36:45 I: WRNord4/ch0/YieldDay: 118.000 Wh
11:36:45 I: WRNord4/ch0/YieldTotal: 316.408 kWh
11:36:45 I: WRNord4/ch0/P_DC: 43.500 W
11:36:45 I: WRNord4/ch0/Efficiency: 95.402 %
11:36:45 I: WRNord4/ch0/MaxPower: 44.000 W
11:36:45 I: 
11:36:45 I: Iv: 8
11:36:45 I: WROst1/ch1/YieldDay: 257.000 Wh
11:36:45 I: WROst1/ch1/YieldTotal: 362.136 kWh
11:36:45 I: WROst1/ch1/MaxPower: 163.300 W
11:36:45 I: WROst1/ch2/YieldDay: 261.000 Wh
11:36:45 I: WROst1/ch2/YieldTotal: 374.083 kWh
11:36:45 I: WROst1/ch2/MaxPower: 172.300 W
11:36:45 I: WROst1/ch0/ALARM_MES_ID: 1.000 
11:36:45 I: WROst1/ch0/YieldDay: 518.000 Wh
11:36:45 I: WROst1/ch0/YieldTotal: 736.219 kWh
[... 1153 lines cut ...]
11:38:55 I: RX 27B Ch40 | 95 83 08 60 50 83 08 60 50 01 00 01 01 46 00 4b 00 f4 01 41 00 40 00 cd 00 02 a2 
11:38:55 I: RX 27B Ch40 | 95 83 08 60 50 83 08 60 50 02 95 f0 00 02 3e 09 00 40 00 37 09 17 13 8b 01 ac 9b 
11:38:55 I: RX 23B Ch23 | 95 83 08 60 50 83 08 60 50 83 00 00 00 12 03 e8 00 e6 00 01 3b c4 f7 
11:38:55 I: procPyld: cmd:  0x0b
11:38:55 I: procPyld: txid: 0x95
11:38:55 I: Payload (42): 00 01 01 46 00 4b 00 f4 01 41 00 40 00 cd 00 02 95 f0 00 02 3e 09 00 40 00 37 09 17 13 8b 01 ac 00 00 00 12 03 e8 00 e6 00 01 
enhaus/ch1/I_DC: 4.730 A
11:38:55 I: WROst1/ch1/P_DC: 147.000 W
11:38:55 I: WROst1/ch1/YieldDay: 262.000 Wh
11:38:55 I: WROst1/ch1/YieldTotal: 362.141 kWh
11:38:55 I: WROst1/ch1/Irradiation: 44.545 %
11:38:55 I: WROst1/ch1/MaxPower: 163.300 W
```

### Error description

- Insgesamt 9 Inverter sind in Betrieb
- Vor dem Update auf 0.7.36 lief alles super
- Nach dem Update: Jetzt passiert es immer wieder, dass die Ahoy DTU die Verbindung zu den Invertern zu "verlieren" scheint - es werden dann (fast) alle Inverter als "available and not producing" angezeigt. Nach und nach werden dann die Inverter wieder als "available and is producing" angezeigt. Dann dauert es keine 30 Sek, bis der Spaß wieder von vorne losgeht.
- Im Log tauchen gar nicht alle Inverter auf

[image]
[image]
[image]
[image]

contentpirate, 2023-09-26:
[image]

rejoe2, 2023-09-26:
Timing-Probleme scheinen immer mal wieder aufzutreten, v.a., wenn man eher viele Inverter hat. "buffer overflow" kenne ich auch.... 

Würde mal empfehlen, auf eine aktuelle dev-Version zu gehen und (zusätzlich) ggf. mal das Ausdrucken der MQTT-Infos über webserial auszuschalten.  
Muss zwar nicht "der Durchbruch" sein (siehe #1173), aber besser sollte die m.E. schon laufen.
Es gibt übrigens neben der "regulären" dev (.59) noch ein paar Varianten von mir in discord unter "all-in-one", von denen zwar mind. die letzte (mit aktivierter "loop") auch einen "Hau" hat, aber mehr Versuche bei an sich aktiven Invertern pro Durchlauf macht; das sollte bei vielen Invertern zumindest verhindern, dass die als "nicht produzierend" gemarkert werden.

contentpirate, 2023-09-27:
> Würde mal empfehlen, auf eine aktuelle dev-Version zu gehen und (zusätzlich) ggf. mal das Ausdrucken der MQTT-Infos über webserial auszuschalten.

Abschalten der MQTT-Infos über Webserial brachte leider keinen Stabilitätsgewinn. Gibt es denn eine (noch) aktuellere Version? Wenn ja, wo?

Ich bin jetzt mal den umgekehrten Weg gegangen und habe v0.7.26 aufgespielt (gleicher Effekt) und danach v0.6.9, bei der wieder alles fein ist. Verbindung ist wieder stabil, ohne irgendwelche Probleme. :-)

contentpirate, 2023-09-27:
Anmerkung: Ich bin mir nicht 100%ig sicher, aber vermutlich war der Update Sprung von der v0.6.0 auf v0.7.36, wodurch dann die o.g. Effekte auftraten

knickohr, 2023-09-27:
Die letzten Stable-Versionen hatten einige kleine Wehwehchen. Geh mal auf die letzte DEV und probiere.  Ich selbst habe 13 Inverter an einer DTU und mit der von @rejoe2 ereähnten Version, besonders die .591 habe ich kaum bis gar keine Aussetzer mehr.

lumapu, 2023-10-04:
gibt es das Problem mit den neuen Einstellungen immer noch?

knickohr, 2023-10-05:
Ich habe es nicht mehr beobachtet.

closed 2023-12-29 by lumapu