lumapu/ahoy #1185: replacement for deprecated REST API /api/record/live after 0.7.26?

jwagnerhki, 2023-09-26:
Release 0.7.26 was the last version offering live data and monitoring points under the REST API endpoint /api/record/live. 

Is there any replacement endpoint in later versions that would provide the same data?

In particular, where to find the "U_DC" field?

The "U_DC" field was extremely useful for battery-attached microinverters. It allowed to detect a battery under-voltage condition and turn the microinverter "off" via the REST API, and similarly, detect voltage recovery and turn the inverter back "on" via the REST API. 

This full control loop was possible via just the REST API.

After v0.7.26 it gets a lot more complicated - U_DC found only in unicast MQTT, while inverter on-off control is possibly only via the REST API.

MetaChuh, 2023-09-27:
@jwagnerhki 
all data you need is available at `/api/inverter/id/[0-x]`, separate for each inverter, to keep the amount of data for each request as small as possible.
note: /api/inverter/id/ endpoints also have been available at older versions, e.g. v0.6.9

/api/record/live was an additional debug dump, with the purpose of exporting as much data as possible at once, but appx at 6 panels, depending on the inverter and panel name lengths given, it cut off the data, due to lack of memory.
this has been discussed thoroughly in past issues, which led to the necessary removal of /api/record/live

content example for /api/inverter/id/0:

```
[inverter/id] => Array
	(
		[0] => Array
			(
				[id] => 0
				[enabled] => 1
				[name] => HM-800
				[serial] => 114184857072
				[version] => 10010
				[power_limit_read] => 100
				[ts_last_success] => 1695747366
				[ch] => Array
					(
						[0] => Array
							(
								[0] => 231.2
								[1] => 0
								[2] => 0
								[3] => 49.95
								[4] => 0
								[5] => 26.5
								[6] => 362.522
								[7] => 3882
								[8] => 0.8
								[9] => 0
								[10] => 0
							)

						[1] => Array
							(
								[0] => 22.9
								[1] => 0.02
								[2] => 0.4
								[3] => 1995
								[4] => 194.068
								[5] => 0.098
							)

						[2] => Array
							(
								[0] => 22.9
								[1] => 0.02
								[2] => 0.4
								[3] => 1887
								[4] => 168.454
								[5] => 0.098
							)

					)

				[ch_name] => Array
					(
						[0] => AC
						[1] => JKM410M S-LOW
						[2] => JKM410M S-HIGH
					)

				[ch_max_pwr] => Array
					(
						[0] => 
						[1] => 410
						[2] => 410
					)

			)
		)
	)
	
```

to get the numbers of available inverters and field names, you can use /api/live
data example:

```
[live] => Array
	(
		[generic] => Array
			(
				[wifi_rssi] => -80
				[ts_uptime] => 74616
				[menu_prot] => 1
				[menu_mask] => 60
				[menu_protEn] => 1
				[esp_type] => ESP8266
			)

		[refresh] => 10
		[ch0_fld_units] => Array
			(
				[0] => V
				[1] => A
				[2] => W
				[3] => Hz
				[4] => 
				[5] => °C
				[6] => kWh
				[7] => Wh
				[8] => W
				[9] => %
				[10] => var
			)

		[ch0_fld_names] => Array
			(
				[0] => U_AC
				[1] => I_AC
				[2] => P_AC
				[3] => F_AC
				[4] => PF_AC
				[5] => Temp
				[6] => YieldTotal
				[7] => YieldDay
				[8] => P_DC
				[9] => Efficiency
				[10] => Q_AC
			)

		[fld_units] => Array
			(
				[0] => V
				[1] => A
				[2] => W
				[3] => Wh
				[4] => kWh
				[5] => %
			)

		[fld_names] => Array
			(
				[0] => U_DC
				[1] => I_DC
				[2] => P_DC
				[3] => YieldDay
				[4] => YieldTotal
				[5] => Irradiation
			)

		[iv] => Array
			(
				[0] => 1
				[1] => 1
				[2] => 1
				[3] => 1
				[4] => 
				[5] => 
				[6] => 
				[7] => 
				[8] => 
				[9] => 
			)

	)
```

if you don't need to know the field names any more, as they usually don't change positions, you can also use /api/index/ to get all available inverters and most of the important values

```
[index] => Array
	(
		[generic] => Array
			(
				[wifi_rssi] => -80
				[ts_uptime] => 74616
				[menu_prot] => 1
				[menu_mask] => 60
				[menu_protEn] => 1
				[esp_type] => ESP8266
			)

		[ts_now] => 1695789702
		[ts_sunrise] => 1695790059
		[ts_sunset] => 1695833072
		[ts_offset] => 0
		[disNightComm] => 
		[inverter] => Array
			(
				[0] => Array
					(
						[enabled] => 1
						[id] => 0
						[name] => HM-800
						[version] => 10010
						[is_avail] => 
						[is_producing] => 
						[ts_last_success] => 1695747366
					)

				[1] => Array
					(
						[enabled] => 1
						[id] => 1
						[name] => HM-300
						[version] => 10014
						[is_avail] => 1
						[is_producing] => 1
						[ts_last_success] => 1695789656
					)

				[2] => Array
					(
						[enabled] => 1
						[id] => 2
						[name] => HM-600
						[version] => 10010
						[is_avail] => 
						[is_producing] => 
						[ts_last_success] => 1695747226
					)

				[3] => Array
					(
						[enabled] => 1
						[id] => 3
						[name] => HM-1500
						[version] => 10018
						[is_avail] => 1
						[is_producing] => 
						[ts_last_success] => 1695789676
					)

			)

		[warnings] => Array
			(
			)

		[infos] => Array
			(
			)

	)
```	

greetings,
metachuh

lumapu, 2023-09-27:
thanks @MetaChuh for explanation

jwagnerhki, 2023-09-28:
Thank you @MetaChuh for your excellent reply with details!

Goot to know about /api/inverter/id/<0-x> for the data fields, and /api/live for fld_name and fld_unit. Indeed I was able to query "U_DC" now from there.

A pity this is undocumented. It would be helpful to have this in the User_Manual.md.

1) Could you perhaps update the [User_Manual.md](https://github.com/lumapu/ahoy/blob/main/User_Manual.md)? There is a section [Control via REST API](https://github.com/lumapu/ahoy/blob/main/User_Manual.md#control-via-rest-api), maybe add a new section "Monitor via REST API"?

2) It would also be nice if the /api could self-document the availability of /api/inverter/id/0 endpoint(s). Under v0.7.26 at least there is only a mention of /api/inverter/list, but not the id endpoint(s), i.e.,
$ wget -q -O- http://ahoy.lan/api/
{"avail_endpoints":{"system":"http://ahoy.lan/api/system","statistics":"http://ahoy.lan/api/statistics","inverter/list":"http://ahoy.lan/api/inverter/list","index":"http://ahoy.lan/api/index","setup":"http://ahoy.lan/api/setup","live":"http://ahoy.lan/api/live","record/info":"http://ahoy.lan/api/record/info","record/alarm":"http://ahoy.lan/api/record/alarm","record/config":"http://ahoy.lan/api/record/config","record/live":"http://ahoy.lan/api/record/live"}}

MetaChuh, 2023-09-28:
@jwagnerhki 
please forgive us for not documenting this for new users.
every community member answers so many questions, within issue tickets, that we simply believe we already did document this everywhere, hence no community pull requests to address this 🙈🙏

we are all humans, trying to share everything what we've found out to the public, but it gets harder and harder to summarize everything, both for first contact users, as well as pros who are facing a major but necessary change.

MetaChuh, 2023-09-28:
2. please try the latest dev version <kbd>[download link to 0.7.60](https://github.com/lumapu/ahoy/commit/963e618c5c687de0252e0b9b9bcc4818773a2612)</kbd> as it contains the changes regarding your request to list at least `/api/inverter/id/0` within the `avail_endpoints`

this has been introduced a while ago, due to popular demand, but is not yet available at an release stage, due to pending issues or concerns not to publish a stable version yet.

thanks for your understanding, and please follow all open and closed issues, as they may contain valuable information for your needs.

greetings,
metachuh

closed 2024-02-17 by lumapu