lumapu/ahoy #1186: Frage: Esport settings (JSON) schöner formatiert

GeThi255, 2023-09-26:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.59

### Github Hash

3a29d6c

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Ahoy-DTU mit einem HM-1500 WR
ESP32
Keine außergewöhnlichen Settings

### Error description

Der JSON Export ist in einer Zeile ohne Formatierung.

`{"wifi":{"ssid":"FARBEN","pwd":"","ap_pwd":"esp_8266","hidd":false,"dev":"AHOY-DTU","adm":"xxxxxxx","prot_mask":60,"dark":false, ...`

Es ist sehr schwierig die unterschiedlichen Stände mit z.B. WinMerge oder NotePad++ zu vergleichen.

Schöner wären ein paar Zeilenumbrüche und Tabs an den richtigen Stellen:

Etwas so:

{
"wifi":{
	"ssid":"FARBEN",
	"pwd":"",
	"ap_pwd":"esp_8266",
	"hidd":false,
	"dev":"AHOY-DTU",
	"adm":"xxxxx",
	"prot_mask":60,
	"dark":false,
	"reb":true,
	"ip":"",
	"mask":"",
	"dns1":"",
	"dns2":"",
	"gtwy":""
},
"nrf":{
	"intvl":1,
	"maxRetry":5,
	"cs":5,
	"ce":4,
	"irq":15,
	"sclk":18,
	"mosi":23,
	"miso":19,
	"pwr":3,
	"en":true
},

Vielen Dank
Gerd

GeThi255, 2023-09-26:
Entschuldigung, sollte kein bug sein. Ist nur eine Frage bzw. ein Erweiterungswunsch.

LeSpocky, 2023-09-27:
Kann man mit beliebigen externen Tools für json nach Wunsch umformatieren. Unter Linux ist `jq` ziemlich genial für sowas.

lumapu, 2023-09-27:
ich bevorzuge auch @LeSpocky Ansatz. In Ahoy selbst würde ich ungern unnötig mit Zeichen jonglieren, der Speicher ist sehr begrenzt. Was du hier als Export bekommst ist im Prinzip fast das was der Ahoy im internen Speicher hat (bis auf die Passwörter).
Man kann das JSON auch zB. in Firefox öffnen, der hat einen eingebauten "parser" dafür. Für Sublime-Text verwende ich "JSON Reindent" als Plugin für diese Zwecke.

GeThi255, 2023-09-28:
Ja, mache ich bisher auch im Notepad++. Ist für mich OK.

closed 2024-02-29 by lumapu