lumapu/ahoy #1187: [Bug] multiple Alarms are shown in unsorted order

servicemensch, 2023-09-26:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

Ba218ed

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

-reboot ahoy at midnight 
- 2 inverter configured

### Error description

Alarms are Shows unsorted wenn multiple Alarms are Listen. Recognized on inverter 2.
Alarms schould be shown in the order of occurance, newest on top.

[image]

lumapu, 2023-09-27:
check with development version, there were changes on this part!
https://fw.ahoydtu.de

lumapu, 2023-09-30:
is there an update? Do you checked it again with the other version?

closed 2023-12-29 by lumapu