lumapu/ahoy #1189: Will Hoymiles HMS 800W-2T get Support?

T3mplate-wow, 2023-09-27:
Hey,
I am one of the „lucky“ guys who bought a HMS 800W-2T. This is the one with included wifi. 
I am wondering if this device is not compatible to a DTU anymore (because of the WIFI) or if this is just not tested. 
Or does anybody know if there is any other way to get the data out without the cloud?

lumapu, 2023-09-27:
for now this inverter is not supported. Only the inverters with NRF or CMT radio are supported. We did no reengineering on the wifi variants until now.

stefan123t, 2023-09-27:
@lumapu I would like to disagree, we just traced the first communication between the App and the DTUBI (built in DTU) of one of the WiFi models. It is basically everything already in the NetProtocol.c/.h files of the original Hoymiles DTU Pro Source code from gitee.
You can look up the details of the various NetProtocol packets received / sent by the DTU from / back to the APP/Cloud in the files InformationData.pb.c/.h and/or APPInformationDate.pb.c/.h

@T3mplate-wow but it will definitely take several months before we can provide the same support as with the other already supported communication for Gen2 (MI) and Gen3 (HM) NRF24L01+-based models as well as the Gen3 (HMS/HMT) CMT2300A-based RF models. The HMS-800**W**-2T models with the built in WiFi DTUBI are a completely different protocol for us.
We will have to implement at least a couple of those NetworkCommands which are listed in the above source code.

T3mplate-wow, 2023-09-27:
Thanks for the fast answer. 
Do you already know: 
Can this be solved completely in software with this model or will special hardware also be necessary?
Will this protocol be able to be queried more frequently or, like the cloud, only every 15 minutes?

LukeTheMast3r, 2023-10-02:
Hallo,
würde mich dieser Anforderung anschließen. Habe mir auch einen HMS800W-2T gekauft.  Hier wurde ja die HMS Serie aufgeführt. Da ich neu in der Materie bin, habe ich das mit neuem Protokoll usw. natürlich nicht gewusst.
Habe nach einiger Frustration endlich alles zum laufen gebracht (ahoy auf ESP32 & CMT2300A) um dann bei der Inverter Seriennummern-Eingabe festzustellen, dass diese nicht akzeptiert wird. Startet bei mir mit 141 ....

Hoffe Ihr habt bald mal Zeit das zu implementieren. :) Solange wird gewartet.!
In der Zwischenzeit wäre cool wenn Ihr einen Kommentar für die 2T Serie in der Kompatibilitätsliste einbauen könntet.
Aber dennoch: Respekt was Ihr bis jetzt alles geschafft habt. Sieht alles mega gut aus.!
Grüße

----
Hello,
would join this requirement. I also bought a HMS800W-2T.  Here, yes, the HMS series was listed. Being new to the matter, I of course didn't know about the new protocol etc.
After some frustration I finally got everything running (ahoy on ESP32 & CMT2300A) only to find out that the inverter serial number is not accepted. Starts with me with 141 ....

Hope you have time to implement this soon. :) We will wait until then!
In the meantime it would be cool if you could add a comment for the 2T series in the compatibility list.
But still: respect what you have done so far. Everything looks mega good!
Greetings

DennisOSRM, 2023-10-09:
> @lumapu I would like to disagree, we just traced the first communication between the App and the DTUBI (built in DTU) of one of the WiFi models. It is basically everything already in the [...]

@stefan123t Would you mind sharing the trace?

stefan123t, 2023-10-15:
@DennisOSRM das Trace file von @PaeserBastelstube findet sich hier im Discord #hms-wifi-serie Kanal
https://discord.com/channels/984173303147155506/1149770941283442698/1156496177882202182
Interessant sind auch die Infos die @jarekin und ich aus der S-Miles Installer App extrahieren konnten. Einfach mal die angepinnten Nachrichten im Kanal durchlesen.

henkwiedig, 2023-10-15:
Please have a look here https://github.com/henkwiedig/Hoymiles-DTU-Proto
These are working POCs for a HMS-800W-2T via the internal Inverter WLAN.

The protobuf message definitions are language agnostic.

DennisOSRM, 2023-10-16:
Danke für den Link zum Discord Channel und zum PoC. Das ist echt super hilfreich. Ich hatte am Wochenende angefangen einen Prototypen eines MQTT publishers für den HMS-800W-T2 (in Rust) zu coden. Bin aber irgendwie an den CRCs hängen geblieben. Wenn ich etwas funktionierendes habe, dann poste ich den Link hier.

stefan123t, 2023-10-17:
>. Bin aber irgendwie an den CRCs hängen geblieben. 

Dazu mal bitte ins Protokol Wiki schauen… sollten CRC8 und CRC16/ModBus sein. Die passenden Polynome sollten dort bzw im Code bereits dokumentiert sein.

https://github.com/lumapu/ahoy/wiki/Protocol#welche-pr%C3%BCfsummen-crc--cyclic-redundancy-check-gibt-es-und-%C3%BCber-welchen-teil-der-nachrichten-werden-sie-gebildet

DennisOSRM, 2023-10-19:
Ea gibt einige Fortschritte und Erkenntnisse zu berichten: 

- Ich habe einen funktionierenden Prototypen, der jede Sekunde die Last-, und Leistungswerte auslesen kann. 
- Das Response Protobuf für den HMS-800w-T2 scheint etwas speziell zu sein. Es passt zu keinem der bekannten proto files, wenn ich nichts übersehen habe. Ich habe das aber reverse engineered und kann ca. 2/3 der Felder direkt zu ordnen.  Alle wichtigen Informationen (aktuelle Leistung, Stromstärke, Netzfrequenz, Tempera, usw) sind für jeden Port dabei. Das müsste schon die meisten Anwendungsfälle abdecken. 
- Die Kodierung der Seriennummer des Inverters ist etwas unerwartet. Der Inverter liefert einen Int64, der als hex Darstellung der Seriennummer wie auf dem Aufkleber auf dem Gerät entspricht. Beispiel: Response liefert 4660, was hexadezimal 0x1234 entspricht und 1234 ist dann die Nummer auf dem Aufkleber. Für die Seriennummer der DTU wird das anders gemacht. Da wird die Nummer als Ascii-String kodiert. 
- Zeitstempel werden als 32 bit epoch behandelt. Damit gibt es eine gute Chance, dass in 15 Jahren beim Overflow die Firmware nicht mehr richtig mit Zeit umgehen kann. Das ist bei neuen Geräten durchaus noch während der zu erwartenden Lebensdauer.
- Es ist möglich den Inverter mit fehlerhaften Requests in einen Zustand zu bekommen, aus dem man nur mit Neustart des Geräts herauskommt.

T3mplate-wow, 2023-10-19:
Wow. Richtig cool. 
Ist schon klar wie eine Einbindung in die eigene Infrastruktur aussehen kann? Wenn ich es richtig verstehe muss man sich mit dem Access Point des WR verbinden. Brauchst man zwingend einen ESP dazwischen oder geht es auch mit WR direkt im eigenen WLAN?

DennisOSRM, 2023-10-19:
Es geht auch im eigenen WLAN.

fsoelch, 2023-10-29:
@DennisOSRM  super cool, vielen Dank!
kannst du schon absehen, wann du deinen Stand Committee kannst?

DennisOSRM, 2023-10-31:
Sobald ich etwas Zeit habe. 😉 Ich denke am Wochenende sollte ich soweit sein.

lumapu, 2023-10-31:
sehr cool was man hier so liest, bin sehr gespannt 😊

SOlangsam, 2023-11-01:
Verfolge es ebenfalls schon einige Zeit und freue mich, wenn es soweit ist!

DennisOSRM, 2023-11-02:
[image]

Hatte es auch schon an anderer Stelle gepostet, aber der Vollständigkeit halber noch mal hier: man sieht wie der Code übersetzt und das Tool gestartet wird. Im Hintergrund updatet die MQTT Instanz von meinem iobroker direkt nach dem Start. 

Das sollte 'ne Idee geben, wie das Tool funktioniert. Ich polier es noch ein wenig und lad es wohl morgen bei github hoch.

DennisOSRM, 2023-11-02:
Code ist hochgeladen: https://github.com/DennisOSRM/hms-mqtt-publisher

fsoelch, 2023-11-02:
Hallo Dennis,

super cool, vielen Dank!
Sobald die Sonne morgen den Wechselrichter wieder weckt werde ich mal testen ob es bei mir auch klappt :)

Ganz lieben Dank!

Florian

ringelbaer, 2023-11-03:
Funktioniert! 
Danke Dennis!

InScene, 2023-11-04:
Klappt auch bei mir. Saubere Arbeit 👍 
Vielen Dank @DennisOSRM

InScene, 2023-11-04:
Ich war so frei und habe das ganze mal in ein Docker Image gepackt.
Das macht es manchen vielleicht einfacher dieses zu nutzen.
Ihr findet es auf Docker Hub. Einfach beim starten die Variablen setzten:
https://hub.docker.com/r/29012019/hms-mqtt-publisher

DennisOSRM, 2023-11-04:
Sehr cool. Wie aktualisiert man die Version im Docker Image?

InScene, 2023-11-04:
> Sehr cool. Wie aktualisiert man die Version im Docker Image?

Leider muss ich dann immer eine neue Version bauen und hochladen. 
Wenn das automatisch bauen können soll, dann wird es kompliziert wegen der Credentials/Keys/Passwörter von git. Man müsste dann diese irgendwie mitgeben. Das war mir jetzt zu kompliziert. Darum ist es eine fertig gebaute Variante.

Aktuell ist es auch nur für amd64 Plattformen. Ich versuche gerade auch ein multibuild, sodass es auch auf einen Raspberry Pi lauffähig ist.

InScene, 2023-11-04:
Ich habe das Docker Image jetzt neu fertig erstellt und es läuft auf meinem Raspberry Pi. Habe es unter der selben Version nochmal hochgeladen.

jogybaer0815, 2023-11-10:
Danke für das Docker image. Läuft zuverlässig.
Sobald mehr Parameter verfügbar sich, um sie in z.B. Iobroker zu verwenden, sparen die neuen Wechselrichter sogar Hardware ein. 
Mir persönlich fehlt jetzt nur noch die Möglichkeit des 'Power Adjusments' per mqtt

InScene, 2023-11-10:
> Danke für das Docker image. Läuft zuverlässig. Sobald mehr Parameter verfügbar sich, um sie in z.B. Iobroker zu verwenden, sparen die neuen Wechselrichter sogar Hardware ein. Mir persönlich fehlt jetzt nur noch die Möglichkeit des 'Power Adjusments' per mqtt

Wir haben das ganze jetzt im original GitHub Repository eingebaut. Es werden jetzt automatisch neue Release als Image veröffentlicht. Man findet es unter folgender Adresse. 

https://hub.docker.com/r/dennisosrm/hms-mqtt-publisher

T3mplate-wow, 2023-11-11:
Meine minimalen Docker Kenntnisse scheinen nicht zu reichen. Weiß jemand was ich falsch mache?
Ich habe eine Docker-Compose Datei erstellt: 

> version: '3'
services:
  hms-mqtt-publisher:
    image: 29012019/hms-mqtt-publisher
    ports:
      - "1883:1883"
      - "9001:9001"
    environment:
      INVERTER_HOST: 10.0.0.218
      MQTT_BROKER_HOST: 10.0.0.59
      MQTT_USERNAME: mqtt-user
      MQTT_PASSWORD: Passwort

Wenn ich das dann mittels "docker compose up“ starte bekomme ich:

> hms-mqtt-publisher-hms-mqtt-publisher-1  | 2023-11-11T15:12:17 [INFO] - inverter: 10.0.0.218, mqtt broker 10.0.0.59
hms-mqtt-publisher-hms-mqtt-publisher-1  | 2023-11-11T15:12:26 [INFO] - Inverter is Online

Aber bei meinem HomeAssistant kommt kein neues MQTT Gerät an.

jogybaer0815, 2023-11-11:
die Variable des MQTT_PORT hast du angelegt?

T3mplate-wow, 2023-11-11:
Oh man. Das hätte ich auch mal so versuchen können. Ich ging wegen Standardport bzw. der Doku davon aus, dass die optional ist. Leider reicht die Sonne heute nicht mehr. Wird morgen getestet

T3mplate-wow, 2023-11-12:
keine Ahnung warum ich das Ticket geschlossen habe...

Leider sehe ich nach wie vor nichts in HomeAssistant.
Im Mosquitto Log sehe ich zwar:

> 2023-11-12 07:40:38: New connection from 10.0.0.45:53370 on port 1883.
2023-11-12 07:40:38: New client connected from 10.0.0.45:53370 as hms800wt2-mqtt-publisher (p2, c1, k5, u'mqtt-user’).

Aber trotzdem wird kein Device angelegt. Da sich der mqtt publisher aber offensichtlich anmeldet kann doch der einzige Grund sein, dass er keine Daten schickt. Oder?
Hat jemand eine Idee wie ich Client seitig analysieren kann oder eine andere Idee?

T3mplate-wow, 2023-11-12:
Gleiches Problem mittels des HA Hoymiles Wifi Addon:
https://github.com/dominikandreas/hms-mqtt-publisher/tree/main/ha-hoymiles-wifi-addon

Plugin läuft, sagt das der Inverter Online ist, aber es taucht kein neues MQTT Device auf...

jogybaer0815, 2023-11-12:
Der Hoymiles muss zwingend im gleichen Netz laufen - ein Gastnetz kann keine Verbindung herstellen

T3mplate-wow, 2023-11-12:
Alle Geräte sind gemeinsam im „normalen“/ nicht Gastnetz

T3mplate-wow, 2023-11-12:
Ich habe auf dem Mosquitto das Debugging eingeschaltet. Diese Kommunikation sehe ich:

> 2023-11-12 13:55:25: Received PINGREQ from hms800wt2-mqtt-publisher
2023-11-12 13:55:25: Sending PINGRESP to hms800wt2-mqtt-publisher
2023-11-12 13:55:30: Received PINGREQ from hms800wt2-mqtt-publisher
2023-11-12 13:55:30: Sending PINGRESP to hms800wt2-mqtt-publisher
2023-11-12 13:55:35: Received PINGREQ from hms800wt2-mqtt-publisher
2023-11-12 13:55:35: Sending PINGRESP to hms800wt2-mqtt-publisher
2023-11-12 13:55:36: Received SUBSCRIBE from hms800wt2-mqtt-publisher
2023-11-12 13:55:36: 	hms800wt2/pv_current_power (QoS 0)
2023-11-12 13:55:36: hms800wt2-mqtt-publisher 0 hms800wt2/pv_current_power
2023-11-12 13:55:36: Sending SUBACK to hms800wt2-mqtt-publisher
2023-11-12 13:55:36: Sending PUBLISH to hms800wt2-mqtt-publisher (d0, q0, r1, m0, 'hms800wt2/pv_current_power', ... (4 bytes))
2023-11-12 13:55:36: Received PUBLISH from hms800wt2-mqtt-publisher (d0, q0, r1, m0, 'hms800wt2/pv_current_power', ... (4 bytes))
2023-11-12 13:55:36: Sending PUBLISH to hms800wt2-mqtt-publisher (d0, q0, r0, m0, 'hms800wt2/pv_current_power', ... (4 bytes))
2023-11-12 13:55:36: Received PUBLISH from hms800wt2-mqtt-publisher (d0, q0, r1, m0, 'hms800wt2/pv_daily_yield', ... (3 bytes))
2023-11-12 13:55:40: Received PINGREQ from hms800wt2-mqtt-publisher
2023-11-12 13:55:40: Sending PINGRESP to hms800wt2-mqtt-publisher

Im Vergleich zu meinen anderen Geräten sind es relativ viele Kommunikationen und wenig Bytes.

lennart24, 2023-11-12:
Hi,
erstmal Danke an Dennis, super Arbeit! 
Die Verbindung mit Homeassistant funktioniert bei mir und man bekommt ja eine echt schöne Übersicht:
[image]
Hatte mir das Tool noch um die Leistung und den Ertrag der beiden Module ergänzt, sowie die Netzfrequenz ergänzt.

> Aber trotzdem wird kein Device angelegt. Da sich der mqtt publisher aber offensichtlich anmeldet kann doch der einzige Grund sein, dass er keine Daten schickt. Oder? Hat jemand eine Idee wie ich Client seitig analysieren kann oder eine andere Idee?

Für die Verbindung mit HA musste ich die Topics explizit abonnieren. Sollte in etwa so in der config aussehen:

```
sensor:
    - name: "Ertrag PV"
      state_topic: "hms800wt2/pv_daily_yield"
      device_class: "energy"
      unit_of_measurement: "Wh"
      state_class: "total_increasing"
```

Hast du das gemacht?

DennisOSRM, 2023-11-12:
Sieht erstmal ok aus von der Menge der Nachrichten. Der Publisher sendet ca. alle 30 Sekunden ein Update und dazwischen kommen die Pings um die Verbindung zu MQTT offen zu halten.

T3mplate-wow, 2023-11-12:
Vielen Dank für den Hinweis. Das habe ich nicht gemacht bzw. musste ich das noch nie bei HomeAssistant. 
Gab es irgendwo einen Hinweis darauf?
Ich habe jetzt folgendes in meiner configuration.yaml ergänzt:

```

mqtt:
  - sensor:
    - name: "Ertrag PV"
      state_topic: "hms800wt2/pv_daily_yield"
      device_class: "energy"
      unit_of_measurement: "Wh"
      state_class: "total_increasing"
```

Jetzt ist ein Sensor da mit einem Wert von 316Wh (ja nicht sehr viel). Mal schauen wie der sich ändert wenn morgen wieder die Sonne scheint. 
Vielen Dank schon einmal!

DennisOSRM, 2023-11-12:
> erstmal Danke an Dennis, super Arbeit! 

Vielen Dank, sehr gerne. 

> Die Verbindung mit Homeassistant funktioniert bei mir und man bekommt ja eine echt schöne Übersicht:

Sehr hübsch!

> Hatte mir das Tool noch um die Leistung und den Ertrag der beiden Module ergänzt, sowie die Netzfrequenz ergänzt.

Kommt demnächst auch direkt ins tool. 😉

T3mplate-wow, 2023-11-13:
Also mit dem Erstellen der mqtt devices in Home Assistant funktioniert das Ganze jetzt. Ich bin total begeistert. Viel häufiger als mit der Cloud. Toll.

Vielen Dank an alle für das Tool, das HomeAssistant Addon, den Docker und die Hilfe bei meinem Problem!
Ich freue mich auch darauf, wenn noch das 2. Modul ergänzt wird. 

3 Fragen die mir noch durch den Kopf gehen:
* Ist hier eigentlich das richtige Forum dafür, oder soll dieses Tool auf https://github.com/DennisOSRM/hms-mqtt-publisher besprochen werden?
* Mein HomeAssistant Addon ist am Abend abgestürzt, als der Hoymiles auch weg war. Ist das bekannt oder soll ich dafür einen Bug einstellen? Ich habe den Watchdog jetzt bewusst deaktiviert gelassen um zu schauen, ob das wieder passiert. 
* Mein Hoymiles wechselt ständig von Offline zu Online. So 10 mal pro Stunde. Ist das mein WLAN was so schwankt oder ist das normal oder weder noch?

T3mplate-wow, 2023-11-13:
> Hi, erstmal Danke an Dennis, super Arbeit! Die Verbindung mit Homeassistant funktioniert bei mir und man bekommt ja eine echt schöne Übersicht: [image] Hatte mir das Tool noch um die Leistung und den Ertrag der beiden Module ergänzt, sowie die Netzfrequenz ergänzt.
> 
Richtig cool. Wie berechnest du deinen aktuellen Stromverbrauch und deine Einspeiseanteil? Ich lese an meinem Stromzähler per Infrarot die Werte aus, was wegen der niedrigen Baudrate aber nur 1x alle 5 Minuten geht. Außerdem gibt meiner nur den Zählerstand aus und keinen Momentanwert (was alle 5 Minuten auch wenig Sinn ergibt). Ich berechne dann immer den Durchschnittsverbrauch der letzten 5 Minuten. Wie viel dann aber eingespeist wurde, bekomme ich nicht heraus. 

Falls du das teilen willst, wäre ich auch an dem Code von deinem Energiedashboard interessiert.

DennisOSRM, 2023-11-13:
> Also mit dem Erstellen der mqtt devices in Home Assistant funktioniert das Ganze jetzt. Ich bin total begeistert. Viel häufiger als mit der Cloud. Toll.

👍🏽

> * Ist hier eigentlich das richtige Forum dafür, oder soll dieses Tool auf https://github.com/DennisOSRM/hms-mqtt-publisher besprochen werden?

Gute Frage, gerne drüben falls das hier offtopic ist. 

> * Mein HomeAssistant Addon ist am Abend abgestürzt, als der Hoymiles auch weg war. Ist das bekannt oder soll ich dafür einen Bug einstellen? Ich habe den Watchdog jetzt bewusst deaktiviert gelassen um zu schauen, ob das wieder passiert. 

Hast Du ne genauere Fehlermeldung?

> * Mein Hoymiles wechselt ständig von Offline zu Online. So 10 mal pro Stunde. Ist das mein WLAN was so schwankt oder ist das normal oder weder noch? 

Entweder das WLAN oder Ertrag scheint schwach zu sein. Ich seh das bei starker Bewölkung zum Beispiel. Die Meldung kommt wenn die Netzwerkverbindung schwach ist.

lennart24, 2023-11-17:
> Gab es irgendwo einen Hinweis darauf?

So richtig nicht. Bei der Ersteinrichtung von der MQTT Integration in HA gibt man einen "Discovery-Präfix" an (steht per default auf homeassistant). Automatisch werden wohl nur Topics, die unter diesem Präfix veröffentlicht werden abonniert. Alle anderen muss man wohl manuell abonnieren. Wie das geht steht dann in den [Beispielen der Integration](https://www.home-assistant.io/integrations/mqtt/#discovery-examples).

> Kommt demnächst auch direkt ins tool. 😉

Ah super :) Dann werde ich wohl auch bald mal auf einen Docker-Container umstellen. Hätte auch schon ein Pull-Request erstellt, aber bis jetzt hatte ich das nur Quick&Dirty reingepatcht. Schön wäre es ja, wenn das Tool dynamisch erkennt, wie viele Eingänge der Wechselrichter hat - dann würde das evtl. auch mit den größeren Wechselrichtern laufen.

> Mein Hoymiles wechselt ständig von Offline zu Online. So 10 mal pro Stunde. Ist das mein WLAN was so schwankt oder ist das normal oder weder noch?

Bei mir geht er eigentlich morgens online und abends offline. Kommt aber ja auch darauf an, wie bei dir die Ausrichtung und Verschattungssituation ist - wenn da zwischendurch die Leistung zu gering ist, kann der Wechselrichter auch mal ausgehen. Müsstest du mal schauen, ob die grüne LED noch blinkt, wenn er offline angezeigt wird.

> Richtig cool. Wie berechnest du deinen aktuellen Stromverbrauch und deine Einspeiseanteil? Ich lese an meinem Stromzähler per Infrarot die Werte aus, was wegen der niedrigen Baudrate aber nur 1x [...]

Ich hab bei mir einen KNX-Energiezähler verbaut, der die Energie in beide Richtungen separat misst und auf den Bus schreibt. Die Werte lese ich einfach mit HA ein.
Der Zählerstand ist ja schon der richtige Wert, den du für diese Diagramme brauchst. Die Balken sind ja einfach nur die Differenz zwischen dem Zählerstand am Ende und am Anfang einer Stunde.
Hast du einen Zweirichtungszähler? Sonst wird das mit der Einspeisung wahrscheinlich nicht ohne zusätzliches Messgerät funktionieren. Bei einem Zweirichtungszähler müsstest du schauen, ob der dir über Infrarot auch den Zählerstand in Einspeiserichtung ausgibt.
Die Diagramme erstellt HA von selbst, wenn man im Energiedashboard die ganzen Sensoren hinzufügt. Sieht bei mir so aus:
[image]
Auch der prognostizierte Ertrag wird automatisch errechnet, wenn man Ausrichtung, Standort und Peak-Leistung der Module angibt.

ckmde, 2023-11-18:
Hallo Dennis,

funktioniert sehr gut. Danke. Benutze das jetzt an FHEM.
Habe das ganze für mich etwas modifiziert. Dabei ist mir aufgefallen, dass die Protocol Buffers Definition der Seriennummer in RealData.proto nicht ganz passt.

message HMSStateResponse {
  bytes dtu_sn = 1;
.
.
ist definitiv ein string dtu_sn  = 1; und nicht bytes...
Habe ich mir mit tcpdump angesehen. Damit hat man dann die korrekte Seriennummer auf HMSStateResponse.dtu_sn, so wie sie auch in der App und auf dem Inverter steht.
Und da ich gerne Daten sammele habe ich die mqtt.rs erweitert um alle möglichen Sachen, wie die Netzspannung, Temperatur, Frequenz sowie die Daten der beiden PV Stränge wie Spannung, Strom etc.
Falls das also Interessant ist, hier die geänderte publish Funktion. Spart dem einen oder anderen ggf. Tipparbeit. Erzeugt allerdings auch deutlich mehr Daten auf meinem FHEM Server. :-)

    fn publish(&mut self, hms_state: &HMSStateResponse) {
        debug!("{hms_state}");

        let pv_current_power = hms_state.pv_current_power as f32 / 10.;
        let pv_daily_yield = hms_state.pv_daily_yield;
        let pv_grid_voltage = hms_state.inverter_state[0].grid_voltage as f32 / 10.;
        let pv_grid_freq = hms_state.inverter_state[0].grid_freq as f32 / 100.;
        let pv_grid_temperature = hms_state.inverter_state[0].temperature as f32 / 10.;
        let pv_port1_voltage = hms_state.port_state[0].pv_vol as f32 / 10.;	
        let pv_port1_curr = hms_state.port_state[0].pv_cur as f32 / 100.;
        let pv_port1_power = hms_state.port_state[0].pv_power as f32 / 10.;
        let pv_port1_energy = hms_state.port_state[0].pv_energy_total as f32 / 10.;
        let pv_port1_daily_yield = hms_state.port_state[0].pv_daily_yield as f32 / 10.;
        let pv_port2_voltage = hms_state.port_state[1].pv_vol as f32 / 10.;	
        let pv_port2_curr = hms_state.port_state[1].pv_cur as f32 / 100.;
        let pv_port2_power = hms_state.port_state[1].pv_power as f32 / 10.;
        let pv_port2_energy = hms_state.port_state[1].pv_energy_total as f32 / 10.;
        let pv_port2_daily_yield = hms_state.port_state[1].pv_daily_yield as f32 / 10.;
        self.client
            .subscribe("hms800wt2/pv_current_power", QoS::AtMostOnce)
            .unwrap();
        match self.client.publish(
            "hms800wt2/pv_current_power",
            QoS::AtMostOnce,
            true,
            pv_current_power.to_string(),
        ) {
            Ok(_) => {}
            Err(e) => warn!("mqtt error: {e}"),
        }
        match self.client.publish(
            "hms800wt2/pv_daily_yield",
            QoS::AtMostOnce,
            true,
            pv_daily_yield.to_string(),
        ) {
            Ok(_) => {}
            Err(e) => warn!("mqtt error: {e}"),
        }
        match self.client.publish(
            "hms800wt2/pv_grid_voltage",
            QoS::AtMostOnce,
            true,
            pv_grid_voltage.to_string(),
        ) {
            Ok(_) => {}
            Err(e) => warn!("mqtt error: {e}"),
        }
        match self.client.publish(
            "hms800wt2/pv_grid_freq",
            QoS::AtMostOnce,
            true,
            pv_grid_freq.to_string(),
        ) {
            Ok(_) => {}
            Err(e) => warn!("mqtt error: {e}"),
[... 77 lines cut ...]
            pv_port2_power.to_string(),
        ) {
            Ok(_) => {}
            Err(e) => warn!("mqtt error: {e}"),
        }        
	match self.client.publish(
            "hms800wt2/pv_port2_energy",
            QoS::AtMostOnce,
            true,
            pv_port2_energy.to_string(),
        ) {
            Ok(_) => {}
            Err(e) => warn!("mqtt error: {e}"),
        }        
	match self.client.publish(
            "hms800wt2/pv_port2_daily_yield",
            QoS::AtMostOnce,
            true,
            pv_port2_daily_yield.to_string(),
        ) {
            Ok(_) => {}
            Err(e) => warn!("mqtt error: {e}"),
        }
    }
}

DennisOSRM, 2023-11-18:
Cool, danke für den Code. Sobald etwas Zeit ist, bau ich das ein.

jogybaer0815, 2023-11-19:
Test scheitert leider: 
'dominikandreas/hms-mqtt-publisher-nightly' oder 'dominikandreas/hms-mqtt-publisher'  findet Docker nicht.
Liegt das an den 2 Versionen mit und ohne "-nightly"?

DennisOSRM, 2023-11-19:
Benutz doch den build vom letzten Release in DockerHub: 'dennisosrm/hms-mqtt-publisher'.

jogybaer0815, 2023-11-19:
Danke. Der läuft ja auch, allerdings sind dort noch keine erweiterten Daten hinterlegt

karstenatgit, 2023-11-21:
Hallo zusammen, klingt super, was ihr hier macht :-)
Ich bin gerade dabei, meine Anlage in Betrieb zu nehmen. Kann ich den Hoymiles eigentlich irgendwie in mein WLAN bekommen, ohne ihn in der HM-Cloud zu registrieren? Noch ist er jungfräulich und nur sein eigener Accesspoint läuft.

karstenatgit, 2023-11-30:
Ohne Hoymiles-Account kann ich mich zwar mit dem integrierten AccessPoint verbinden, ja. Aber ich kann damit dem Wechselrichter keine Zugangsdaten für mein WLAN geben. Oder hab ich da was übersehen?
[image]
[image]

karstenatgit, 2023-11-30:
Ok, also offenbar nur mit Account bei Hoymiles. Sonst kommt man nicht zur Verbindung mit dem eigenen WLAN. Naja, werde ich demnächst mal machen. Und vorher überlege ich mir, wie ich den Datenverkehr dabei mitschneide, vielleicht können die Experten hier was damit anfangen. Ein Raspi als WiFi-Repeater mit einem Wireshark sollte doch dafür gehen...

DennisOSRM, 2023-11-30:
Es gibt einen Request um die u.a. die WLAN Konfiguration zu ändern. So wie es aussieht kann man das am Gerät setzen, ohne dass dort irgendein Cloudzugang gecheckt würde. 

Ausprobiert habe ich das aber nicht. Das war mir aber dann doch zu riskant am eigenen Gerät. 😉

karstenatgit, 2023-11-30:
Ja, dieser Request wäre es dann vermutlich. Gibt es auch einen für "Reset auf Auslieferungszustand", falls was schiefgeht?

karstenatgit, 2023-12-01:
Ist der interne Access Point des DTUBI denn aus, sobald die WLAN - Konfiguration steht?

philiip11, 2023-12-01:
Der interne Access Point ist immer an, wenn es hell genug ist. Auch dann, wenn eine Verbindung zum WLAN besteht.

KFU141, 2023-12-02:
Servus beinand,
ich bin auch Besitzer eines HMS 800W-2T und total genervt davon, dass man nicht wie bei den Vorgängergeräten mit Open DTU per MQTT an die Daten kommt. Daher nochmal Hut ab, was du hier aufgesetzt hast Dennis.
Allerdings struggle ich gerade den Docker Container auf meinem Raspi zum Laufen zu bekommen. Die Variablen, die das Image braucht hab ich alle gesetzt aber irgendwas fehlt noch, dass der Container wirklich läuft. Kann hier jemand helfen und ggf. den Befehl posten mit welchem ich den Container anlege und starte?
Danke

DennisOSRM, 2023-12-02:
> Servus beinand, ich bin auch Besitzer eines HMS 800W-2T und total genervt davon, dass man nicht wie bei den Vorgängergeräten mit Open DTU per MQTT an die Daten kommt. Daher nochmal Hut ab, was du [...]

Schau mal hier, ob das Dein Problem behebt: https://github.com/DennisOSRM/hms-mqtt-publisher/issues/45#issuecomment-1834592232

KFU141, 2023-12-02:
Danke für die schnelle Antwort, aber leider funktioniert auch das nicht. Der Container startet alle paar Sekunden neu und hat weiterhin Probleme mit fehlenden Argumenten.

`e6a152c1696d   dennisosrm/hms-mqtt-publisher:latest   "/bin/sh -c hms-mqtt…"   19 minutes ago   Restarting (101) 21 seconds ago                                                    hms-mqtt-publisher
`

Ich habe das config.toml im zugehörigen Volume angepasst und die simple_mqtt Infos eingetragen. Inverter, MQTT Broker und Port. Passwort hab ich in meiner Umgebung nicht.

`toml config unparsable: Error { inner: Error { inner: TomlError { message: "invalid string\nexpected `\"`, `'`", original: Some("inverter_host = \"\" \n\n[home_assistent] \nhost = \"\"\nusername = \"\"\npassword = \"\"\nport = \n\n"), keys: [], span: Some(84..85) } } }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace`

Abgelegtes config.toml:
`inverter_host = "192.168.178.144"

[simple_mqtt]
host = "192.168.178.141"
port = 1883
`

KFU141, 2023-12-09:
Hi @DennisOSRM, ich hab es nun endlich geschafft den Container zum Laufen zu bekommen. Ich habe in den Topics für die Gesamtenergie des jeweiligen Solarpannels aber noch einen Kommafehler gefunden. Meine Anlage hat seit Aufzeichnung ~177kWh Ertrag zu verzeichnen. Das wären also gerundet 90kWh+87kWh. In den Topics kommt es aber mit 8989.3 statt 89893 an. Entweder muss das Komma zwei Stellen nach vorn verschoben werden dann kann man kWh als Einheit nehmen, oder eben eine Stelle nach hinten dann sind es Wh. Eine sinnvolle Einheit dazwischen ist mir nicht bekannt. Das kann man natürlich korrigieren indem man zB x10 rechnet für die Ausgabe, aber wäre super, wenn es schon passend per MQTT kommt:
[image]

KFU141, 2023-12-09:
Danke für den schnellen Fix. Scheint das du leider meine Ergänzung im Kommentar verpasst hast. Da ist noch ein zweiter Wert, der nicht passt.
Mir ist zudem aufgefallen, dass sobald der Docker Container läuft und die Daten an den MQTT Broker gehen ist keine Verbindung aus der Hoymiles Cloud auf den Inverter mehr möglich. Ist das so gewollt? Wäre eine parallele Verbindung möglich? Es war von unterwegs ganz hilfreich die Werte zu sehen, auch wenn sie nur alle 30min kommen. Zudem muss man nun den Docker stoppen, um ein Update der Firmware aus der Cloud triggern zu können. Das sollte man beachten.

[46 more replies]