lumapu/ahoy #1192: Waveshare 1,54" ePaper rev.1

elias0172, 2023-09-28:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.7.59
Github Hash: _______

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Ich habe Schwierigkeiten bei der Inbetriebnahme der dtu mit einem ePaper Display. Die dtu funktioniert, vielen Dank für die Bereitstellung des Projektes!
Ich möchte ein Display von Waveshare, ich vermute in der V1, nutzen, es ist kein V2 Aufkleber auf der Rückseite angebracht, das Display ist bereits sechs Jahre alt, daher gehe ich von einer V1 aus. Ich habe die verschiedensten Pin-Konfigurationen getestet, alle ohne Erfolg. Bei der Nutzung der Arduino Beispiele des ESP funktioniert das Display, gleiches Anschlussschema. Ich habe auf der Waveshare Webseite gelesen, dass man für V1 Modelle einen anderen Code verwenden soll, damit es funktioniert. 
https://www.waveshare.com/wiki/1.54inch_e-Paper_Module_Manual#Resources
Des weiteren gibt es such optisch auf Bildern unterschiedliche / fehlende Bauteile zwischen V1/V2.

teletapi, 2023-09-28:
Es wird in Ahoy ausschließlich das waveshare Epaper 1,54" mit er revision 2.1 unterstützt. alles andere wäre Zufall

knickohr, 2023-09-29:
Funktioniert leider nicht. Das wäre eine komplett andere Library und fast eine Neuprogrammierung der ePaper Routine. Außerdem ist dieses Modul mit der V1 schon seit ca. 2 Jahren deprecated. Ja ich weiß, es schwirren noch Tausende von diesen Midulen, vor allem bei Billiganbietern, im Internet rum.

elias0172, 2023-10-01:
Danke für die Rückmeldung. Schade aber ich habe Verständnis.

closed 2023-10-02 by elias0172