lumapu/ahoy #1193: Feature Request :  Kennzeichnung der Settings mit zwingendem Reboot

knickohr, 2023-09-29:
Stört mich schon seit je her das man immer rebooten muß damit die Settings übernommen werden. Einige Settings können jedoch auch ohne Reboot aktiv werden. Ich bitte um die Kennzeichnung (z.B. *) der Settings die einen zwingenden Reboot erfordern.

Bei der Angelegenheit kann man dann auch den Haken „ Reboot device after successful save“ default erdfernen. Statt dessen eine Fußnote :

* changeing this setting needs a reboot

einfügen.

stefan123t, 2023-09-29:
Prinzipiell einen Hinweis einblenden, wenn ein Setting geändert wurde, das einen Reboot erfordert ist gut.
Verwendung von `*` halte ich  für problematisch, das ist normalerweise schon für Pflichtangaben (!) reserviert.
Entweder **) oder 2) fände ich daher besser, wenn es nicht gleich ein zusätzlicher Label/Dialog, etc. wird.

knickohr, 2023-09-29:
Wie die Fußnote aussieht ist mir schurzupurzi 😅

knickohr, 2024-06-02:
Uiii, alte Baustelle, aber immer noch aktuell 🫣

stefan123t, 2024-10-29:
@lumapu ja das hatte ich Dir auch neulich per PN mal geschrieben, daß das bei OpenDTU schon lange möglich ist und mir nach einem ca. Jahr Pause in den AhoyDTU Settings auch aufgefallen war, dass ich zum Speichern immer einen Reboot ausgelöst habe, obwohl es eigentlich nicht zwingend notwendig ist.

+1 von meiner Seite.