lumapu/ahoy #1194: Webseitenaufruf beim ESP32 dauert sehr lang

teletapi, 2023-09-29:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.60

### Github Hash

963e618

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

.

### Debug Serial Log output

```shell
.
```

### Error description

Ich habe beim ESP32 (38Pin) festgestellt das nachdem ich die 0.7.60 geflasht habe der Webseitenaufruf sehr langsam geht und auch der Wechsel zwischen den Menue seiten(SYStem,Update;Settings usw.) teilweise bis zu 20-25 Sekunden dauert.

Beim ESP32 S3 oder Fusionsboard konnte ich das nicht feststellen.

HorstyS, 2023-09-30:
Das hatte ich auch am Anfang nach dem Upgrade. Hat sich wieder einigermaßen gelegt aber ist definitiv etwas langsamer als vor der .60

rnwgnr, 2023-09-30:
Das gleiche bei mir mit der 0.7.60 - mit der 0.7.59 ging das alles noch super zügig.

knickohr, 2023-09-30:
Und da ist es wieder, das Schwuppdizitätsproblem 😢

Was viel schlimmer ist, es gibt Aussetzer im Abfrageintervall der Inverter. Bei meinen 13 Invertern wird mindestens einer pro Zyklus vergessen. Er wird nicht einmal abgefragt, also kommen auch keine Werte zurück. Kann man in der seriellen Konsole sehr gut erkennen, er wird einfach vergessen, also nicht mal ein TX gesendet.

Da es immer wieder ein anderer Inverter ist, manchmal auch mehrere pro Zyklus, gehe ich davon aus das es nicht inverterspezifisch ist.

lumapu, 2023-09-30:
bin gespannt wo wir uns das wieder eingefangen haben, das waren leider einige Pull-Requests ...

lumapu, 2023-09-30:
also ein Diff zwischen `0.7.59` (eaa69c5) und `0.7.60` (963e618) sieht nicht tragisch aus. Seid ihr sicher, dass es nicht schon vor `0.7.59` reingekommen ist?

Diff: https://github.com/lumapu/ahoy/compare/eaa69c5...963e618

HorstyS, 2023-10-01:
Ziemlich. Ich hatte zuvor die .59 und sofort nach dem Update auf die .60 ist es passiert. 
Ich habe eine einfache Konfiguration mit nur einem Inverter und ohne MQTT oder sonstigen Spielerein (aber ePaper Display). Aber wie gesagt, es hat sich nach ein paar Stunden ziemlich gelegt und ist wieder schneller geworden. Aber nicht so schnell wie meine zweite DTU mit der .59 (mit OLED Display).

rnwgnr, 2023-10-01:
Bin mir auch sehr sicher das es von .59 auf .60 reingekommen ist, die .59 war super fluffig.

Mit der .60 verliert das Board auch immer wieder mal die WLAN-Verbindung, Aktualisierung des Displays nur alle 1 bis 2 Minuten auch bei kürzer eingestellten Intervallen.

Wollte über die Weboberfläche das Update auf .61 einspielen, bin aber mehrfach gescheitert. Nach dem Klick auf "Update" lädt die Seite entweder ewig oder bricht direkt mit nem Ladefehler ab.

Komm ich noch irgendwie an die .59? Bin recht neu im "DTU-Business". :)

knickohr, 2023-10-01:
„ Wollte über die Weboberfläche das Update auf .61 einspielen, bin aber mehrfach gescheitert. Nach dem Klick auf "Update" lädt die Seite entweder ewig oder bricht direkt mit nem Ladefehler ab.“

Das deutet tatsächlich auf ein Schwuppdizitätsproblem hin. Habe ich auch immer gehabt. Lösung :  Die DTU im AP-Modus starten und dort den Update über OTA machen.

lumapu, 2023-10-01:
> Komm ich noch irgendwie an die .59? Bin recht neu im "DTU-Business". :)

ja hier: https://fw.ahoydtu.de

rnwgnr, 2023-10-01:
> Das deutet tatsächlich auf ein Schwuppdizitätsproblem hin. Habe ich auch immer gehabt. Lösung : Die DTU im AP-Modus starten und dort den Update über OTA machen.

Hab ich grad mal probiert, hat funktioniert. Komischerweise ist der AP scheinbar bei der Verbindung übers Notebook immer "abgestürzt" (Verbindung fehlgeschlagen, AP dann weg), übers Handy gings dann problemlos. Mit 0.7.59 schwuppt es jetzt wieder.

teletapi, 2023-10-01:
Alsoi der Webseitenaufruf und die Menuewechsel funktionieren mit der 0.7.62 wieder Top beim ESP32

teletapi, 2023-10-01:
Also der Webseitenaufruf und die Menuewechsel funktionieren mit der 0.7.62 wieder Top beim ESP32

teletapi, 2023-10-01:
Ich musste das Issue wieder öffnen da ich versehentlich nicht die 0.7.62 auf den ESP geflasht hatte sondern das Stable,0.7.36 und nachdem ich dann die richtige 0.7.62 wieder geflasht habe erkannt habe das die Version 0.7.62 wohl fehlerhaft ist und dadurch nicht wirklich getestet werden kann

knickohr, 2023-10-01:
Zu - Auf - Zu - Auf - Blinker 😂

closed 2023-10-01 by teletapi

rnwgnr, 2023-10-01:
Closed weil gefixed in .63?

knickohr, 2023-10-01:
Korrekt