lumapu/ahoy #1195: Datum und Uhrzeit werden im Browser falsch angezeigt

rmayergfx, 2023-09-30:
Bin gerade außerhalb der Zeitzone und der AhoyDTU arbeitet. Was soll ich sagen, die Zeit des inverters wird auf die lokale Zeit des Endgerätes modifiziert in der Anzeige, was so nicht sein kann bzw darf. Gehe ich Remote per VPN auf das Gerät muss er mir doch die lokale Zeit anzeigen, in der das Gerät steht und nicht die Uhrzeit meiner aktuellen Zeitzone. Auch in der Darstellung zum Sonnenaufgang bzw -untergang ist diese falsch dargestellt.

Joh49, 2023-09-30:
Kann es die Zeit beim letzten VPN Server sein?

lumapu, 2023-09-30:
ganz neues Szenario - da müssen wir konzeptionell was umbauen 😀

rmayergfx, 2023-10-01:
> Kann es die Zeit beim letzten VPN Server sein?

Bitte ganze und verständliche Sätze schreiben damit man auch versteht was du meinst. 
Es hat definitiv nichts mit dem VPN Server zu tun, die Zeit wird vom lokalen Server geholt, dieser läuft in der Zeitzone MESZ. Ein Rechner im VPN Netzwerk vor Ort zeigt auch die richtigen Zeiten an. Das Problem kann man auch ganz einfach reproduzieren, am Gerät auf dem der Browser offen ist den automatischen Abgleich der Systemuhr ausschalten und eine andere Zeitzone auswählen.

gokhanakca, 2024-04-26:
Kann selbes Problem in Version 0.8.83 auf einem ESP8266 bestätigen. 
Die ahoy Oberfläche scheint immer die Uhrzeit des Browsers zu verwenden. Auch in den Ausgaben der Web Konsole. 
Habe zu Hause angerufen, um dort die Oberfläche aufrufen zu lassen- es wird die deutsche Uhrzeit angezeigt.