lumapu/ahoy #1196: Keine Verbindung zu HM-350

NewBee51, 2023-09-30:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [x ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version:0.7.36
Github Hash: _______

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [x] Setup settings (use our templates ... to be added)

NewBee51, 2023-09-30:
Hallo Zusammen,

ich habe mir eine Ahoi - DTU mit dem Wemos D1 ESP32 gebaut. Das Aufspielen der Software hat gut funktioniert, die DTU ist auch im WLAN erreichbar und kann konfiguriert werden.

Leider lässt sich mein WR HM-350 nicht verbinden, die Seriennummer habe ich korrekt eingegeben.

Was kann ich sonst noch prüfen, testen ?

Viele Grüße

NewBee51

Pinning:

[image]

System Status:

[image]

NewBee51, 2023-09-30:
Hier noch ein Auszug aus dem Serial Log:

14:09:16 I: (#0) resetPayload
14:09:16 I: (#0) prepareDevInformCmd 0x0b
14:09:16 I: TX 27B Ch23 | 15 82 84 26 43 80 49 19 40 80 0b 00 65 18 0f ec 00 00 00 00 00 00 00 00 12 ff 1e 
14:09:26 I: (#0) resetPayload
14:09:26 I: (#0) enqueCommand: 0x01
14:09:26 I: (#0) enqueCommand: 0x0b
14:09:26 I: (#0) prepareDevInformCmd 0x01
14:09:26 I: TX 27B Ch40 | 15 82 84 26 43 80 49 19 40 80 01 00 65 18 0f f6 00 00 00 00 00 00 00 00 78 46 dd 
14:09:36 I: (#0) resetPayload
14:09:36 I: (#0) prepareDevInformCmd 0x0b
14:09:36 I: TX 27B Ch61 | 15 82 84 26 43 80 49 19 40 80 0b 00 65 18 10 00 00 00 00 00 00 00 00 00 73 52 21 
14:09:46 I: (#0) resetPayload
14:09:46 I: (#0) enqueCommand: 0x01
14:09:46 I: (#0) enqueCommand: 0x0b
14:09:46 I: (#0) prepareDevInformCmd 0x01
14:09:46 I: TX 27B Ch75 | 15 82 84 26 43 80 49 19 40 80 01 00 65 18 10 0a 00 00 00 00 00 00 00 00 d9 26 ff 
14:09:56 I: (#0) resetPayload
14:09:56 I: (#0) prepareDevInformCmd 0x0b
14:09:56 I: TX 27B Ch3 | 15 82 84 26 43 80 49 19 40 80 0b 00 65 18 10 14 00 00 00 00 00 00 00 00 73 ad ca 
14:10:06 I: (#0) resetPayload
14:10:06 I: (#0) enqueCommand: 0x01
14:10:06 I: (#0) enqueCommand: 0x0b
14:10:06 I: (#0) prepareDevInformCmd 0x01
14:10:06 I: TX 27B Ch23 | 15 82 84 26 43 80 49 19 40 80 01 00 65 18 10 1e 00 00 00 00 00 00 00 00 d9 d9 14 
14:10:16 I: (#0) resetPayload
14:10:16 I: (#0) prepareDevInformCmd 0x0b
14:10:16 I: TX 27B Ch40 | 15 82 84 26 43 80 49 19 40 80 0b 00 65 18 10 28 00 00 00 00 00 00 00 00 72 ac f6

rmayergfx, 2023-09-30:
Welches motherboard? Pinout prüfen, siehe https://ahoydtu.de/getting_started/

rejoe2, 2023-09-30:
Und runter mit der Sendeleistung, wenn du nicht grade Dutzende von Metern weg bist vom Inverter!

NewBee51, 2023-09-30:
[image]
Ich habe diese Platine - in der Beschreibung steht leider nur das Pinning für Open-DTU, ich brauche aber das für Ahoy.

lumapu, 2023-09-30:
PCB gefunden, leider kein Schaltbild. Nach meiner Analyse (verfolgen der Leiterbahnen) musst du folgendes ändern:
* `IRQ` auf `GPIO17`
* `CE` auf `GPIO16`

Quellen, die ich genutzt habe (Pinout von Wemos D1 ESP32 und NRF24)
* https://community.openmqttgateway.com/t/esp32-mini-kit-pinout-is-there-a-mapping-table-to-lolin-esp32-pinout/2320
* https://howtomechatronics.com/tutorials/arduino/arduino-wireless-communication-nrf24l01-tutorial/

rmayergfx, 2023-10-01:
> [image] Ich habe diese Platine - in der Beschreibung steht leider nur das Pinning für Open-DTU, ich brauche aber das für Ahoy.
Wenn ihr schon spezielle Boards benutzt dann bitte auch gleich in die Anfangspostings mit reinschreiben und auch die Infos auf der Homepage zu den Boards lesen: https://github.com/turrican944/OpenDTU-PCB

lumapu, 2023-10-01:
da hätte ich mir paar Minuten sparen können - habe nicht nachgeschaut, ob es irgendwo dokumentiert ist - naja jetzt ist es verifiiert 🙈

NewBee51, 2023-10-01:
Guten Morgen, ich habe das Pinning jetzt ausprobiert, leider ohne Erfolg. Ich habe auch testweise einmal IO 16 und IO 17 getauscht. Immer noch keine Verbindung zum HM-350

rmayergfx, 2023-10-01:
Hier das Pinout vom Wemos D1 ESP32 https://www.fambach.net/d1-mini-esp8266-modul-2-2/
Modul stromlos machen und die entsprechenden PINS mit dem Multimeter durchpiepsen! Hatte auch mal ein Mainboard von einem anderen Hersteller das leider nicht sauber durchkontaktiert war!

NewBee51, 2023-10-01:
Hurra, es geht jetzt. Ich habe zuerst die Platine durchgepiepst, da war alles O.K. Danach habe ich nochmal alle Lötstellen am Funkmodul nachgelötet - und jetzt funktioniert es. Vielen Dank nochmal für die Unterstützung hier

lumapu, 2023-10-01:
sehr cool, dann kann man hier zumachen!

closed 2023-10-01 by lumapu