lumapu/ahoy #1198: [Bug] 0.7.61 Inverter is not available

Ollipop030, 2023-10-01:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.61

### Github Hash

5a01cc8

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

none

### Error description

Grad das Update von 0.7.60 auf die 61 gemacht. Die Inverter sind "not available", auch nach ein paar Minuten Wartezeit. Unter System fehlen auch die ganzen RX/TX Counter. Bin dann wieder zurück auf die 0.7.60, hier sind die Inverter nach ein paar Sekunden zu sehen.

dtuuser, 2023-10-01:
Bei mir auch so. Bin auch zurück auf 7.60 und alles wieder da

BastyESP, 2023-10-01:
Kann ich bestätigen, das war wohl nichts....bin wieder zurück auf die 0.7.60

Peertje-62, 2023-10-01:
Good morning,
idem, back to 0.7.59
regards

lumapu, 2023-10-01:
bei mir auch, denke ich kenne das Problem

lumapu, 2023-10-01:
neue Version baut, ich hoffe das Feedback ist dann wieder besser 😅

kiu77, 2023-10-01:
0.7.6.2
Ebenfalls kein Kontakt zum Inverter

Senderti, 2023-10-01:
Auch mit 0.7.62 besteht keine Verbindung zum Inverter.

dtuuser, 2023-10-01:
Mit 7.63 klappt es wieder. Danke

MiniOh, 2023-10-02:
Ich bin von der .60 auf die .65 und die beiden HMS-2000 komminuzieren nicht mehr.
Wieder zurück zur .60.
Die 5 HM Inverter funtionieren wieder, die beiden HMS nicht.
Seit 4 Stunden "Inverter #3: HMS-2000-03 is not yet available"

Neustart der DTU und auch erneutes Flashen der .60 half nicht.

lumapu, 2023-10-02:
das sollte maximal 15 min so sein, dann sollten deine Inverter wieder kommunizieren

MiniOh, 2023-10-02:
Ja, 5 HM und 2 HMS.
Die beiden HMS haben heute bis zum Tagesende nicht mehr kommuniziert.
Auch nicht mit dem Downgrade auf 0.7.60

Mal sehen wie es morgen aussieht, nachdem die Inverter offline waren.

MiniOh, 2023-10-03:
@lumapu 
Funktioniert heute morgen wieder mit den beiden HMS.

lumapu, 2023-10-05:
ich denke es ist bei allen jetzt gelöst oder habe ich was verpasst?

MiniOh, 2023-10-06:
Hey, nein, bei mir funktioniert es bei HM + HMS noch nicht.
Auch nicht mit der 0.7.66.

Das Verhalten ist wie folgt:
Mit der 0.7.60 funktionieren HM und HMS Inverter problemlos.
Nach dem Update auf 0.7.66 antworten HM und HMS einige Minuten lang wie erwartet.
Nach ca. 20 - 60 min gibt es keine Rückmeldung mehr seitens der HMS.
Auch nach dem Downgrade auf 0.7.60 antworten die HMS nicht mehr.
Erst nachdem die Inverter Offline waren, also idR am näcshten Morgen funktioniert die Kommunikation mit der 0.7.60 wieder.

Die GUI sieht wie folgt aus:
[image]

kiu77, 2023-10-06:
Habe ich den Build 0.7.66 verpasst? Ich finde den garnicht.

lumapu, 2023-10-06:
https://fw.ahoydtu.de

lumapu, 2023-10-06:
@MiniOh danke für's Feedback, dann arbeiten wir hier weiter an dem Fehler. Evtl. gibt es Parallelitäten zu #1202

lumapu, 2023-11-18:
@MiniOh hast du die neuesten `0.8.x` Versionen schon testen können?

Peertje-62, 2023-11-19:
Yes , I have , development version 0.8.7.,
very nice, no problems, (also in Grafana)
Thx & Kind regards

[image]

MiniOh, 2023-11-23:
Hallo @lumapu 
nein noch nicht. Hatte bisher die 0.7.60 seit Wochen ohne jeglichen Fehler laufen.
Habe jetzt mal die 0.8.12 installiert und berichte.

MiniOh, 2023-11-23:
Erste 2 Erkenntnisse meinser seits nach dem Update von 0.7.60 auf 0.8.12.

#1
Die DTU war nach dem Update nicht mehr erreichbar. Stromlos machen half, allerdings war die DTU nach 5min wieder nicht erreichbar. Beim 3. Reboot lief die DTU jetzt mal für 15min.
Freeheap ~130000 (Sonst waren es meinst 170000)

#2
Es findet keine Kommunikation mit den HMS Invertern statt. Ich habe bisher noch nichts an den Settings angepasst.
20db / 860MHz sind eingestellt.

[image]

Uptime: 0 Days, 00:04:46
ESP-Time: 2023-11-23, 09:02:47

System Infos:
Inverter #0: HM-600-01 is available and is producing 6.9W
Inverter #1: HM-1500-01 is available and is producing 26.9W
Inverter #2: HM-1500-02 is available and is producing 23.4W
Inverter #3: HMS-2000-03 is not yet available
Inverter #4: HM-1500-04 is available and is producing 26.3W
Inverter #5: HMS-2000-05 is not yet available
Inverter #6: HM-1500-06 is available and is producing 25.3W

Radio NRF
NRF24L01	connected
NRF24 Data Rate	250 kbps
DTU Radio ID	81479844
Radio CMT
CMT2300A	connected
DTU Radio ID	81001765
MqTT
connected	true
#TX	11194
#RX	0

MiniOh, 2023-12-13:
@lumapu 

Kleines Update von mir.
Ich habe nun mal die 0.8.21 installiert.

Das Update von 0.7.60 auf 0.8.21 läuft nicht ganz problemlos.
Nach dem Update ist die DTU nicht per IP erreichbar.
Paar mal USB Port neu einstecken schaffte aber Abhilfe.

Nach dem ändern der Radio Settings der HMS Inverter auf 865 MHz funktioniert die Kommunikation mit den HMS Invertern. Mit dem default Wert von 863 MHz werden keine Daten geliefert.

Bisher läuft die DTU die ersten 30min stabil.

Free Heap ca. 140.000. Mit der 0.7.60 waren es ca. 170.000, falls das von Interesse ist.
7 aktive Inverter.

[image]

closed 2023-12-29 by lumapu