lumapu/ahoy #1200: NRF System Daten fehlen und kein WR wird gefunden

teletapi, 2023-10-01:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [X] I will attach/upload an Image of my wiring

### Version

0.7.62

### Github Hash

169eea7

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

.

### Debug Serial Log output

```shell
.
```

### Error description

Nachdem flashen auf die 0.7.62 fehlen in System Menue die Daten fürs NRF und es wird auch kein Wechselrichter mehr gefunden. Ich denke da hat sich iwo ein Fehler eingeschlichen
[image]

Riker18, 2023-10-01:
Die Daten sind jetzt unter Live - aber NRF geht trotzdem nicht  - CMT geht

rmayergfx, 2023-10-01:
Siehe: https://github.com/lumapu/ahoy/issues/1198

lumapu, 2023-10-01:
@teletapi so sieht es jetzt aus:
[image]

dtuuser, 2023-10-01:
Bei mir klappt es auch wieder. Danke

teletapi, 2023-10-01:
oh danke

closed 2023-10-01 by teletapi