lumapu/ahoy #1201: Update nicht möglich[Bug]

persil14, 2023-10-01:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.59

### Github Hash

3a29d6c

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Einstellungen nicht geändert, nur Update versucht

### Error description

Update auf 0.7.62 nicht möglich. Meldung Update: Neustart erfolgreich ... automatisches Neustart.. aber immer noch die 0.7.59?

rmayergfx, 2023-10-01:
Same issue again?

lumapu, 2023-10-01:
hier bei der Fehlersuche in Version `0.7.62` nutze ich dauernd OTA und es funktioniert sicher in alle Richtungen.

knickohr, 2023-10-01:
Versuch mal die DTU in den AP-Modus zu bringen und dann den Update zu machen.

persil14, 2023-10-01:
Ok

Sent by from my Smartphone.  

Am 1. Oktober 2023, 16:31, schrieb knickohr:

Versuch mal die DTU in den AP-Modus zu bringen und dann den Update zu machen.

kiu77, 2023-10-01:
Zeigt er auch auf dem Display die alte Version an oder nur im Browser?
Bei letzterem dann ein reload der Seite machen.

persil14, 2023-10-02:
Display hab ich nicht. Zeigt im Browser die Version an. Aber auch vom Smartphone Android 13 nicht möglich. Grüße Peter 

Sent by from my Smartphone.  

Am 1. Oktober 2023, 20:25, schrieb kiu77:

Zeigt er auch auf dem Display die alte Version an oder nur im Browser?
Bei letzterem dann ein reload der Seite machen.

knickohr, 2023-10-02:
Dann wird wohl nur noch ein Erase & Flash mit den entsprechenden Tools über die USB-Schnittstelle möglich sein.

elisenjens, 2023-10-02:
Bei mir das gleiche Problem; OTA funzt nicht mehr; für USB Flash muss ich jedes Mal umbauen; wird OTA wieder zum Leben erweckt für 8266 oder muss ich nun die Hardware upgraden?

lumapu, 2023-10-02:
hab eine Idee hierzu, evtl. könntest du testweise mal MqTT deaktivieren (IP von Broker löschen). Haben hier neulich was umgebaut, nicht dass das hi l in deinem Fall nötig wäre.

elisenjens, 2023-10-02:
Bin erst morgen wieder zuhause; mach ich dann prompt.

elisenjens, 2023-10-04:
Habe jetzt die die release-Version mit Erase installiert; dann hat das update auf 0.7.65 funktioniert: Warte jetzt auf Morgen wegen der Reset der Max-Werte...

elisenjens, 2023-10-05:
Die MAX-Werte sind über Nacht gelöscht; mit der 0.7.65 komme ich heute aber nicht in die Live-Anzeige??? Nach Restart funzt es...

lumapu, 2023-10-05:
die Max-Werte werden um 00:00 Uhr zurückgesetzt sobald der Haken dafür gesetzt ist. Hier sieht es nach einem heap Problem beim ESP8266 aus. Wie ist denn die `heap-frag`?

elisenjens, 2023-10-05:
Muss ich morgen checken; jetzt liefert AHOY nix mehr

elisenjens, 2023-10-06:
Heute das gleiche Bild: komme nicht mehr in die Anzeigen rein; kann deshalb auch nicht sagen, wie die heap-fraq zu dem Zeitpunkt aussieht; nach einem Reboot sieht es dann so aus:

Radio NRF

NRF24L01

connected

NRF24 Power Level

MIN (recommended)

NRF24 Data Rate

250 kbps

MqTT

connected

true

#TX

126

#RX

0

System Information

mac

98:CD:AC:1C:DE:17

sdk

2.2.2-dev(38a443e)

cpu freq

80

heap free

15424

sketch used

566

wifi rssi

-14

ts uptime

103

version

0.7.65

esp type

ESP8266

heap frag

24

max free_blk

10960

core version

3.1.2

flash size

4096

reboot reason

External System

Sun

Sunrise

6.10.2023, 07:29:33

Sunset

6.10.2023, 18:47:04

Communication start

6.10.2023, 07:29:33

Communication stop

6.10.2023, 18:47:04

Night behaviour

not communicating

elisenjens, 2023-10-06:
Heap fraq pendelt sich jetzt bei 24 ein

elisenjens, 2023-10-06:
heap fraq nach 3 Std Laufzeit 25

lumapu, 2023-10-06:
das ist komisch, die sollte wie bei mir relativ gering sein, habe nach 1 Tag und 17 Stunden eine `heap-frag` von 2 (auch auf einem ESP8266, nur dieser zeigt die fragmentation überhaupt an).

[image]

elisenjens, 2023-10-06:
Du bist auch auf 66…

 

 <https://user-images.githubusercontent.com/22733082/273235818-73497cdf-7f6c-4436-b7af-a6ec48670efb.png>

lumapu, 2023-10-06:
das macht hier keinen Unterschied

elisenjens, 2023-10-06:
hab jetzt mit der .66 einen heap frag von 24 nach reboot

lumapu, 2023-10-07:
kannst du mir einen Export deiner Einstellungen an contact@ahoydtu.de schicken?

elisenjens, 2023-10-07:
Moin, heute Morgen ist der heap frag wieder 24, aber Ahoy funzt; ich komme in alle Optionen ohne Reboot. Trotzdem anbei die Einstellungen.

Viele Grüße Jens

elisenjens, 2023-10-11:
Ein paar Tage lief es; heute kommen die Menüpunkte wieder nicht. Nach Neustart funzt es wieder und der Heap Frag = 26. Hat mein Export geholfen?

elisenjens, 2023-10-15:
Auch heute funzt es nicht Heap Frag = 26; 25 scheint die magische Grenze zu sein; wird es in der 0.8 dazu Verbesserungen geben? Ich weiß nicht, ob ich da weiter testen soll...

elisenjens, 2023-10-15:
meine Einstellungen haben keine Erkenntnisse gebracht?

lumapu, 2023-10-15:
habe bis jetzt nichts bekommen 🤔. Könntest du es nochmal versuchen?

elisenjens, 2023-10-15:
Na klar, hier nochmal in Anlage meine JSON

elisenjens, 2023-10-15:
JSON per Mail-Antwort angekommen?

elisenjens, 2023-10-15:
das Beantworten der Mails mit Anhang funzt wohl nicht; deshalb hier nochmal meine JSON
[2023-10-15_15-47-17_v0.7.66_ahoy_setup.json](https://github.com/lumapu/ahoy/files/12909723/2023-10-15_15-47-17_v0.7.66_ahoy_setup.json)

elisenjens, 2023-10-17:
@Lumapu: gibts Erkenntnisse aus meiner Konfig?

lumapu, 2024-01-05:
sorry, ich habe nie in die Config geschaut. Könntest du nochmal die aktuelle Release-Version testen?

elisenjens, 2024-01-05:
Update funzt in 0.8.21 OTA, aber morgens ist die DTU offline; muss nachts immer einen Reboot machen. War in den Vorversionen nicht so

lumapu, 2024-01-05:
das ist aber nicht die Release Version

closed 2024-02-17 by lumapu