lumapu/ahoy #1202: Several alarms „DTU command failed“

lille2000, 2023-10-02:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

O.7.63

### Github Hash

1bbe979

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

ESP8266 with two inverter (HM-600 and -350), e-ink display 

### Error description

After updating to 0.7.64 I had several alarms (~10 within one hour) inside inverter HM-350 which were never observed before „ DTU command failed”

lille2000, 2023-10-02:
[image]
[image]

Ollipop030, 2023-10-02:
Ist auch unter 0.7.63 der Fall. Der Inverter ist erst 3 Stunden wach, hab schon 23 Alarme "DTU command failed".

knickohr, 2023-10-02:
Ist bekannt. Bis man genau weiß was der Fehler aussagt, müssen wir es erst mal akzeptieren. Und ein Bug ist es auf jeden Fall nicht, da die Meldung direkt vom Inverter kommt und somit uns was sagen will.

lumapu, 2023-10-02:
habt ihr eine Art von Limitierung aktiv? Da ist bekannt, dass das der Fehler aus dem inverter kommt wenn man (ahoy) zu schnell den Wert zurückliest

knickohr, 2023-10-02:
Ja und nein. Es kommt bei meine, HM-300ern sowohl bei dem limitierten als auch von den unlimitierten.

MartinBachmannHD, 2023-10-02:
Ich hatte es auch einmal die letzten Tage im unlimitierten HM-600... 
Ob Ahoy aus versehen in einer der letzten Versionen doch 'irgend etwas' an den Inverter schickt, was der dann prompt nicht versteht?

Ollipop030, 2023-10-02:
> habt ihr eine Art von Limitierung aktiv?

Ja in der Tat. Allerdings hatte ich bisher nie so viele Alarme. Bei den ganzen 0.7.5er Releases hatte ich vielleicht 8 Alarme den ganzen Tag, mit dem selben Zero Export. Habe heute nach 5 Stunden schon 39.

lille2000, 2023-10-02:
> habt ihr eine Art von Limitierung aktiv? Da ist bekannt, dass das der Fehler aus dem inverter kommt wenn man (ahoy) zu schnell den Wert zurückliest

Nein, bei mir ist nichts limitiert. Es ist auch definitiv erst seit 0.7.63 so.

reserve85, 2023-10-02:
Bei mir ähnliches Phänomen, die ganze Kommunikation ist sehr schlecht geworden? RX no answer und RX Fail schießen in die Höhe seit Update von .61 zu .65.
ZeroExport ist bei mir aktuell deaktiviert.

Ich kann nachher noch Bilder liefern, bin gerade unterwegs.

lumapu, 2023-10-02:
ich sehe bei mir den Fehler auch, allerdings nur bei den inverter, bei den ich heute Morgen das neue Power-Limit-Popup getestet habe.
Habe einmal auf 99% limitiert, dann wieder auf 100%.
Die Fehler sind aber erst heute Mittag im Fehlerspeicher - echt komisch.

Ollipop030, 2023-10-02:
Seit knapp 3 Stunden läuft bei mir wieder 0.7.59 mit laufendem ZeroExport. Beide Inverter seit dem kein Alarm mehr.

BastyESP, 2023-10-02:
[image]

Mir geht das schon ein paar Versionen so, mal einen Tag fast nichts bzw. nur Start und 2-3, und heute von 3 Invertern "947+717+156" Alarme also mal eben knapp über 1800....

Zu dem was ich sehen kann als Alarm ist es faktisch nur die die Zwei, frage aber auch mit 2 DTU die Wechselrichter ab, bzw nur 2 Wechselricheter mit Open und 3 mit Ahoy....trotzdem müsste es ja dann Jeden Tag auftreten...ist aber nicht so.

knickohr, 2023-10-02:
Es wäre schön, wenn ihr alle eure ahW dazuschreibt, also ob ESP8266 oder ESP32 damit man das Problem, ich nenne es absichtlich nicht Bug, besser eingrenzen kann.

Als Beispiel :  Ich habe 13 Inverter an einem ESP32S3 (Fusion-Board) hängen und habe für heute genau 13 Fehler, alle lauten „Inverter Start“ 😉

BastyESP, 2023-10-02:
ESP32 (30-PIN) 
0.7.63
3 Inverter
2627 Alarme
Da ich nur die letzten 10 jeweils sehe, lauten alle 2 - DTU Command Failed

Besonderheit, Abfrage von 2 Invertern parallel mit Open DTU.

Hoffe das war richtig so.

You69Man, 2023-10-02:
Hatte seit der Früh weg die 0.7.63 auf einem ESP32 am laufen (1xHM800). Abgesehen von der Inverter Start Meldung sehe ich keine weiteren 'Alarme' in der Live View. Oder wo sollten die sein?

Ollipop030, 2023-10-02:
Bin auf dem ESP32. Habe gerade die Logs vom ZeroExport nochmal angesehen. Heute bis Mittag hatte ich ja die 7.63 drauf, da kam quasi im Minutentakt ein Timeout bei ackknowledge nach dem Limitsetzen. Danach lief ab kurz nach Mittag wieder 0.7.59. Da hatte ich diesen Fehler den ganzen Tag nur noch 1x.

lille2000, 2023-10-03:
ESP8266, 2 Inverter HM-600 und 350, Häufigkeit bei beiden ca. 10x , abwechselnd. Heute ist der andere betroffen als gestern, kein Power Limit

Vielleicht hilft der Trace. Es ist zwar ein Offset von  2 Stunden im Log, aber sie sind definitiv zeitgleich aufgenommen.

`
05:34:45 I: TX 27 CH61 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) [57 42 80 05 00 65](tel:57 42 80 05 00 65) 1b a7 f5 [00 00 00 00 00 00 00](tel:00 00 00 00 00 00 00) 00 60 df 4d 
05:34:45 I: TX 27 CH75 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) [57 42 80 11 00 65](tel:57 42 80 11 00 65) 1b a7 f5 [00 00 00 00 00 00 00](tel:00 00 00 00 00 00 00) 00 74 cb 59 
05:35:15 I: TX 27 CH3 | [15 82 85 46 62 88 94](tel:15 82 85 46 62 88 94) [57 42 80 05 00 65](tel:57 42 80 05 00 65) 1b a8 [13 00 00 00 00 00 00](tel:13 00 00 00 00 00 00) 00 00 34 0d 46 
05:35:15 W: (#0) nothing received: complete retransmit
05:35:15 I: (#0) prepareDevInformCmd 0x11
05:35:15 I: TX 27 CH23 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) [57 42 80 11 00 65](tel:57 42 80 11 00 65) 1b a7 f5 [00 00 00 03 00 00 00](tel:00 00 00 03 00 00 00) 00 74 8f 1e 
05:35:15 I: (#1) enqueCommand: 0x00
05:35:15 I: (#1) enqueCommand: 0x0b
05:35:15 I: TX 27 CH40 | [15 82 85 46 62 88 94](tel:15 82 85 46 62 88 94) [57 42 80 00 00 65](tel:57 42 80 00 00 65) 1b a8 [13 00 00 00 00 00 00](tel:13 00 00 00 00 00 00) [00 00 31 08 43](tel:00 00 31 08 43) 
05:35:45 I: (#0) enqueCommand: 0x11
05:35:45 I: (#0) enqueCommand: 0x0b
05:35:45 I: TX 27 CH61 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) [57 42 80 11 00 65](tel:57 42 80 11 00 65) 1b a8 [31 00 00 00 00 00 00](tel:31 00 00 00 00 00 00) [00 00 81 99 35](tel:00 00 81 99 35) 
05:35:45 I: TX 11 CH75 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) 57 42 83 d8 
05:35:45 W: (#1) nothing received
05:36:15 I: TX 27 CH3 | [15 82 85 46 62 88 94](tel:15 82 85 46 62 88 94) 57 42 80 0b 00 65 1b a8 4f [00 00 00 00 00 00 00](tel:00 00 00 00 00 00 00) 00 f9 ab 7f 
05:36:15 I: TX 11 CH23 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) 57 42 83 d8 
05:36:15 I: TX 11 CH40 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) 57 42 84 df 
05:36:15 I: TX 27 CH61 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) 57 42 80 0b 00 65 1b a8 4f [00 00 00 00 00 00 00](tel:00 00 00 00 00 00 00) 00 f9 ab 1b 
05:36:15 I: TX 11 CH75 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) 57 42 82 d9 
05:36:45 I: (#0) enqueCommand: 0x0b
05:36:45 I: TX 27 CH3 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) 57 42 80 0b 00 65 1b a8 6d [00 00 00 00 00 00 00](tel:00 00 00 00 00 00 00) 00 58 2b 18 
05:36:45 I: alarm ID incremented to 5
05:36:45 I: (#0) enqueCommand: 0x11
05:37:15 I: (#1) enqueCommand: 0x00
05:37:15 I: (#1) enqueCommand: 0x0b
05:37:15 I: TX 27 CH23 | [15 82 85 46 62 88 94](tel:15 82 85 46 62 88 94) [57 42 80 00 00 65](tel:57 42 80 00 00 65) 1b a8 8b [00 00 00 00 00 00 00](tel:00 00 00 00 00 00 00) 00 f7 c3 d6 
05:37:15 I: TX 27 CH40 | [15 82 85 46 62 88 94](tel:15 82 85 46 62 88 94) 57 42 80 0b 00 65 1b a8 8b [00 00 00 00 00 00 00](tel:00 00 00 00 00 00 00) 00 3c c9 1c 
05:37:15 I: TX 11 CH61 | [15 82 85 46 62 88 94](tel:15 82 85 46 62 88 94) 57 42 81 be 
05:37:45 I: TX 27 CH75 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) [57 42 80 11 00 65](tel:57 42 80 11 00 65) 1b a8 a9 [00 00 00 03 00 00 00](tel:00 00 00 03 00 00 00) 00 47 16 e7 
05:37:45 I: TX 11 CH3 | [15 82 85 46 62 88 94](tel:15 82 85 46 62 88 94) 57 42 81 be 
05:37:45 I: (#0) enqueCommand: 0x0b
05:37:45 I: TX 27 CH23 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) 57 42 80 0b 00 65 1b a8 a9 [00 00 00 00 00 00 00](tel:00 00 00 00 00 00 00) 00 9d 49 7b 
05:37:45 W: (#0) nothing received: complete retransmit
05:37:45 I: (#0) prepareDevInformCmd 0xb
05:37:45 I: TX 27 CH40 | [15 82 95 76 26 88 94](tel:15 82 95 76 26 88 94) 57 42 80 0b 00 65 1b a8 a9 [00 00 00 00 00 00 00](tel:00 00 00 00 00 00 00) 00 9d 49 7b
`

und die Alarme von inverter #0
[image]

BastyESP, 2023-10-03:
[image]
Bei mir gehts heute so weiter...bin schon bei >300

Ollipop030, 2023-10-03:
> An welcher Ressource solls denn liegen, ist der ESP32 in 6 Monaten, vollkommen verschlissen? Oder NRF Modul abgenutzt? Hab sogar den Kondensator damals nachgerüstet als ich das 2.42er ins Gehäuse verbaut habe und es auf Low gestellt...

Nein, wird wohl an Ahoy liegen. Versuche mal ein Downgrade auf 0.7.59, die Version hat auch heute nur den Alarm für den Inverterstart angezeigt. Wie @reserve85 schon angemerkt hat, ab der 0.7.60 ist wohl der Wurm drin.

Ollipop030, 2023-10-04:
Gerade spaßeshalber mal die 0.7.65 probiert. DTU Uptime ist jetzt bei knapp 15 Minuten. Inverter 1 zeigt nur den Startalarm, Inverter 2 zeigt knapp alle 30 Sekunden einen Alarm. Scheint, als ob 1 Inverter fehlerfrei funktioniert, ab zweien häufen sich dann die Fehler.

lille2000, 2023-10-04:
Bei mir sind diese „invalid“ Alarme immer zeitlich gruppiert. Wenn ein Alarm eingetragen ist, dann folgt immer ein zweiter Alarm nach 30s. Dann ist immer einer Pause/Lücke bis zum nächsten Alarmpäarchen. 

„alleine“ treten die Invalids nicht auf, auch nicht drei hintereinander.

Ollipop030, 2023-10-04:
Habe den zweiten Inverter wieder aktiviert und Ahoy neu gestartet. Komischerweise seit 2,5 Stunden keinen Alarm mehr gehabt. Ganz kurios. Bin gespannt, was dann morgen passiert.

reserve85, 2023-10-04:
Wie sieht denn bei dir die Radio Statistik aus? Bei mir ist mit der V0.7.65 fast kein RX-Paket mehr erfolgreich empfangen worden.

Ollipop030, 2023-10-04:
TX count	3366
RX success	1526
RX fail	10
RX no answer	550
RX fragments	9279
TX retransmits	2250

und 

TX count	3324
RX success	1978
RX fail	39
RX no answer	120
RX fragments	7017
TX retransmits	1162

Ist im Rahmen denke ich. Allerdings fällt auf, dass der Obere mehr als 4x so viele "no answer" hat als der Untere. Das ist auch der, der seit einer Stunde wieder Alarme ausgibt.

Ollipop030, 2023-10-04:
> Welcher Power Level ?

Minimum.

reserve85, 2023-10-04:
Ja beobachte das mal, bei mir konnte ich regelrecht zuschauen wie die fails hochzählten und gleichzeitig inverter errors auftraten.

BastyESP, 2023-10-04:
Also ich hab ja wie angedroht auf High gestellt gehabt, gebracht hats nichts. 

HM-800 Alarme 1199
HM-400 Alarme 1681
HM-300 Alarme 89

Irgendwie sollte man die Anzeige einfach nicht beachten, dann gehts.

lumapu, 2023-10-04:
kann mir einer (oder auch alle?) ein Log zukommen lassen, gerne auch an contact@ahoydtu.de
Würde mir das gerne mal ansehen, meine 3 inverter verhalten sich leider seit dem Report des Fehlers wie Musterknaben 😏

BastyESP, 2023-10-04:
@lumapu 
Kannst du mal grob was zu schreiben, wie lange oder was man da abgeben muss? 
Dann mach ich das morgen Nachmittag gerne mal.

lumapu, 2023-10-04:
das Log am besten aus der Webserial-Console kopieren, nachdem der Fehler `DTU command failed` gekommen ist

lumapu, 2023-10-04:
ich habe das Gefühl, dass es mit den Retransmits zu tun hat, bin mir aber nicht sicher. Hier muss ich an einer saubereren Kommunikation arbeiten, sodass jeder Inverter einzeln verarbeitet wird, bis klar ist, ob man einen Timeout hat oder noch Retransmitted. Bisher durchlaufen wir die loop über alle Inveter von vorne und senden alles raus was noch offen ist - ein bisschen kuddelmuddel - historisch bedingt.

lille2000, 2023-10-05:
Hier ist ein Debug-Log

[Logs_DTU_CommandFailed_075307.txt](https://github.com/lumapu/ahoy/files/12812869/Logs_DTU_CommandFailed_075307.txt)

Um 07:53 :xx waren drei "DTU Command failed" eingetragen (+ 1x "Inverter Start").
Seltsam ist: Um 07:59 springt der Zeitstempel plötzlich auf 6:00 zurück und die Alarmliste ist leer.  
um 8:07 sind wieder drei (!) Einträge drinnen (2x Fail, 1x Start)

Ollipop030, 2023-10-05:
Ich habe gestern und heute mal 0.7.65 und 0.7.59 verglichen. Beide Tests mit exakt gleichen Einstellungen und laufendem Zero Export.

ESP32, NRF mit externer Antenne, Kondensator verbaut. Power Level minimum, knapp 3 Meter Abstand und nur Dämmung dazwischen.

**0.7.65:**

Uptime: 0 Days, 07:53:05 (wobei die Alarme sich erst nach drei Stunden Uptime häuften)

HM-1500 Alarms: 45

```
TX count        6644
RX success	3968
RX fail		75
RX no answer	222
RX fragments	13953
TX retransmits	2171
```

HM-300 Alarms: 82

```
TX count	6718
RX success	3557
RX fail		27
RX no answer	688
RX fragments	13434
TX retransmits	2761
```

```
heap free	176968
heap total	294708

```
Manchmal mehrere Alarme in wenigen Minuten, dann wieder für längere Zeit kein Alarm.

**0.7.59:**

Uptime: 0 Days, 07:54:54

HM-1500 Alarms: 1 (Startalarm)

HM-300 Alarms: 1 (Startalarm)

```
TX count	13184
RX success	9106
RX fail		26
RX no answer	123
RX fragments	24788
TX retransmits	4825
```
```
heap free	169704
heap total	289188
```

Morgen versuche ich mal eine 0.7.60er, dann mit höherem Power Level.

BastyESP, 2023-10-05:
Also das was @lumapu fordert, wenn Fehler auftreten ist ja auch ein Geduldsspiel, da es mal wie schon beschrieben, es Minuten lang Fehler hagelt, und dann auch wieder lange ruhe ist...natürlich bewegt sich wenn man es anschaut, ewig nichts an der Alarm anzeige...hab es aber nun an 2 Alarmen hinbekommen, und schicke die Mail mal raus, wenn mehr benötigt wird, einfach was sagen, dann mach ich es morgen nochmal. Schicke erstmal die Mail raus...

lumapu, 2023-10-05:
Zuerst, @BastyESP hat scheinbar 3 Inverter, 2x 1-Kanalig und 1x 2-Kanalig. Meiner Analyse nach alle 3 HM-Wechselrichter, daher alles über das gleiche Funkmodul und kein Mischbetrieb.

beim Log fällt auf, dass kurz vor der Message ein Kuddel-Muddel ist:
* der Inverter 1 wird abgefragt
* keine Antwort kommt
* nach 10s wird ein kompletter Retransmit angefordert (mit gleichem Timestamp)
* ursprüngliche Nachricht kommt komplett an, Alarm ID bei 0x02ef
* Inverter 1 ist wieder an der Reihe
* Antwort kommt normal, mit der Alarm ID 0x02f1 (2 höher als vorher)

Wir müssen noch mehr Logs sehen, aber ich habe die Vermutung, dass ein Retransmit (evtl. auch nur der complete-Retransmit) nicht 10s später mit dem gleichen Timestamp nochmal erfolgen darf.

lumapu, 2023-10-05:
@BastyESP evtl. kannst du einfach nebenher die Webserial laufen lassen und bei Gelegenheit einfach alles per `STRG`+`C` rauskopieren und in einem Editor als Text-File speichern. Dann ist es kein Geduldsspiel mehr. Anschließend kannst du darin nach "alarm ID incremented to" suchen und die 50 Zeilen vorher und nachher mir schicken.

BastyESP, 2023-10-05:
@lumapu Hab noch viele Zeilen, denn das Notebook lief noch, und die Konsole war offen, allerdings hab ich alle ca 10sek eine abfrage, wo wieder kommt das es Nacht ist und keine Kommunikation läuft, wären wenn ich auf Seite suche wohl 74 Alarme...nur wie ich in 10 sek, alles markieren und wenigstens mal sichern kann, ist mir ein rätsel.
 Editor würde mir ja reichen...evtl solltet ihr dafür mal ne Export als "txt" anbieten...

Würde jetzt mal Anbieten, nach Alarm zu suchen, und soviel wie möglich in den 10sek die ich habe zu Kopieren, und dir das zu senden...wäre das OK? Also ich versuch es mal und schicks dir als Mail.

lumapu, 2023-10-05:
nur 904 😂
danke für die Mail. Nutze doch einfach die Tastenkombinationen in Windows: zuerst `STRG` + `A`, dann direkt `STRG` + `C`. Dann hast du alles im Zwischenspeicher und muss es im Editor der Wahl nur noch einfügen 😉
Wenn die das Scrollen stört kannst du einfach unter der Konsole auch auf `Auto-Scoll` klicken, dann steht dort `manual scroll` und du hast alle Zeit der Welt.

lille2000, 2023-10-05:
Ich teile die Vermutung, dass die Inverter 2 identische Requests (= Retransmit mit der selben Payload inkl. dem Zeitstempel) nicht mögen. Vermutlich wird die erste Nachricht empfangen und sogar beantwortet. Die Antwort geht vollständig verloren. Die DTU sendet die identische Nachricht nochmal und der Inverter lehnt die Nachricht als „verbraucht/bereits geantwortet“ ab und trägt einen Fehler ein. 

vielleicht sollte ein Retransmit die Zeitstempel  vor dem erneuten Senden aktualisieren.

BastyESP, 2023-10-05:
Der Tipp mit Strg+A war Spitze, damit ging es, hab es mir mal als txt gespeichert, aber hast ja erstmal wahrscheinlich Auswahl.
Das mit dem Manual scroll etc wollte ich alles nicht probieren...da die DTU wohl vor 2h abgekackt war (zumindest Uptime knapp über 2h, und ich nur von den Bestand im Browser gelebt habe...wenn der neu Läd sind alle Daten weg...

lumapu, 2023-10-05:
ja dieser Kuddel-Muddel muss weg. 

So muss es sein:
1. Inveter wird abgefragt
2. Antwort wird geprüft, bei fehlenden Nachrichten wird neu angefragt
3. Antworten werden geprüft, wenn vollständig -> ok, wenn nicht -> failed
4. nächster Inverter wird abgefragt

zwischen 1. und 3. sollte max. ca. eine Sekunde sein und nicht wie jetzt teilweise 10s.

@BastyESP super, ja dein letztes Log ist sehr umfangreich, das verdaue ich morgen 😉. Im Schnelldurchlauf habe ich gesehen, dass Fehlereinträge für alle 3 Inveter dabei sind, das ist perfekt.
Das Log ist nur im Browser gespeichert, nicht im ESP. Evtl. kannst du morgen mal paar Stunden auf die zweite DTU verzichten und nochmal für mich loggen.

Ollipop030, 2023-10-06:
Habe auch mal mitgeloggt. Lief knapp eine Stunde ohne Alarm. ESP32, 0.7.66

[Log.txt](https://github.com/lumapu/ahoy/files/12828160/Log.txt)

Ollipop030, 2023-10-06:
> Du hast scheinbar ein sehr kurzes Intervall. Was mir hier auffällt wäre die sofortige Abfrage des gesetzten Powerlimits schon nach einer Sekunde.

Das ist der wohl meiner Nulleinspeisung geschuldet, sieht dann so aus:
```
2023-10-06 10:13:52,146 INFO     Ahoy: Inverter "HM-1500": setting new limit from 397 Watt to 402 Watt
2023-10-06 10:13:52,829 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
2023-10-06 10:13:52,830 INFO     Ahoy: Inverter "HM-300": setting new limit from 79 Watt to 80 Watt
2023-10-06 10:13:53,514 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
```
Ist alles sehr schnell verarbeitet, war aber bisher unkritisch, siehe auch meinen Beitrag von gestern.

lumapu, 2023-10-06:
keine Kritik am eingestellten Intervall, ich will nur verstehen wie es zu dem Fehlercode kommen könnte. War eher selbstkritisch gemeint 😉

stefan123t, 2023-10-07:
> @Ollipop030 danke für dein Log. Du hast scheinbar ein sehr kurzes Intervall. Was mir hier auffällt wäre die sofortige Abfrage des gesetzten Powerlimits schon nach einer Sekunde. Evtl. sollte man hier die Zykluszeit beachten (Ahoy). Noch ein Hinweis bzgl. der SN: Wenn du sie unkenntlich machen willst, dann bitte auch bei allen TX nach der 15 die nächsten 4 Gruppen unkenntlich machen und bei RX identisch nach der 95. Evtl. sollten wir hier einen Feature-Request anlegen, der quasi privacy hier automatisch aktiviert.
> 
> z.B.
> 
> ```
> TX 27 CH3 | 15 ** ** ** ** 85 37 85 60 80 0b 00 65 1f b1 7b 00 00 00 00 00 00 00 00 b2 09 8c 
> ```

Es sollte noch die Inverter Nummer #1..16 erkennbar sein oder noch besser die Modell Nummer und das Produktionsdatum also die ersten sieben/acht Stellen der Seriennummer stattdessen verwenden ?

stefan123t, 2023-10-07:
> Zuerst, @BastyESP hat scheinbar 3 Inverter, 2x 1-Kanalig und 1x 2-Kanalig. Meiner Analyse nach alle 3 HM-Wechselrichter, daher alles über das gleiche Funkmodul und kein Mischbetrieb.
>
> beim Log fällt [...]

Wie kommt es dazu dass beim Retransmit nach 10 Sekunden der Timestamp immer noch gleich ist. Das würde für mich auch das Problem bei der Abfrage mit mehreren DTUs erklären, da ja beide DTUs ab und an mal eine Abfrage mit dem gleichen oder sogar einem “älteren” Timestamp als die andere DTU senden und dann der WR einen Protokollfehler mit der DTU vermerkt.
Ich versuche mich gerade zu erinnern die aller ersten Logs von der DTU WLite hatten auch retransmits drin aber mW nur von einzelnen Teilpaketen, oder waren da auch ganze Retransmits dabei ? Bei den Teilpaketen wurde der Timestamp mW nicht immer erhöht aber er war auch nicht niedriger als der vom WR (schickt der eigentlich einen außer in den Alarm Messages) … bei mir sind noch viele Fragezeichen. 

@BastyESP Ich hatte das in der Vergangenheit ja auch oft zu Hauf mit den (2) DTU command errors. Nach Umstellen auf LOW am NRF +PA/LNA ( das Ding mit der externen Antenne ) wurde es mit plötzlich viel besser. Davor hatte ich lange Zeit MIN, MAX und HIGH probiert, weil ich auf Sende/Empfangsprobleme getippt hatte: aber das war dann eher kontraproduktiv trotz Kondensator!

BastyESP, 2023-10-07:
Hab gestern komplett verpasst das aufzunehmen, und als ich wieder dran gedacht habe, war es dann Dunkel -.-

P.s. Uptime der OpenDTU sind mittlerweile 30 Tage, da komm ich mit Ahoy schon alleine weil ich die vielen Versionen Installiere nicht hin -.-
[image]

Bin ja erst hochgegangen auf LOW auf HIGH, um es zu versuchen, hat aber nichts verbessert, das ich auf MIN besser fahre, bezweifel ich beinahe, der HM-400, war eher empfangstechnisch schon mein Sorgenkind, hatte früher als ich nur die 2 (HM-800 + HM-400) Aktiv hatte, immer HIGH, weil ich auf LOW, sonst teilweise nur alle 0,5-1min vom HM-400 antworten erhalten habe, nach dem ich ein Gehäuse gedruckt hatte, und die Verhältnisse besser wurden (War vorher nur in so einem kleinem Shelly 1PM Karton reingestopft + der Ort wo es jetzt steht ist minimal höher), hab ich den Kondensator verbaut und LOW versucht, lief  bisher eigentlich bisher ganz gut, kann aber MIN gerne versuchen...will mich davor nicht streuben...hab eben nur die vorherigen Sorgen mit den Schlechten erreichbarkeiten schon gehabt.

Die beiden HM (800/400) liegen Quasi flach auf einem extra Blech, was sie abschirmt vor Sonnenstrahlung, auf einem Trapezblech Dach, das schränkt natürlich die Funkerei etwas ein, kann mal schauen auf was die OpenDTU funkt...die macht mir in den Sinne aber auch keinerlei sorgen, schaue selten ins Menü rein, und das Display (1,54" OLED) macht was es soll...

[image]

Würde ich jetzt mal als LOW übersetzen...

Mein Problem wäre jetzt auch einfach zu Lösen (Schalter womit man die Alarm anzeige ausschalten kann) ;) denn es läuft ja auch mit den Tausenden Alarmen...allerdings bin ich ja nicht der einzige der das hat, nur ich bin der einzige Pflegefall der nebenher noch ne zweit DTU mitlaufen lässt ;)

Ollipop030, 2023-10-07:
> Evtl. sollten wir hier einen Feature-Request anlegen, der quasi privacy hier automatisch aktiviert.

Erledigt: #1211

lumapu, 2023-10-09:
@stefan123t beim retransmit muss der alte timestamp geschickt werden, sonst passt der CRC nicht.
Ich glaube den kompletten retransmit haben wir eingeführt ist besser gesagt @Sprinterfreak glaube ich, der war damals uns allen ein gutes Stück voraus.

Falsch ist aus meiner Sicht der retransmit nach 10s, der muss viel eher kommen. Bin schon dran es zu reparieren.

lille2000, 2023-11-08:
Mit 0.8.1 habe ich nach einem Tag keine Alarme „DTU command failed” mehr gehabt. 
 Andere Auffälligkeiten habe ich nicht bemerkt.

lille2000, 2023-11-14:
Bisher nicht mehr aufgetreten, kann als gelöst geschlossen werden. 👍

closed 2023-11-30 by lille2000