lumapu/ahoy #1203: Merkwürdiges Verhalten der Web-Schnittstelle

kiu77, 2023-10-02:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.60

### Github Hash

963e618

### Build & Flash Method

was already installed

### Setup

Mit 2,42"-OLED-Display und eByte-Funkmodul (externe Antenne)

### Error description

Ich habe mit diesem Aufbau so ziemlich alle Versionen via OTA seit 0.6.9 drauf gemacht, also upgedated. 
2 Inverter waren eingetragen.
Die Version 0.7.60 hatte ich seit ein paar Tagen drauf, ohne Probleme bisher.
Dann gestern war die integrierte Webseite nicht mehr bedienbar, da extremst träge.
Neustart und neu Booten hat nichts gebracht.
Nur ein neues Flashen mit dem Online-Flasher der AhoyDTU-Webseite inklusive Erase hat es gebracht.

Die Sache ist natürlich nicht reproduzierbar, aber irgendwas muss da intern im ESP32 passiert sein
und ich wollte nur darauf hinweisen, dass das wohl unter Umständen mal passieren kann.

Eine andere DTU bei meiner Freundin läuft schon seit 4 Monaten non stop problemlos unter 0.6.x

rnwgnr, 2023-10-02:
Schau mal in #1194 - die .60 hatte ein Problem auf dem ESP32.

kiu77, 2023-10-02:
Ja, das wird es genau gewesen sein. 
Also kann man den thread schließen @lumapu 
Danke @rnwgnr

rnwgnr, 2023-10-02:
Als Ersteller solltest du das Issue auch wieder schließen können

kiu77, 2023-10-02:
Closed, da Problemlösung klar...

closed 2023-10-02 by kiu77