lumapu/ahoy #1204: Zusatz: DTU sendet/empfängt sporadisch 10 min. lang keine Signale vom WR

Gubi2023, 2023-10-02:
Zusatz: DTU sendet/empfängt sporadisch 10 min. lang keine Signale vom WR
[image]

_Originally posted by @Gubi2023 in https://github.com/lumapu/ahoy/issues/1178#issuecomment-1735205119_

Gubi2023, 2023-10-02:
Das Problem tritt mit 7.64 immer noch auf.

Gubi2023, 2023-10-02:
[image]

knickohr, 2023-10-02:
Bitte auch hier etwas zur HW dazu schreiben. Es wäre schön, wenn ihr auch den einleitenden Fragebogen ausfüllt. So ist es nur Stochern im Nebel und man sieht eventuell das Problem nicht. Danke !

rmayergfx, 2023-10-04:
> I just tested with ESP-8266, new installed 0.7.65 and HM-inverter is working.

Not useful! Didn’t you read the comment from @knickohr?
We need all information about the used hardware and inverter for the checklist!

technics42, 2023-10-04:
I wrote I am using ESP-8266 and using HM-inverter. (HM-type not HMS-type)
What else "HW" information do you need? ( I don't think there is another option than NRF24L01+ to connect to inverters)

I would say, my information is useful.
Just saying "we need **all** information" is pretty useless.
(If you just reply, you don't get any checklist.)

lumapu, 2023-10-04:
hier wäre hilfreich wie lange tatsächlich das Log analysiert wurde, @Gubi2023 du schreibst für 10 Minuten kein Signal, aber wie häufig tritt das auf? Jede Stunde / 2x Tag / etc.?

Gubi2023, 2023-10-05:
ich habe das Verhalten jetzt nochmals genauer beobachtet. Die DTU startet immer wieder ohne ersichtlichen Anlass neu, dann sind im Webserial 10 Min keine Daten. Starte ich das Webserial in einem neuen Fenster, läuft die Kommunikation sofort weiter. Mir ist nur die Ursache des Neustarts nicht klar. Manchmal alle 30 min dann läuft sie wieder 14 Std am Stück.... Könnte es am fehlenden Kondendsator liegen?

knickohr, 2023-10-05:
Ach jetzt !

Der Aussetzer kommt nach einem Reboot.

lumapu, 2023-10-05:
Hier haben wir ein anderes Problem, wie @knickohr schon schreibt. @Gubi2023 wie ist dein `heap-frag` Wert? ist er kleiner als 10?
Hast du @knickohr seine goldenen Ahoy-Regeln (Kondensator, etc.) beachtet?

Gubi2023, 2023-10-06:
Hast du @knickohr seine goldenen Ahoy-Regeln (Kondensator, etc.) beachtet?
wo finde ich die?
 @Gubi2023 wie ist dein heap-frag Wert? ist er kleiner als 10? unterschiedlich: mal auf 12, mal auf 2.
das ärgerliche ist, dass jedesmal beim reboot die ganzen max-Werte gelöscht werden. war das schon immer so?

Gubi2023, 2023-10-06:
[image]

dtuuser, 2023-10-12:
Platform
ESP32-D0WD-V3

nRF24L01+ plus (powerlevel Low)

Antenna
external antenna

Power Stabilization
Elko (~100uF)

Eigenes Netzteil 500mA

Heute nochmal in Ruhe die neueste Version eingespielt. Nach Update und Reboot kommen die Werte. Dann 10 Minuten kein Empfang. Nach 10 Minuten dann wieder Werte. Usw.
Mit der 60er Version kein Problem. Empfang topp, keine Empfangslücken

knickohr, 2023-10-12:
Ist der Serial Debug aktiviert ?  Wenn ja, schalte den mal aus ob es dann besser wird.

dtuuser, 2023-10-12:
Nein, Serial Debug ist aus

dtuuser, 2023-10-15:
Vielleicht hilft das weiter....oder ich interpretiere das falsch

Im Webserial Fenster erscheinen manchmal mehrere Abrufversuche innerhalb einer Sekunde

13:31:28 I: TX 11 CH61 | 15 80 40 51 95 89 95 62 48 82 a5 
13:31:28 I: TX 11 CH75 | 15 81 86 70 88 89 95 62 48 82 5e 
13:31:28 I: TX 11 CH3 | 15 80 40 51 95 89 95 62 48 82 a5 
13:31:28 I: TX 11 CH23 | 15 81 86 70 88 89 95 62 48 82 5e 

Und beim Alarm erscheint dann umgehend ein weiterer Eintrag 

DTU command failed 2

dtuuser, 2023-10-16:
Update
Habe gestern mal den ESP komplett gelöscht und die 66 direkt installiert. 
Problem bleibt bestehen

lumapu, 2023-11-18:
bitte nochmal mit aktueller `0.8.x` testen

Gubi2023, 2023-11-20:
> ich habe das Verhalten jetzt nochmals genauer beobachtet. Die DTU startet immer wieder ohne ersichtlichen Anlass neu, dann sind im Webserial 10 Min keine Daten und der Zeitstempel ist anschliessend um 1 Std. falsch. Starte ich das Webserial in einem neuen Fenster, läuft die Kommunikation sofort weiter mit richtigem Zeitstempel. Mir ist nur die Ursache des Neustarts nicht klar. Manchmal alle 30 min dann läuft sie wieder 14 Std am Stück.... Könnte es am fehlenden Kondensator liegen?

Dieses Verhalten ist auch noch mit V8.10 zu beobachten.

rmayergfx, 2023-11-20:
@Gubi2023 
Welche Hardware nutzt du? Ist ein Elko verbaut, kann irgendwie keine Info von deinem System im Thread sehen.
Hast du denn mal ein anderes Netzteil ausprobiert. Hatten die Tage erst das Problem das ein defektes Netzteil solche Fehler produziert hatte. Sporadische Reboots hatte ich bisher noch keine.

dtuuser, 2023-11-20:
Ich bekomme das hier im Webserial, obwohl nach Sonnenuntergang keine Abfragen mehr laufen sollten

18:58:21 I: heuristic test mode
18:58:21 I: (#0) request timeout: 500ms
18:58:41 I: heuristic test mode
18:58:41 I: (#0) request timeout: 500ms

knickohr, 2023-11-20:
Du hast die Optionen bei den Invertereinstellungen nicht gesetzt. Das ganze ist jetzt unter Settings - Inverter - Edit zu finden 😉

Gubi2023, 2024-01-25:
> > ich habe das Verhalten jetzt nochmals genauer beobachtet. Die DTU startet immer wieder ohne ersichtlichen Anlass neu, dann sind im Webserial 10 Min keine Daten und der Zeitstempel ist anschliessend um 1 Std. falsch. Starte ich das Webserial in einem neuen Fenster, läuft die Kommunikation sofort weiter mit richtigem Zeitstempel. Mir ist nur die Ursache des Neustarts nicht klar. Manchmal alle 30 min dann läuft sie wieder 14 Std am Stück.... Könnte es am fehlenden Kondensator liegen?
> 
> Dieses Verhalten ist auch noch mit V8.65 zu beobachten. V8.59 lief da stabiler!

Gubi2023, 2024-02-21:
Leider auch noch mit Release 8.83 immer wieder Neustarts der DTU ohne ersichtlichen Anlass. Manchmal alle 20 min, dann läuft sie wieder 2 Std am Stück. Das ärgerliche ist, dass jedesmal beim reboot die ganzen max-Werte gelöscht werden. war das schon immer so bzw. kann man die irgendwie wiederherstellen??

Gubi2023, 2024-03-06:
Hallo, liest das überhaupt noch jemand oder muss ich einen neuen Issue öffnen?
heute auf V0.8.91 upgedatet, seither wieder ständige Neustarts der DTU. Mit 0.8.89 lief sie 2 Tage durch!

Gubi2023, 2024-03-16:
und wieder mal: 
V0.8.93 sehr stabil,
V 0.8.94 DTU-Neustarts im Minutentakt!!

lumapu, 2024-03-16:
dann bitte ein serielles Log, damit ich nachvollziehen kann wo es hakt.

Gubi2023, 2024-03-16:
[0.8.94 - 6af07f5.txt](https://github.com/lumapu/ahoy/files/14623375/0.8.94.-.6af07f5.txt)

Gubi2023, 2024-03-16:
ESP8266, kein Display, kein Mqtt, 240316_ahoy_0.8.94_6af07f5_esp8266-all.bin geladen

lumapu, 2024-03-16:
ich meine ein USB Log, sorry. Dein Log ist wenig aussagekräftig.
Ich würde dir im allgemeinen empfehlen das Log nicht mit unnötigen Informationen zu fluten.
So wurde ich es einstellen:

[image]

Gubi2023, 2024-03-16:
tut mir leid, ich bin nur ein "dummer" Anwender, mit USB-Logs kann ich nicht dienen...

Gubi2023, 2024-03-23:
ESP8266, kein Display, kein Mqtt,:  ab V0.8.94 bis 8.97 habe ich ständige Neustarts der DTU im Minutentakt, egal welche FW geladen wurde (.min  -all). V8.93 läuft halbwegs stabil.
und nein, USB-Logs kann ich keine machen. (weiss nicht, wie das gehen soll)
Wurde ab da was grundlegendes geändert?

lumapu, 2024-03-23:
https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md

-> neuer Webserver ab 0.8.94

Gubi2023, 2024-03-23:
na schön, dann bleibe ich eben auf 0.8.93, wenn mein Webserver das nicht ab kann.

Gubi2023, 2024-03-23:
oder gibt es eine Abhilfe??

Gubi2023, 2024-04-25:
tja, auch mit V0.8.112 habe ich immer noch sporadische unerklärliche Neustarts der DTU, obwohl  ich inzwischen ESP32 verwende.... (V0.8.97 lief problemlos durch)

Gubi2023, 2024-04-25:
@rmayergfx:
ich habe jetz ein neues Gerät im Einsatz:  AllianceApps OpenDTU mit ESP32 (AZ-Delivery ESP32 NodeMCU Module WLAN WiFi Development Board mit CP2102)EBYTE NRF24L01+ Funkmodul mit Antenne
Adapter-PCB mit Kondesator 16V 100 uF
Status-LEDs sind auf default

rmayergfx, 2024-04-25:
Somit sind wir nicht mehr bei einem Standard ESP32 sondern du musst auch das entsprechende Paket für OpenDTU nehmen:
ahoy_v0.8.112\ESP32-S3 240424_ahoy_0.8.112_b291ad1_opendtufusion-de.bin
Wie soll man da helfen wenn du die Boards tauschst wie deine Unterhosen :-)

Spaß beiseite. Das richtige BIN einspielen und erneut testen.
LED bitte alle auf OFF stellen!
Welche Sendeleistung steht beim Modul in den Settings vom WR?

Gubi2023, 2024-04-25:
@rmayergfx : 
jetzt verwirrst du mich aber: Ich hab die DTU seit 2 Wochen im Einsatz und immer die _esp32-wroom32-de.bin FW verwendet (auf Empfehlung von metachuh), lief auch immer halbwegs. Z.Zt. hab ich V103 drauf, mit der gehts.  LEDs sind alle aus, Sendeleistung 100%, WLAN geht doch nur 2,4 GHz!

rmayergfx, 2024-04-25:
@Gubi2023 
Dann frag doch mal bei metachuh nach, er hat dir das Board ja empfohlen.
Welches Board hast du denn nun gekauft? Bei  AllianceApps  gibt es ja 3 verschiedene Platinen:
AllianceApps OpenDTU
AllianceApps OpenDTU Fusion
OpenDTU Fusion - Community Edition

Gubi2023, 2024-04-25:
wie geschrieben: AllianceApps OpenDTU. Wie gesagt: Mit V103 geht alles! Warum spielen dann auf einmal die anderen Faktoren eine Rolle?

rmayergfx, 2024-04-26:
Die o.g. Faktoren haben schon immer eine Rolle gespielt. Elko, Netzwerkstörungen durch andere WLAN Geräte, Netzteile das defekt ist oder zu geringe Leistung. Daran wird sich auch in Zukunft nichts ändern.
Auch die 103 ist eine Entwicklerversion. Bleib doch einfach auf der für dich passenden Version und warte ab. Warum möchtest du immer die aktuellste dev auf dem System haben, wenn eine Version für deine Umgebung stabil läuft, dann bleib doch auch bei dieser.

@lumapu 
Zwecks der richtigen Binary, welche ist für das normale AllianceApps OpenDTU am besten geeignet?
So wie ich das sehe ist das ja nur eine Platine und die Standard Bauteile, da sollte das \ESP32\xxxx_esp32-wroom32-de.bin 
doch eigentlich laufen.

rmayergfx, 2024-04-26:
Bitte jetzt nicht so argumentieren. Für Feedback ist man jederzeit dankbar, aber dann bitte auch mmit den entsprechenden Eckdaten. Wenn du dieses Issue hier selbst mal von Anfang an durchliest, wirst du feststellen, das du nun 3x die Hardware getauscht hast und leider auf einige Fragen überhaupt keine Antwort gegeben hast. Wie soll man dann als Entwickler einen Fehler eingrenzen?

Gubi2023, 2024-04-26:
@rmayergfx 
können wir diese Diskussion irgendwo über PN weiterführen?

rmayergfx, 2024-04-26:
PN Sind nicht öffentlich und somit für Dritte nicht einsehbar, das ist auch gut so, aber für einen Issue vollkommen irrelevant. 
Aus meiner Sicht ist das Thema soweit abgeschlossen, die Release 113 läuft bei dir stabil, also kann aus meiner Sicht hier abgeschlossen werden, wüsste auch nicht was ich noch per PN diskutieren sollte.

Gubi2023, 2024-04-26:
@rmayergfx
Ich hätte dir gerne noch mein Sicht zu dem Issue mitgeteilt, da da etwas aus dem Ruder läuft. 
Im Übrigen ist das Problem mit den sporadischen Neustarts bei mir immer noch vorhanden, sprich ich lass den Issue weiterhin offen

Gubi2023, 2024-04-26:
tja, sporadisch heisst sporadisch; kaum hatte ich ich es geschrieben, ist die DTU wieder ausgestiegen... (Nach 16 Std)
Und solange sie nicht mehrere Tage durchläuft, ist es für mich nicht ok.

closed 2024-05-10 by Gubi2023