lumapu/ahoy #1207: Ich habe einen Hoymiles HM-300 und die Ahoy DTU installiert. Version 7.36.  In der Eingabemaske für den Inverter kann ich nirgends die Leistung anpassen. Kann mir jemand helfen?n

Giverny1991, 2023-10-05:
## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

rejoe2, 2023-10-05:
Sollte in der aktuellen dev bereits gelöst sein.

Bitte vor dem Öffnen eines issues auch mal die Changelog ansehen

knickohr, 2023-10-05:
Um Bitte :  Die Titelzeile ist eine Zeile und kein Freitextfeld 😞

lumapu, 2023-10-05:
Die Leistungsbegrenzung ist in Version `0.7.36` unter `Serial / Control` zu finden.

closed 2023-12-29 by lumapu