lumapu/ahoy #1209: 0.7.66 mach meine DTU quasi unansprechbar träge [Bug]

kiu77, 2023-10-06:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.66

### Github Hash

1234567

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Aufbau auf eigener Platine, die zuverlässig läuft.

### Error description

Als ich heute OTA auf 0.7.66 upgedatet habe, war direkt nach dem Klick auf Update kein Bedienen der Web-Oberfläche der DTU mehr möglich.
Ich konnte mich nicht mehr einloggen. Die Live-Ansicht kam irgendwann später, aber mit Aussetzern, und selbst das Display ging zwischendurch aus. 
Ich habe dann neu geflachst mit Erase auf 0.7.36 und die Setting eingetragen - ging. Zweiter Versuch auf 0.7.66 - hing wie zuvor.
Also neu geflasht und dann nur auf 0.7.65 upgedatet. Da läuft stabil und ist responsiv.

Karlo49, 2023-10-07:
Kann ich bei meiner ESP8266 bestätigen.
Mit der 0.7.66 sehr Träge. Anzeigeänderungen kommen erst nach ca 5 sec.
Auch mit der 0.7.65 Probleme mit RX fail. ca 10 pro Stunde und heap frag auf 9!
Zurück auf 0.7.60 RX fail über 24 Stunden auf 0. heap frag auf 2 bis 3. Erst nach 2 bis 3 Tagen steigt der heap frag.
Alles mit der 3,3 Volt Filterung aus #1083

Joh49, 2023-10-07:
Mit ESP32 bei mir das gleiche.
Anzeigeänderung kommt wenn man sie nicht mehr erwartet.
Zurück auf 0.7.59 und es läuft wieder.
Nur mit Display ohne weiteren Anhang.

Riker18, 2023-10-08:
Das Problem hab ich heute auch mit einem ESP32 - hab schon neu geflasht mit Erase aber die Weboberfläche ist fast nicht erreichbar.  In Verbindung mit CMT ist das - ob es an den Änderungen liegt die noch nicht alle drin sind?  Ich bekomme OTA auch kein Downgrade geflasht

kiu77, 2023-10-08:
@Riker18 
Ja, da muss man wieder neu flashen mit einem Flash-Tool oder via AhoyDTU.de.

theunknownstarman, 2023-10-11:
Auf der 8266 Plattform bisher keine Fehler feststellbar. 
Bei der ESP32 führt das Update auf diese Version zu einem permanenten Ausfall, der sich bei mir insbes. bei MQTT Kommunikation zeigt (Siehe Bild - Update auf diese Version ca. 11:30 Uhr. Um ca. 15:30 Uhr wieder Rücksprung auf die Version 7.36)

[image]

KoLSMS, 2023-10-13:
Hallo,

ich kann nun nicht 100 % sagen das ein Zusammenhang besteht, aber ich hatte ähnliche Probleme mit der DTU...

Als ich den WLAN Kanal von 9 auf 8 geändert habe war das Problem mit der trägen bis kaum ansprechbarer DTU tatsächlich direkt gelöst...

Vielleicht probiert ihr das auch mal?!

Gruß...

kiu77, 2023-10-13:
@KoLSMS 
So nett der Workaround auch klingt, eine richtige Lösung ist das nicht, da einige Leute auf eine dynamische Kanalnutzung angewiesen sind.

lumapu, 2023-10-15:
das kann nur ein Zufall sein, kann mir nicht vorstellen, dass der Kanal Auswirkungen auf die Stabilität hat.

KoLSMS, 2023-10-15:
> das kann nur ein Zufall sein, kann mir nicht vorstellen, dass der Kanal Auswirkungen auf die Stabilität hat.

Also bei mir lokal hat es geholfen, wenn auch fast auszuschließen das es daran liegt...

Gruß...

kiu77, 2023-10-15:
Persönliche Erfahrung mit über 30 Stück ESP32:
Manchmal klappt es nicht so richtig mit der WLAN-Verbindung. Ich kann sogar sehen, dass der WLAN-Pegel des SoCs schwach ist (bei 30cm Distanz zum Rechner). Prinzipiell funktioniert die Sache dann nur so halbwegs.
In diesen Fällen hilft merkwürdigerweise nur das neue Flashen mit Erase - warum auch immer.
Aber ich hab längst aufgehört mich zu wundern, dass digitale Sachen durchaus gelegentlich analoges Verhalten zeigen ;-)

Senderti, 2023-11-05:
Ich habe einen ESP32 und auch bei mir wurde die WebUI mit 0.7.66 auch unfassbar träge und unzuverlässig, teilweise sogar Auswirkungen auf die Nachrichtenverarbeitung. Ich habe dann zur funktionierenden 0.7.65 gegriffen und die läuft auf meinem ESP32 wunderbar.

Senderti, 2023-11-12:
Mit den aktuellen 0.8er builds habe ich keine Probleme mehr.

theunknownstarman, 2023-11-12:
... ich schon. Bin wieder auf die Stable Version zurück.
Bei mir brechen sowohl bei der ESP32 als auch bei der 8266 Plattform die MQTT Verbindungen nach einiger Zeit ab.

kiu77, 2023-11-12:
MQTT nutze ich (noch) nicht, daher klappte es bei mit mit der neuesten Version auf ESP32 gut.

closed 2023-12-29 by lumapu