lumapu/ahoy #1210: Power Limit of MPPT Channel

Stefan1357, 2023-10-07:
## Hardware
  * ESP32

Modelname: Tsun TSOL M800
Retailer URL: https://www.alternate.de/

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* circuit board

### Power Stabilization:
* [x] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.7.36
Github Hash: ba218ed :: 0.7.36

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

I use am TSUN TSOL M800 and connected three 425W modules. One on channel 1, and two separated with one blocking diode each, to channel 2. The first module is pointing to south, the second to west and the third to east. I set the power limit to 133%. Therfore I should get at least 700W maximum on AC. But I see, that the converter limits the input power of both chabnels to 320W.
Does anyone know whether this limit can be adjusted by software or is it a hardware limit? 

[image]
[image]
[image]

Solarteur, 2023-10-07:
As has been stated here many times, the power of the inverter cannot be increased beyond the nominal limit (=100%). 

In my opinion, the observed limit of 320W is due to the seasonal reduced solar radiation.

Stefan1357, 2023-10-07:
I live in Germany and just realized, that the converter is a M800 (DE) version. In Germany the power is by law limited to 600W. I assume that therfore the firmware has a limit of 600W or 320W per channel. I hoped, I can extend the power to 800W or at least to 700W with ahoy DTU.

I'm sure, this limit is not due to season or limited radiation. That's why I attached the screenshots.

Stefan1357, 2023-10-07:
Okay, I found another commend, where they described the problem.

https://www.photovoltaikforum.com/thread/192651-hoymiles-hm600-700-800-gleiche-hardware/?pageNo=4

The total power of the converter (I'm my case 600W) is divided by two and this is the power limit of each MPPT. This is very annoying, but not an issue of Ahoy. Therfore I will close this issue.

Thanks

closed 2023-10-07 by Stefan1357