lumapu/ahoy #1211: Feature Request: Privacy Debug Mode

Ollipop030, 2023-10-07:
Wie in #1202 von @lumapu und @stefan123t schon angemerkt: Wäre vielleicht gut eine Möglichkeit zu haben, im Log die Seriennummern unkenntlich zu machen. Ein paar Zahlen sollten da evtl. aber erhalten bleiben, um den WR noch identifizieren zu können.

Noch eine Idee:
Da der ESP ja keinerlei Speichermöglichkeit für Logdateien hat, könnte man nicht über eine SMB Freigabe in eine TXT schreiben? Man bräuchte hier ein Feld für den Pfad (\\Nas\Freigabe) und jeweils ein Feld für Benutzer und Passwort. Alle 24 Stunden wird dann eine neue TXT angelegt, mit dem Datum als Dateinamen. So könnte man ohne Weiteres mehrere Tage loggen, ohne die ganze Zeit ein Browserfenster geöffnet zu haben.

MetaChuh, 2023-10-07:
@Ollipop030 
ich bin kein freund von obscuring logs.
im normalfall, kann maximal ein sehr naher nachbar dies ausnutzen.

gebe dir aber recht, bzgl. eurer angst, falls die „strompolizei“ mal die gegend abfährt, und eine liste der bekannten hm seriennummern hat, und auch in der lage ist eine scanner software zu schreiben, die alle sn‘s in allen gegenden pollt, jedoch ist die chance für dieses szenario imho derzeit sehr gering.

smb: du kannst alle serial daten live loggen,
ein zusätzliches abspeichern der daten, und bereitstellen via smb, auch wenn nur eine datei auf einem externen nas, per file append, ergänzt wird, ist imho im ersten moment ein nice to have, jedoch nicht notwendig.

gedankenweg erklärung:
es stellt eine ressourcenkonsummierende zusatzaufgabe für den esp dar, der imho in die kategorie debug, but not for many fällt

greetings, metachuh

Ollipop030, 2023-10-07:
> gebe dir aber recht, bzgl. eurer angst, falls die „strompolizei“ mal die gegend abfährt, und eine liste der bekannten hm seriennummern hat, und auch in der lage ist eine scanner software zu [...]

Das ist gar nicht der Hintergrund. Die Idee kam im o.g. Issue auf, da wir dort lange Log dateien einsenden, und ich dort die Seriennummern verschleiert habe. Also eher für die eine Veröffentlichung von Logdateien hier auf github.

> nice to have, jedoch nicht notwendig.

Ursprung der Idee ist auch hier o.g. Issue. Dort warten wir auf Alarme von Wechselrichtern, die manchmal aber erst nach Stunden auftreten, und wir so lange einen Browser laufen lassen, und später den gesamten Inhalt in eine Textdatei kopieren. Sollte auch nur für so einen Fall sein und nicht dauerhaft verwendet werden.

MetaChuh, 2023-10-09:
brainstorming:
aufteilen einiger features in 2 getrennte esps.

eine "Ahoy DTU" resilliant esp, die sich nur um die die basic nrf24 und/oder cmt kommunikation, sowie api/mqtt bereitstellung kümmern muss,
plus eine zweite esp als "Ahoy Plus", die das web interface (welches bei reger nutzung ebenfalls reboots verursacht), sowie alle weiteren zusatzfunktionen bereitstellt.

ziel: dass die primary "communication and data delivery unit" möglichst rebootless, constant, and fail safe bleibt, während die data processing (Ahoy Plus Unit), alle zusatzfunktionen beinhaltet, mit dem zusätzlichen vorteil, dass man dadurch auch nur eine Ahoy DTU betreiben muss, jedoch verschiedene iterationen einer Ahoy Plus Unit darauf zugreifen können.

was meint ihr ?
disclaimer: reines brainstorming, no pressures, no demands, no needs, no requests please.

closed 2023-12-29 by lumapu