lumapu/ahoy #1212: MQTT-Analyse

knickohr, 2023-10-10:
Ich habe mal ein bißchen mit MQTT rumgespielt und das analysiert. Dabei sind mit ein paar Punkte sauer aufgestoßen :

Erst mal. Ahoy hat mit 14 Invertern ganze 835 Topics 😵

- Warum sind die Alarm-Messages retained ?  Gibt’s dazu einen besonderen Grund ?
- Schön wäre es wenn man die Alarme in einen 4er-Päckchen zusammen fassen könnte. Also code, start, end und string in einem JSON. Macht das etwas übersichtlicher und gleich mal über 500 Messages weniger.
- Das last_success kommt mir noch zu oft. Es reicht eine einzige Message pro abgefragten Inverter, vor allem bei nicht antwortenden Invertern.
-

knickohr, 2023-10-10:
Nun, das ist (leider) in einem Satz gesagt :  Ein Projekt muß seine Topics ändern.

Schon allein die Baumstruktur ist nicht identisch 😕

stefan123t, 2024-01-07:
Die MQTT Topics von AhoyDTU sind im UserManual: https://github.com/lumapu/ahoy/blob/main/User_Manual.md
und die MQTT Topics von OpenDTU sind hier dokumentiert: https://tbnobody.github.io/OpenDTU-docs/firmware/mqtt_topics/

Müsste man mal nebeneinander stellen und mit nem diff vergleichen, was vergleichbar ist

stefan123t, 2024-01-07:
Anbei der Vergleich der beiden dokumentierten MQTT Topics.
@Knickohr kannst Du ggf. die beiden Tabellen bzw. die Topics überprüfen anhand einer aktuellen Version, 
falls Du sowohl AhoyDTU als auch OpenDTU irgendwo am Laufen hast ?

[MQTT_Topics.xlsx](https://github.com/lumapu/ahoy/files/13854007/MQTT_Topics.xlsx)
[AhoyDTU_MQTT_Topics.md](https://github.com/lumapu/ahoy/files/13854008/AhoyDTU_MQTT_Topics.md)
[OpenDTU_MQTT_Topics.md](https://github.com/lumapu/ahoy/files/13854009/OpenDTU_MQTT_Topics.md)

knickohr, 2024-01-07:
Auf die Schnelle habe ich noch den fehlenden undokumentierten Topic für Ahoy gefunden :

ctrl/power/[id] mit Payload 0/1
Schaltet den Inverter soft aus/ein, also ohne dieses Startgedöns, er ist sofort online

stefan123t, 2024-01-07:
@knickohr habe Deine Anmerkungen oben eingearbeitet.
Hier noch der Hinweis auf die Diskussion Max. Leistung aller WRs zusammen anpassen #1608 
https://github.com/tbnobody/OpenDTU/discussions/1608#discussioncomment-7988430

Hier wäre eine Zusammenfassung mehrerer Wechselrichter oder Inverter zu einem / mehreren PV-Pools sinnvoll.
Damit könnte man dann auch das Power Limit je Inverter / PV-Pool für einen gewissen Zeitraum vorgeben.
Z.B. 
* der Batterie Pool von Sonnenuntergang bis Sonnenaufgang mit 150 W Grundlast
* der restliche Zoo an PV-Panelen & Invertern bis zu einem max. Einspeisung von 600W / 800W je nachdem was der Gesetzgeber gerade für richtig befindet.
* man könnte auch den aktuellen Verbrauch (von einem SML / Shelly 3EM Pro) zu einem oder mehreren PV-Pools gegenrechnen für eine Nulleinspeisung / Zero-Export Control.

knickohr, 2024-01-07:
Ja, korrekt, das sollte aber dann direkt in der DTU passieren und nicht außerhalb. Ich möchte in Zukunft die ganze Power-Limit Gedöns nur noch innerhalb der DTU gesteuert sehen. Einzig allein die Vorgabe des Limits, keine Nulleinspeisung wo sich das Limit andauernd ändert, könnten wir per MQTT übergeben. Ansonsten nur noch lesende Ausgaben über MQTT bezüglich Limit.

knickohr, 2024-01-07:
Vorschlag :

- Die  ganzen General und Total in einem Topic
- Alarmdaten als Topic
- (Noch nicht vorhandene) Gridprofile als en Topic
- Die Inverterdatenn als ein Topic, wobei ich die Firmware und Hardwareinfos da raus lassen würde. Vielleicht zu den Gridprofilen dazu ?
- Die schreibbaren MQTT-Topics würde ich einzeln lassen

Dann hätten wir 3 inverterspezifische Topics, je nach Anzahl der Inverter. Wobei 2 ja nur einmal abgefragt werden. Aktualisierung sobald neue Daten vorhanden.

Das General mit dem Total würde einmal die Minute reichen.

wären bei meinen 16 Invertern dann insgesamt 54 Topics, wobei sich nur 17 zeitlich wiederholen. Im Gegensatz zu aktuell über 1000 Topics.

Da OpenDTU doch nicht ganz so unterschiedlich erscheint, müßte man sich nur auf die Namensgebung einigen.

knickohr, 2024-02-18:
Das sieht schon mal gar nicht schlecht aus 👍

Auch wenn jetzt die HW und Firmware-Infos als JSON kommen, müssen sie nicht retained sein. Oder ist das so wichtig ? 😉

[mqtt-retained.txt](https://github.com/lumapu/ahoy/files/14321728/mqtt-retained.txt)

lumapu, 2024-03-02:
wenn sie nicht retained sind, dann verschwinden sie nach kurzer zeit für immer oder?

knickohr, 2024-03-02:
Der Broker hält sie halt nicht vor. Genau genommen werden sie einfach nur raus geplärrt. Der Client muß sie dann einsammeln und selbst speichern.

Bei retained werden die Werte im Broker gespeichert. Und jedesmal wenn sich ein Client subscribed, bekommt er sie vor die Nase gepfeffert.

knickohr, 2024-06-02:
JSON-Payload ist mit der 0.8.123 optional auswählbar 👍