lumapu/ahoy #1213: Yield Total ist an manchen Tagen im Minus. (nur beim HM-1500 nicht beim HM-600)

stiech82, 2023-10-11:
Hallo Zusammen,

ich lasse seit einiger Zeit meine Datenan influxDB schicken, unter anderem Yield total. Jetzt wollte ich mir das in Grafana anzeigen lassen und musste sehen, dass Yield Total von Tag zu Tag weniger wird und nur an wenigen Tagen mehr wird.

Ich mache eine Nulleinspeisung, schicke also regelmäßig ein neues Power limit. Offline geht die Hoymiles nur wenn ich daran arbeite. Sie werden höchsten via DTU auf "off" gestellt. Aber sie sind immer noch erreichbar bzw. online.

Scheinbar ist das Problem nur beim HM-1500. Beim HM-600 sehen die Werte für mich realistisch aus.

HM-1500 Yield total:
04.10: 700 kWh
05.10: 645 
06.10: 611
07.10: 586
08.10: 588
09.10: 594
10.10: 562
11.10: 468 (heute bis jetzt ca. 9 Uhr morgens)

HM-600 Yield total:
04.10: 613
05.10: 615
06.10: 617
07.10: 618
08.10: 619
09.10: 621
10.10: 623

stiech82, 2023-10-11:
[image]

Diese Tabelle zeigt die Werte um wieviel sich der Wert an dem jeweiligen Tag verändert hat. (difference).

Ich hatte mich eh gewundert warum laut Weboberfläche von AhoyDTU der HM-1500 so wenig eingespeist hat und dachte er hatte vielleicht mal einen Reset als ich an der Anlage rummontiert habe. Aber ich war an der Anlage das letzte mal vor über 2 Wochen dran.

Ollipop030, 2023-10-11:
Hast du evtl. in deiner Config vom Grafana irgendwo einen Fehler? Der Yield Total kommt ja direkt vom Wechselrichter, das ist kein Wert den Ahoy aufaddiert.

> HM-1500 Yield total:
> 04.10: 700 kWh
> 05.10: 645
> 06.10: 611
> 07.10: 586
> 08.10: 588
> 09.10: 594
> 10.10: 562
> 11.10: 468 (heute bis jetzt ca. 9 Uhr morgens) (Dieser Wert steht auf der Weboberfläche von AhoyDTU)

Also diese Wete kommen von Grafana, bis auf den letzten? Oder hast du die täglich von Ahoy abgelesen?

Beobachte mal den Yield Total in Ahoy die nächsten Tage, dort sollte er ja größer werden und nicht so schwanken. Was macht denn Yield Day, zählt der normal hoch?

stiech82, 2023-10-11:
@Ollipop030 Danke erst einmal.

Das Bild war das erste was ich gesehen habe und dachte so wie du, dass ich da einen Fehler in Grafana eingebaut habe. Im Falle von dem Bild ist es die Differenz vom ersten und letzten Wert des jeweiligen Tages.
Wenn ich einfach nur den höchsten Wert "max" vom Tag nehme, dann sieht man wie die Werte von Tag zu Tag kleiner werden, aber an manchen Tagen steht dann auch mal mehr da. Hier die Ansicht wenn ich den "max"-Wert eingestellt habe.
[image]

Ich habe jetzt erst die Daten so wirklich ausgelesen. Daher konnte ich nur den aktuellen Wert von heute von Yield total aus Ahoy-DTU auslesen.
Ich behaupte jetzt einfach mal, dass ich mir sicher bin, dass dieser Wert in Ahoy DTU nicht stimmt. Ich habe immer nur mit dem HM-600 und dem HM-1500 eingespeist. Wenn es nach meinen Ladereglern geht, haben die für 898€ Energie von der PV weitergeleitet. Nehme ich aus Ahoy-DTU von beiden Wechselrichtern den Wert "Yield Total". Dann sind das gerade mal 455€.

Ich weiß, es ist mit Verlusten über den Speicher und auch von DC zu AC zu rechnen, aber die Verluste sind dann doch enorm.

Der Yield Day scheint bei beiden nicht zu stimmen. Der HM-600 zeigt aktuell 112014 Wh an (Yield total 624 kWh). Der HM-1500 zeigt 12074 Wh an (Yield total 469 kWh).
Der Speicher war heute früh leer und die PV hat heute weniger als 5kWh produziert.

kiu77, 2023-10-11:
Notiere doch mal für ein paar Tage die Yield Total Werte, die Ahoy-DTU anzeigt

stiech82, 2023-10-12:
> > Wenn es nach meinen Ladereglern geht
> 
> Ahso, du hast den 1500 an einem Akku. Hier gab es mal ein Issue, da kamen auch krumme Werte bei der Leistung in Verbindung mit einem Akku: #1168 Aber keine Ahnung woran es da gelegen hat.

Ja das Ticket habe ich auch schon gesehen, aber das ist ja der AC Power und nicht der Yield-Wert.
Bei mir passt der AC-Power bisher was ich so gesehen habe. Ich habe am AC noch einen Shelly hängen und wenn ich die Werte  mit AC-Power von Ahoy-DTU vergleiche, dann passt das relativ gut.
Selbst wenn er nicht passen würde, der müsste ja ins Minus gehen damit der Yield-Wert runter geht.

Aktuell zeigt der HM-1500 Yield Day 16264Wh an. Ich frage mich halt woher der Wert kommen soll. Liegt der Hoymiles wirklich so daneben oder liegt da ein Fehler in Ahoy-DTU vor? Wie erwähnt, die AC Power passt.

stiech82, 2023-10-12:
> Notiere doch mal für ein paar Tage die Yield Total Werte, die Ahoy-DTU anzeigt

Danke dir. Ja das mache ich jetzt auch mal, aber ich bin mir sicher, dass die Werte in MQTT richtig übertragen wurden. Der Yield Total ist viel zu niedrig im Vergleich zu dem was die PV bisher produziert hat.
Wie kann das sein, sollte der Fehler nur bei MQTT liegen, weil dann doch nur die Möglichkeit besteht, dass der Hoymiles sich genullt hat. Aber wie kann das passieren? Ab Kauf waren die immer in Betrieb, höchstens maximal wenige Stunden nicht.

Ollipop030, 2023-10-12:
> liegt da ein Fehler in Ahoy-DTU vor?

Ich schätze, das kann man auschließen. Sonst hätten sich hier schon mehr Leute gemeldet. Ich habe selbst einen HM1500 (an Modulen, nicht am Akku), Ahoy zeigt bei mir die Yield Day fast genau so wie der Zwischenzähler im Zählerschrank, Abweichungen sind täglich nur im Promillebereich.

Du scheinst aber auch im Forum vom Akkudoktor aktiv zu sein:
https://www.akkudoktor.net/member/stiech82/activity/

Bist das du? Vielleicht ist bei der ganzen Bastelarbeit irgendwas im Wechselrichter hochgegangen? Du scheinst da ja sehr aktiv zu sein.

stiech82, 2023-10-12:
> > liegt da ein Fehler in Ahoy-DTU vor?
> 
> Ich schätze, das kann man ausschließen. Sonst hätten sich hier schon mehr Leute gemeldet. Ich habe selbst einen HM1500 (an Modulen, nicht am Akku), Ahoy zeigt bei mir die Yield Day fast genau so wie der Zwischenzähler im Zählerschrank, Abweichungen sind täglich nur im Promillebereich.
> 
> Du scheinst aber auch im Forum vom Akkudoktor aktiv zu sein: https://www.akkudoktor.net/member/stiech82/activity/
> 
> Bist das du? Vielleicht ist bei der ganzen Bastelarbeit irgendwas im Wechselrichter hochgegangen? Du scheinst da ja sehr aktiv zu sein.

Ja das bin tatsächlich ich. Immer diese Stalker ;). Nein Spaß.

Das komische ist halt, dass Yield Total nur beim HM-1500 so extrem falsch ist. Beim HM-600 sieht es realistisch aus.
Den HM-600 habe ich schon länger und mit dem habe ich hauptsächlich rumprobiert und experimentiert. Als dann alles super lief, habe ich einfach den HM-1500 ergänzt.

Die Frage aber auch ist: Was soll da hochgegangen sein? Hat der HM-1500 getrennt vom AC-Power einen extra Zähler als Hardware für Yield?
Ich hätte gedacht der nimmt einfach die AC-Power und rechnet die in Energie um. Zumindest frage ich mich, wieso die das anders machen sollten. Macht ja keinen Sinn zwei verschiedene Zähler für AC-Power und den Yield-Werten zu nehmen.
Wenn dann stimmt mit der Software was nicht, aber das würde ja bedeuten, dass die Software so einen knacks abbekommen hat, dass sie AC-Power in eine falsche Formel packt und deswegen falsche Yield-Werte ausgibt.

Natürlich alles nur Vermutung von mir.

Wenn die AC-Power auch falsch wäre, dann würde ich dir zu 100% zustimmen und vermuten, dass da was am WR kaputt ist. Aber so....... ?

stiech82, 2023-10-12:
OK ich habe vielleicht einen möglichen Fehler. Ich habe die Werte "Yield total" von ch0 genommen und nicht von "total". Ich lass jetzt mal beide Werte von beiden WR in die Datenbank schreiben.

Nach MQTT sehen die beide gleich aus, aber vielleicht wird da wirklich mal nach und nach was anderes übertragen.

Oder sollte ch0 und total gleich sein?

kiu77, 2023-10-12:
;-)

lumapu, 2023-10-15:
> Oder sollte ch0 und total gleich sein?

bei einem Wechselrichter ja, bei mehreren nicht. Je Wechselrichter ist der Wert YieldTotal monoton steigend, niemals fallend.

stiech82, 2023-10-16:
Ich habe jeweils einen WR über eine DTU laufen. Also in dem Sinne nur ein WR.

PS: Seit ich hier gefragt habe ging der Wert auch nicht mehr ins Minus. Aber wenn ich zurück schaue, passiert das scheinbar auch nur sehr selten.

kiu77, 2023-10-16:
Wie gesagt: An AhoyDTU kann das kaum liegen, denn die Software liest nur die Daten des WR aus...

stiech82, 2023-10-16:
Ja danke, aber woran liegt es dann? Ich will ja nicht wissen woran es nicht liegt, sondern woran das liegt.

Aktuell bin ich an dem Punkt: AhoyDTU oder der WR selbst übermittelt falsche Werte. Oder es gibt gewisse Punkte bzw. Situationen an denen der Wert zurückgesetzt wird.
Nur ich kann kein Muster erkennen. Ich hatte mal mit dem Akkustand verglichen. Mal war der Akku gut gefüllt, mal war er früh leer, mal war er abends leer.... mal geht Yield total an dem Tag ins Minus und mal nicht.

Achja, vergessen zu erwähnen. Tatsächlich hatte ich auch eine falsche Einstellung in Grafana. Aber jetzt nach der Anpassung, passen die meisten Tage. Am Ausgang der beiden Wechselrichter hängt ein Shelly. Die Daten vom Shelly und den zwei WR passen so gut wie jeden Tag, außer eben mal hier und da ein Tag bei dem der HM-1500 mit Yield total im Minus ist.

Ollipop030, 2023-10-16:
> > Notiere doch mal für ein paar Tage die Yield Total Werte, die Ahoy-DTU anzeigt
> 
> Danke dir. Ja das mache ich jetzt auch mal,

Was ist denn dabei rausgekommen, ist das einigermaßen plausibel?

stiech82, 2023-10-18:
> > > Notiere doch mal für ein paar Tage die Yield Total Werte, die Ahoy-DTU anzeigt
> > 
> > 
> > Danke dir. Ja das mache ich jetzt auch mal,
> 
> Was ist denn dabei rausgekommen, ist das einigermaßen plausibel?

Bis jetzt sehen die Werte in MQTT und Ahoy-DTU normal aus. Zumindest beim HM-1500. Jetzt ist der HM-600 plötzlich im Minus für heute. Leider habe ich mir von dem die Werte von AhoyDTU nicht notiert.

stiech82, 2024-03-01:
Ich denke das Thema hat sich erledigt.
Seitdem läuft alles super und keine Werte haben sich erneut zurück gesetzt oder sind nicht plausibel.

Vielleicht habe ich auch einfach nur ausversehen die Werte zurückgesetzt...

DanielR92, 2024-05-19:
Könnte man dieses Issue dann nicht closen?

stiech82, 2024-05-19:
Es kam eine lange Zeit nicht mehr, aber jetzt ist es wieder passiert. Bei einem Wechselrichter sogar 3 Tage hintereinander wurde jeden Tag der Wert kleiner, so das mein SmartHome ein Minus an diesem Tag ausrechnet. Ich lass mir die Differenz vom ersten und letzten Wert des Tages anzeigen.
Ein paar Wochen vorher ist es auch ab und zu beim anderen WR passiert.

Wenn ich über die IP in den WR schaue, stimmt zumindest der letzte Wert mit dem Wert der in meinem SmartHome gespeichert wurde, überein.

closed 2024-08-16 by lumapu