lumapu/ahoy #1214: Restarting when in Live-view [Bug]

technics42, 2023-10-11:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.65

### Github Hash

50ca076

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Inverter intervall 20 sec
Polling of inverter data by Rest-api 20 sec

### Error description

When in Live-view, sometimes ahoy-dtu is restarted.
Version was up and running for 2-3 days without any problems, then yesterday when in Live-view ahoy was suddenly restarted.
Today this already happened twice.
Problem never occurred on version 0.7.36 with same settings/setup.

MetaChuh, 2023-10-12:
@technics42 

i can confirm your experience.
as long as only api polls are used, the dtu seems (more or less regardless of the version) rock solid, with few reboots, or scenarios where the api is delivering „null“.

since 0.6.9 i‘ve personally separated the html part to another system running apache2.4
(if of interest: in my case a synology ds918+, but i guess any apache or nginx setup on any device would work the same way, except only apache setups supporting classic .htaccess)

as long as i limit myself to setups on the native gui, and only use it, while not polling the api constantly from any logger, the rest of the subsystem seems to be rock solid, almost regardless of the ahoy version used.

off topic and private: this rock solid behaviour with a few tweaks, might be, what led me to be sometimes over defendant, regarding this project and our devs, despite of finding out issues, and many of us trying to publish their necessary workarounds.
i apologize to you, for one of my answers a few weeks ago.

technics42, 2023-10-12:
Thank you for your comment! No need to apologize.
A few weeks ago this project was new to me and I thought it would already be a very stable system without any need for tweaks...
Seems like you wrote, in most cases it is stable, but there is still some way to go.

I am also struggling with crashes due to webserver for an own project. (also using async-webserver)
It seems like such crashes can appear, if you refresh the webpage yourself, while another request is already triggered/handled. And maybe this also depends on the clients browser.
So if you keep page open (e.g. live-view) and you are doing a refresh by browser while update was triggered from server, this crash may occur.
But I can't really prove it, currently it's just a possible cause.

Peertje-62, 2023-10-13:
Hi, had the same problem, #1208 
7.66 / ESP8266 & ESP32
ESP8266 was okay, ESP32 was not, 
Did not make the connection between access GUI & Reboot cause
changed from JSON to MQTT
runs fine now (both)
regards

closed 2023-12-29 by lumapu