lumapu/ahoy #1216: [Bug] Power Limit wird via MQTT nicht gesendet

Alech385, 2023-10-18:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

 0.7.36

### Github Hash

ab1234

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Setup ist soweit i. O. Kommunikation zu den Invertern ist einwandfrei. 

### Error description

Hallo, ich habe einen HM-600 Inverter die Kommunikation mit der AhoyDTU (ESP32) sowie mit meinem Smart Home System läuft via MQTT einwandfrei. Leider habe ich das Problem, dass ich den Wert "Power Limit" nicht erhalte. Also das Topic: 

"<TOPIC>/<INVERTER_NAME_FROM_SETUP>/ch0/PowerLimit"  wird einfach nicht aufgerufen, alle anderen Werte jedoch schon. Hat jemand das gleiche Problem? Ich benötige den Wert, da ich eine Nulleinspeisung umsetzen möchte. 

Vielen Dank im Voraus! 

Grüße

lumapu, 2023-10-29:
sind die anderen topics aktuell?
Nur zur Kontrolle: Du willst den gelesenen Power Limit Wert wissen, der automatisch per MqTT gesendet werden sollte?
Was steht in den Feld oder gibt es das überhaupt nicht?

Alech385, 2023-10-30:
Die anderen Topics sind aktuell. Das Topic für PowerLimit wird  über MQTT irgendwie nicht gesendet.

lumapu, 2023-10-31:
bei mir ist der Wert da, wurde gleich bei der ersten Inverterkommunikation aktualisiert:
[image]

Alech385, 2023-11-11:
Hallo nochmal, 
ich bekomme tatsächlich nur folgendes vom Umrichter: 
[image]

Alech385, 2023-11-11:
Hi, ich habe meinen fehler gefunden. Das Topic gibt mir ja nur eine Rückmeldung, wenn ich den Wert für PowerLimit selbst setze.

closed 2023-11-11 by Alech385