lumapu/ahoy #1218: [Bug] IndexError: bytearray index out of range

haufsolutions, 2023-10-22:
### Platform

RaspberryPi

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

latest

### Github Hash

e5d78bf5b42363751bc3956c0ce465d1f75d43bb

### Build & Flash Method

was already installed

### Setup

mqtt:
host: edited
username & password: ''

inverters:
serial: edited

### Debug Serial Log output

```shell
xxx:/home/pi/ahoy/tools/rpi# python3 -um hoymiles --log-transactions --verbose --config ahoy.yml | tee -a log2.log
Using python Module: RF24
Traceback (most recent call last):
  File "/home/pi/ahoy/tools/rpi/hoymiles/__main__.py", line 153, in main_loop
    poll_inverter(inverter, dtu_ser, do_init, transmit_retries)
  File "/home/pi/ahoy/tools/rpi/hoymiles/__main__.py", line 206, in poll_inverter
    while com.rxtx():
  File "/home/pi/ahoy/tools/rpi/hoymiles/__init__.py", line 660, in rxtx
    for response in self.radio.receive():
  File "/home/pi/ahoy/tools/rpi/hoymiles/__init__.py", line 432, in receive
    time_rx=datetime.now()
  File "/home/pi/ahoy/tools/rpi/hoymiles/__init__.py", line 262, in __init__
    if f_crc8(payload[:-1]) != payload[-1]:
IndexError: bytearray index out of range
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/ahoy/tools/rpi/hoymiles/__main__.py", line 416, in <module>
    main_loop(ahoy_config)
  File "/home/pi/ahoy/tools/rpi/hoymiles/__main__.py", line 153, in main_loop
    poll_inverter(inverter, dtu_ser, do_init, transmit_retries)
  File "/home/pi/ahoy/tools/rpi/hoymiles/__main__.py", line 206, in poll_inverter
    while com.rxtx():
  File "/home/pi/ahoy/tools/rpi/hoymiles/__init__.py", line 660, in rxtx
    for response in self.radio.receive():
  File "/home/pi/ahoy/tools/rpi/hoymiles/__init__.py", line 432, in receive
    time_rx=datetime.now()
  File "/home/pi/ahoy/tools/rpi/hoymiles/__init__.py", line 262, in __init__
    if f_crc8(payload[:-1]) != payload[-1]:
IndexError: bytearray index out of range
```

### Error description

I installed the RPI method using the NRF24 wrapper on Debian 10.
I noticed the error will mostly vary if I switch between  tzlocal versions. 
Or maybe I'm missing something?

lumapu, 2023-10-31:
the Raspberry version is only slightly maintained. The focus of this repository is most regarding ESP8266 and ESP32. Hopefully somebody is out there who can help you. I mention @PaeserBastelstube because he provided the last merged PR.

stefan123t, 2024-10-29:
@haufsolutions what is your current status with this issue can you provide some more information on your issue ?

@PaeserBastelstube Knuti can you maybe help the user of the Raspberry Pi here, does he use the right python version ?

PaeserBastelstube, 2024-10-30:
Hi @haufsolutions,
please check outdated python modules:
```
python3 -m pip list --outdated
sudo python3 -m pip list --outdated
```
You start `python3` as user, maybe there are different module versions between `system` and `user `environment.
.
Question: What is the reason that you want to change the time zone?
The log file shows an error in the inverter payload. When AHOY requests data from the inverter, the request data contains a Unix timestamp. Changing the tzlocal between request and response may result in an error you reported.