lumapu/ahoy #1220: Bad Connection to Hoymiles HM1500

Starfoxfs, 2023-10-25:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ESP32-D0WDQ6
Retailer URL: https://www.berrybase.de/esp32-nodemcu-development-board?c=2473

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [x ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.7.36
Github Hash: _______

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

I have a question about the NRF24L01+ with onboard antenna. My Hoymiles HM1500 is about 4 meters away from the NRF24 module and I have extremely poor reception. I have now clamped the AZ Delivery Adapter Board for the power supply between ESP32 and NRF24 and since then I have had even worse reception than without the power supply board. I haven't tried a capacitor yet. What else could be done to improve reception?

[image]

stefan123t, 2023-10-25:
Try the capacitor :)

Joh49, 2023-10-25:
Try an other NRF24L01+ and if possible with the original chip

knickohr, 2023-10-25:
Rempeln the adapter board and connect directly with short wires, better solder it ! Don’t forget the capacitor.

Starfoxfs, 2023-10-27:
Now i have bought some Panasonic 16V 100μF with 5000Hour Lifetime, soldert it to the NRF24 and the Wire Connections so short as possible.

What can i say, no changes no Signal improvements

I have many RX Fragments, many TX retransmits and RX success every time about 70% of TX counts

e.g.
TX Count 16
RX Success 11
RX Fails are 0
RX no answer 5
RX fragments 34
TX retransmits 14

knickohr, 2023-10-27:
Only 16 TX counts ?

Make some hundert TX and we discuss again 😉

Starfoxfs, 2023-10-28:
Ich schreib mal auf Deutsch ist ja einfacher.

Inzwischen geht es sehr gut, ich hab den Stromadapter von AZ-Delivery wieder entfernt, einen 16V 100μF Kondensator aufgelötet und die Leitungen sehr kurz gehalten und angelötet.

Jetzt funktioniert es mit sporadischen Verlusten sogar durch eine Betonwand. Das ist für mich ausreichend gut.

Aktuell steht der Empfänger sowieso näher und empfängt sehr gut.

knickohr, 2023-10-28:
Ach, sieh mal an 😉

Laß jetzt mal so ca. 1000 TX durchrauschen und dann geh mal mit dem Power-Lever hoch oder runter und beobachte wieder 1000 TX.

Es gibt einen „Sweet Spot“, bei denen die fail und no answer am geringsten sind. Das ist dann die optimale Einstellung.

MetaChuh, 2023-10-29:
bestätige @knickohr 's beobachtungen und tipps.

ist leider nicht, dass wir eine perfekte eierlegende wollmilchsau einstellung für alle haben, sondern dass viel individuelles trial and error dahinter steckt, weil vieles was bei a funzt, bei b abkackt
(kann dann so vieles sein, angefangen von 2.4ghz von wlans bis wetterstationen der nachbarn dicht an der location, bis nrf modul unterschiede, und mehr)

glaub' wichtig ist, dass wir wie bisher alle unsere infos teilen, wie ähnliches gelöst oder ein workaround gefunden wurde.

info: der freak kern von uns hat mehrere ahoy dtus mit verschiedenen nrf modulen gleichzeitig in betrieb,
um zb. mehrere inverter auf einzelne dtus in gruppen aufzuteilen, und/oder mit den gleichen invertern, parallel neuere firmwares zu vergleichen, esp und nrf modul unterschiede der verschiedenen lieferanten zu testen etc.

da die hardware für 'ne 2. dtu nicht so viel kostet, ist meine empfehlung, dir eine 2. aufzubauen und hardware und settings ergebnisse mit deiner bestehenden zu vergleichen und dann zu posten.

ps: auf wunsch posten wir auch hier links der bauteile mit der wir gute erfahrungen hatten, die nicht auf ahoydtu.de gelistet sind.

greetings metachuh

closed 2023-12-29 by lumapu