lumapu/ahoy #1221: 0.8.1 keine Kommunikation mit den Invertern

Ollipop030, 2023-11-05:
Moin moin,

grad die 0.8.1. auf einem ESP32 installiert. Hier scheint es genau einmal zu einer Kommunkation zu kommen, wenn die DTU startet, danach nicht wieder.

Es werden viele Fehler gezählt:

[image]

Hier noch ein kurzes Log:
[Log.txt](https://github.com/lumapu/ahoy/files/13259779/Log.txt)

knickohr, 2023-11-05:
Ich würde da mal ansetzen :  15:14:45 W: local buffer to small for payload fragments

Sieht so aus als ob das Datenpaket von Deinem 4-Ch WR nicht mehr rein paßt.

dtuuser, 2023-11-05:
Bei mir nicht mehr die 10 Minuten Pausen wie bei der 7.66. Aber manchmal 2 bis 3 Minuten kein Empfang

lumapu, 2023-11-05:
ok, Problem erkannt. Es wurden maximal 10 Fragmente zugelassen. In der nächsten Version wird das verdoppelt. Diese Alarmabrage bereitet mir zunehmend Kopfschmerzen .. ich hoffe wir bekommen das vernünftig in Griff.

Gubi2023, 2023-11-08:
bei mir sind auch mit 8.1 die 10-minütigen Pausen noch vorhanden... und anschliessend default-Zeit 
[image]

knickohr, 2023-11-08:
Du hast keine Unterbrechung von +10 Minuten, sondern von -50 Minuten. Da stimmt was mit der Uhr (NTP ?) nicht !

rmayergfx, 2023-11-08:
@knickohr 
was bedeuten diese Werte?
11:48:35 I: Status: | -6 0 -6 -6 -6
11:50:05 I: Status: | -6 2 -6 -6 -6

Das wechselt bei mir ständig hin und her und 

Uptime: 2 Days, 19:10:49

kann ich auch nicht nachvollziehen, woher kommt das 19:10:49?

Wo finde ich dazu informationen?

knickohr, 2023-11-08:
Das ist die Heuristik. Er versucht den besten Sendekanal aud den 5 möglichen zu ermitteln. Eine -6 ist schlecht, eine 4 gut. Es sollte mindestens ein Kanal mit >0 dabei sein, dannn steht eine gute Verbindung.

Mach Dir da mal keinen Kopp, das ist ganz neu und der allererste Wurf. Dashalb auch noch keine Doku.

2 Tage, 19 Stunden, 10 Minuten, 49 Sekunden Uptime 😅

Ollipop030, 2023-11-10:
Hab heute morgen die 0.8.3 installiert, der HM1500 hat wieder eine stabile Verbindung, allerdings werden fleißig Fehler gezählt.
[image]

Komischerweise aber nicht beim HM300:
[image]

Ollipop030, 2023-11-14:
Bezüglich des Fehlerzählers habe ich nochmal einen Vergleich gemacht, Power Level auf LOW.

Hier die Version 0.8.7:
[image]
[image]

Und hier die Version 0.7.64:
[image]
[image]

Mal eine Frage in die Runde: Bin ich der einzige mit dem Phänomen? Ich hatte erst meine Nulleinspeisung in Verdacht, weil ich hier ständig Timeouts bekomme beim Limit setzen mit den Versionen >0.8. Die habe ich für diesen Vergleich aber ausgeschaltet.

Gubi2023, 2023-11-14:
Bei mir sieht es auch nicht viel besser aus: DTU hat sich vor 2,5 Std neu gestartet, seither nachstehende Statistik:[image]

knickohr, 2023-11-14:
@Ollipop030 

RX fail in Verbindung mit sehr wenige No Answer deutet darauf hin das Du mit zu wenig Power sendest. Denn würde Dich der Inverter verstehen und nicht antworten, dann wäre es genau anders herum.

Aaaber wenn ich mitbete Summe aus fail und no answer anschaue, dann ist die neue Version nicht schlechter als die alte. Denn beide Summen sind ungefähr gleich, was ja auch die Anzahl der success deutlich macht.

rmayergfx, 2023-11-14:
@knickohr 
Danke für den Hint. Ich habe gerade bei mir auch mal nachgesehen und musste leider feststellen, das irgendeines der Updates von 0.7.59 auf 0.8.xx mir die Einstellungen im Bereich Inverter gelöscht hat. Ich hatte plötzlich dort nur noch MIN (recommended) drinstehen und auch "Pause communication during night " war wieder deaktiviert.
Aktuell bin ich auf 0.8.7 und die Werte sind nun, nachdem ich wieder auf HIGH gesetzt habe wesentlich besser geworden.

> CH qualities: 4 -6 -4 -6 -6

@lumapu 
Könntest du bitte im Popup zu den Radio Statistics die eingestellte Sendeleistung mit anzeigen, damit man nicht erst in die Untermenues muss?

Ollipop030, 2023-11-14:
> das Du mit zu wenig Power sendest

Danke, probiere ich mal. In der Vergangenheit hieß es ja, so niedrig wie möglich.

> dann ist die neue Version nicht schlechter als die alte

Bei meiner Nulleinspeisung schon. Die 0.7er Versionen laufen wunderbar, die 0.8er Releases sind für mich leider unbrauchbar, damit bekomme timeouts im Minutentakt. 

> Könntest du bitte im Popup zu den Radio Statistics die eingestellte Sendeleistung mit anzeigen, damit man nicht erst in die Untermenues muss?

Könnte ahoy nicht anhand der hohen Fehlerzahl die Sendeleistung selbst einstellen?

knickohr, 2023-11-14:
Soweit sind wir noch nicht. Erst mal muß die Kommunikation (wieder) stabil funktionieren.

Ollipop030, 2023-11-14:
Kurzer Zwischenstand mit Powerlevel HIGH:

[image]

[image]

Übrigens Kondensator ist verbaut, externe Antenne vorhanden. 
Probiere morgen mal noch ein wenig herum, jetzt ist dunkel.

Falls irgendwelche Logs benötigt werden sagt bescheid.

Gubi2023, 2023-11-14:
> Kurzer Zwischenstand mit Powerlevel HIGH:

Wie hast du den Power Level auf high bekommen? bei mir kommt Fehlermeldung beim speichern....

Ollipop030, 2023-11-14:
> Wie hast du den Power Level auf high bekommen? bei mir kommt Fehlermeldung beim speichern....

Die Fehlermeldung hab ich nicht. Ich kann das problemlos wechseln, bei beiden Invertern.

rmayergfx, 2023-11-14:
Dito kann ich so bestätigen. Habe schrittweise auf dem ESP32 auf 0.8.7 erhöht. Keinerlei Probleme beim Speichern.

Gubi2023, 2023-11-16:
Merkwürdig: mit dem Handy kann ich die Änderungen speichern, nur am PC beomme ich eine die Fehlermeldung....

rmayergfx, 2023-11-16:
> Merkwürdig: mit dem Handy kann ich die Änderungen speichern, nur am PC bekomme ich eine die Fehlermeldung....

Welches Betriebssystem? Welcher Browser? Hast du schon mal den Cache am Browser gelöscht? Das ist ganz wichtig, gerade bei neuen Releases. Manchmal hilft auch die Seite zwangsweise neu zu laden (Mit Shift und Reload Button) der einzelnen Seite. Damit zwingst du diese Seite neu zu laden ohne den lokalen Cache zu nutzen. 
Bevor ich es ganz vergesse, natürlich auch den eventuell vorhandenen Adblocker/Skriptblocker abschalten, da passieren oft auch merkwürdige Dinge.

closed 2023-12-29 by lumapu