lumapu/ahoy #1223: Update dev ahoy_v0.8.2

Peertje-62, 2023-11-08:
Hi, Sorry to bother but 
last ahoydtu_dev.zip contains ahoy_v0.8.1 instead of ahoy_v0.8.2
Kind regards

## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [x ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2023-11-09:
Last build has failed.

Peertje-62, 2023-11-09:
Okay, Thx

closed 2023-11-09 by Peertje-62