lumapu/ahoy #1224: 0.8.4 / ESP8266 Fehler in Settings

mtet2, 2023-11-11:
## Hardware
  * [*] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [*] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [*] external antenna (SMA)

### Power Stabilization:
* [*]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.4
Github Hash: efdac96

### Build & Flash Method:
* [*] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

0.8.4 / ESP8266 Fehler in Settings

„It's not a bug, it's a feature“?

unter 
- Settings/Radio fehlt der Powerlevel
- Settings/Inverter kann man die Inverter nicht mehr editieren

Es sind 3 Inverter eingetragen:
enabled     HM1500
disabled    HM600
enabled     HM600

Unter der 0.8.4 werden von den zwei Invertern keine Daten mehr empfangen

Wieder zurück auf die 0.8.3 (OTA geflasht) und die Settings aus der Sicherungsdatei geladen, funktioniert es wie vorher.

Mein Firefox_119.0 läuft unter Linux.

Und Danke für das gelungene DTU-Projekt

Otto-Normal, 2023-11-11:
Zur Info, auch das WiFi-Symbol unten links bleibt dunkel und beim nRF24L01-Modul fehlt die Einstellung vom Powerlevel.

Betsen Dank!

knickohr, 2023-11-11:
Ich habe auch gerade ein Update auf die 0.8.4 gemacht. Da hängen 16 Inverter dran und alles funktioniert auf Anhieb. Allerdings mit einem ESP32 !

Das Power Level ist jetzt unter Settings/Inverter/Edit/Radio 😉
Ach ja, beim ESP8266 werden nur 2 Inverter unterstützt. Ist das vielleicht das Problem ?
Das dunkle WiFi-Symbol ist nur ein Schönheitsfehler, es funktioniert wie bisher (Anzahl der Boden im Symbol).

MetaChuh, 2023-11-11:
@mtet2 @knickohr 
am esp8266 hab' ich das auch gerade gemerkt:
inverter lassen sich löschen aber nicht editieren oder hinzufügen
auch wenn nur noch ein inverter vorhanden ist, lässt sich dieser nicht editieren und kein weiterer hinzufügen

MetaChuh, 2023-11-11:
@knickohr 

was passiert wenn du deinen letzen inverter mit 0.8.4 am esp32 rauslöscht ?
lässt sich dann dieser mit add inverter wieder hinzufügen ?

ps: letztes mal waren's noch moderate 13 inverter bei dir ... hammer plus 1 inverter pro monat ... und ich dümple noch bei 5 rum 😂
thx & greetings

Gubi2023, 2023-11-12:
> inverter lassen sich löschen aber nicht editieren oder hinzufügen
  
geht mit 8.5 immer noch nicht, somit auch kein Zugriff auf Powerlevel-Änderung....

lille2000, 2023-11-12:
Der Powerlevel ist seit einiger Zeit unter „live“ zu finden.

jedoch kann ich das grundsätzliche Problem bestätigen; „edit“ und add inverter“ haben bei mir auch mit 0.8.5 keine Funktion.

Gubi2023, 2023-11-12:
> Der Powerlevel ist seit einiger Zeit unter „live“ zu finden.

???? (finde ich nicht)

lumapu, 2023-11-12:
ist in dem dritten Tab in den Einstellungen des Inverters zu finden

mtet2, 2023-11-12:
Bei der 0.8.5 und der 0.8.6 (die sich auch als 0.8.5 ausgibt) sind die Fehler:  „edit und add inverter“ bei mir noch vorhaden.
ESP8266

lumapu, 2023-11-12:
ok, teste ich bei Gelegenheit nochmal

lille2000, 2023-11-12:
Unter „APC“ (zwischen Inverter Namen und „Alarm“)

lille2000, 2023-11-12:
> > finden.
> 
> ????

Sorry, ich hatte es missverstanden und den Sende-PowerLevel mit dem Inverter-Power-Beschränkungen vertauscht . Letzter sind unter Live/APC und erstere grad nicht erreichbar unter „edit“

MetaChuh, 2023-11-13:
@lumapu am esp8266 geht auch unter 0.8.6 das inverter anlegen und editieren nicht, auch wenn man nur einen bestehenden inverter eingetragen hat.

ob bei esp32 das inverter löschen und neu anlegen, sowie editieren geht, weiß ich nicht.
hab' grad nur einen esp32 da und der ist und bleibt derzeit produktiv mit 0.6.9

hast du einen 8266 bei dir zum testen ?

lumapu, 2023-11-13:
ja logisch 😉. Fehler ist auch schon gefixt und wird heute noch veröffentlicht

lille2000, 2023-11-14:
Funktioniert mit 0.8.7 wieder 👍🙏

MetaChuh, 2023-11-14:
@mtet2 
funzt mit 0.8.7 auch bei dir alles beschriebene und kannst dieses issue schließen ?

mtet2, 2023-11-17:
Mit der 0.8.8 ist es wieder IO.
siehe: https://github.com/lumapu/ahoy/issues/1232

closed 2023-11-17 by mtet2