lumapu/ahoy #1226: 0.8.6 Settings / Inverter / Edit und Add Inverter ohne Funktion

elisenjens, 2023-11-13:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.6

### Github Hash

a82777b

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Standard

### Error description

Edit und Add Inverter ohne Funktion; Einstellen der Sendepower nicht möglich

MetaChuh, 2023-11-13:
duplikat zu: https://github.com/lumapu/ahoy/issues/1224

MetaChuh, 2023-11-14:
@lumapu
[0.8.7](https://github.com/lumapu/ahoy/actions/runs/6856740765) (commit https://github.com/lumapu/ahoy/commit/1bc3a0f06f5a6e3fd5b0ad87c703110751b63afa) am 8266 getestet, funktioniert und sieht sehr nice aus, danke 👍

@elisenjens 
für dich auch ?
wenn je dann bitte das issue schließen, danke

closed 2023-11-14 by elisenjens